AllskyTeam/allsky #1167: Not saving images

RLWagner, 2022-04-17:
I built a new Bullseye SD card (64GB) for a RPi3, cloned allsky and followed the instructions for the install of the basic package and the webUI.  I installed it to a different user location /home/wagnerr/allsky as I don't use the pi login.  I had to rebuild sunwait as it was giving the Issue 1132 error.  Then allsky looks like it's working - it shows a live image, saves a live image in ~/allsky/tmp/image.jpg.  However the ~/allsky/images folder was never created and, after I created it manually, no images are ever stored there.  Config is to shoot daytime autoexposure images every 300s
Possibly unrelated: on start up allsky was warning that maxgain =200 is not a legal value - I edited /etc/raspap/settings_ZWO.json to correct both daymaxgain and nightmaxgain to 100 which seemed to correct the warning, though now the program on startup seems to spend a lot more time (~20sec) processing settings_ZWO.json.  
Also it warns that  "Value of 50 greater than max value allowed (20) for control 'Offset' (#5)."  In this case the only 'Offset' I can find in settings_ZWO.json is set to 0.

EricClaeys, 2022-04-23:
@RLWagner   Please review the Wiki page on Reporting Issues and attach the log file.  It should give an indication of what's happening (or not happening).  Also, I assume DAYTIME_SAVE is set to "true" in config/config.sh?

I suggest never editing the /etc/raspap files directly - instead, always use the WebUI.  It does some post-processing after changes are made.  Not sure why it would take 20 seconds to process the file unless the changes you made introduced a formatting problem.  Try updating the file via the WebUI - it will overwrite the file.

After you do the steps in the Wiki page, look in the top of the log file, it will list all the capabilities of the camera including their number (1 to x).  What it calls "Offset" we call "Brightness".  The max allowed varies depending on camera - on my ASI178MC the max is 600.

EricClaeys, 2022-05-12:
@RLWagner   Any update?

RLWagner, 2022-05-12:
Hi Eric:
No update right now, I got pulled away by other commitments.  I am going 
to try reinstalling from scratch and will debug from there.

Rick

EricClaeys, 2022-06-03:
@RLWagner   Rick, any update?

EricClaeys, 2022-06-19:
Closing Issue - no update from user in over a month.
The `saveImage.sh` script should create the allsky/images folder - if not, there will be a message in the log file for every image; please check there.  It would likely be a permissions issue.

The gain and offset should be lowered for this camera per my prior comment.  The next version of Allsky will know each camera's minimum and maximum values and will produce a message in the WebUI which will make fixing things easier.

closed 2022-06-19 by EricClaeys