AllskyTeam/allsky #1169: soft temperature limit

pmccudden, 2022-04-18:
My hardware (Pi 3 Model B+ Revision 1.3 (1 GB)) and the AllSky software are all working, but I get a "Soft temperature limit currently active/Soft temperature limit has occurred" message almost continuously.  I'm not sure why (or even if this is something I should be concerned with) - I'm not running any other software or hardware as far as I know.  Any help would be appreciated.  Thanks in advance - Paul

EricClaeys, 2022-04-19:
@pmccudden,  Paul, I assume you are on the "System" page of the WebUI?  It does a `sudo vcgencmd get_throttled` which returns a hexadecimal code.  0 means everything is fine.  The message you are getting is basically saying your Pi is too hot and it was too hot in the past.  **What is the CPU temperature?**
Do you have a fan to cool the Pi?

Mine currently says `Soft temperature limit has occurred` which means it was too hot in the past (I believe since the last boot) but is ok now.

You can get more info [here](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#maximum-power-output).  It's not very useful regarding the messages though.

If you are ok with this answer, would you please close the issue?

In the future, if you have a question like this, please post it in the Discussions link above.  I'm trying to keep the Issues list for actual bugs.

Thanks - Eric

pmccudden, 2022-04-19:
Thanks, Eric.  The temperature was up to 60 C or so in an indoor dark room, and since this board will be in constant sunlight in a case I'm worried that it will overheat.  I will post in Discussions from now on, but I thought this might have been a software issue causing the board to run hot.

closed 2022-04-19 by pmccudden