AllskyTeam/allsky #1176: Timelapse, Keogram, Startrails Only Show in the Admin Panel

gktozer, 2022-04-24:
Good morning and thanks for the absolutely wonderful application. I have an issue i am sure is simply related to configuration.
My Admin Panel / Images shows the Images, Timelapse, Keogram, and Startrails perfectly, and has accumulated them successfully over the last 3 nights. However the website only shows the live image. The others simply show "No xxx images" where xxx is timelapse, keogram, or Startrails, depending on which left-hand menu selection was made.
Any help would be greatly appreciated!

gktozer, 2022-04-24:
And sorry for putting this on the issues page vs. discussion.

EricClaeys, 2022-04-24:
@gktozer   Glad you like Allsky and thanks for the complement.

The WebUI uses the images from the `allsky/images` folder.  The Allsky Website uses _copies_ of the timelapse, keogram, and startrails images, since the website often runs on a different server.  That means those images have to get copied to the Website.
Check that you're copying the files - make sure `UPLOAD_KEOGRAM`, `UPLOAD_VIDEO`, and `UPLOAD_STARTRAILS` are `true` in config/config.sh.

You also need to tell Allsky _where_ to copy the files to.  I assume your Website is on your Pi, so in config/ftp-settings.sh, make sure `PROTOCOL="local"` and set the *_DIR variables according to the instructions in the file.

If those settings are already `true`, then please follow the instructions on the Reporting Issues Wiki page and I'll take a look at the log file.

EricClaeys, 2022-05-12:
@gktozer    Any update?

EricClaeys, 2022-05-27:
Closing Issue - no response from user in over a month.

closed 2022-05-27 by EricClaeys