AllskyTeam/allsky #1179: Camera shows only "Allsky Software is starting up"

gabrielkumar, 2022-04-27:
Hi there,

I have a problem with setting up the camera.
On the Allsky Web-Administration Panel there is no live-view of the camera, it only shows "Allksy software is starting up" and it doesn´t change at any time.

One only time i got a picture, but then after a reboot never again.
I´m using Pi2, Pi OS Buster and the PI HQ Camera.
Already installed libcamera, camera in settings is also enabled.

Do you have any ideas why it is not working? I tried everything and nothing works :(

Thank you!

gabrielkumar, 2022-04-28:
> 

Thank you for your answer.

The log says:

`
Apr 28 07:05:46 allsky allsky.sh[525]: ERROR: the system should be configured for the legacy camera stack
Apr 28 07:05:46 allsky allsky.sh[525]:  >>> Unable to take picture, return code=65280, r=255
Apr 28 07:05:46 allsky allsky.sh[525]: xxxx Got signal 255 in capture_RPiHQ.cpp
Apr 28 07:05:46 allsky allsky.sh[525]: STARTING EXPOSURE at: 2022-04-28 07:05:46   @ 10.0 sec
`

Do you know whats wrong?

EricClaeys, 2022-05-12:
Closing Issue.  This isn't an Allsky issue - it's a non-standard camera configuration issue.  The next version of Allsky will catch these issues before even trying to start the "capture" program.

closed 2022-05-12 by EricClaeys

ettoreribaudo, 2024-02-10:
Hello boys my name is Ettore and my home in in Italy, i have a same problem, in local is ok but in web remote i have a message AllSky is starting up.
http://www.allskymasseris.it/allsky/

EricClaeys, 2024-02-10:
I see this when I visit your Website:

[image]

ettoreribaudo, 2024-02-11:
> I see this when I visit your Website:
>
> image.png (view on web)

Yes i try my all Sky.