AllskyTeam/allsky #1180: Not sure what's going on - no live view, photos have words on it

pjdonovan84, 2022-04-28:
I have a raspberry pi zero 1.1, bullseye, raspbery pi HQ camera.  I determined it was deleting bad photos after looking at the logs, but after stopping it from deleting photos I don't know what setting isn't working.  here's the tail

pi@Zerosky:~ $ tail /var/log/allsky.log
Apr 28 02:32:33 Zerosky allsky.sh[1350]: STARTING EXPOSURE at: 2022-04-28 02:32:33   @ 1.0 sec
Apr 28 02:32:34 Zerosky allsky.sh[1350]:   > Capture command: raspistill --output '/home/pi/allsky/tmp/image-20220428023233.jpg' --thumb none --burst -st --timeout 1 --nopreview --mode 3 --exposure off --shutter 1000000 --analoggain 4 --exif IFD0.Artist=li_1000000_4 --awb off --awbgains 2.8,2 --vflip --saturation 1 --brightness 0 --quality 95
Apr 28 02:33:04 Zerosky allsky.sh[1350]:   > Saving NIGHT image 'image-20220428023233.jpg'
Apr 28 02:33:05 Zerosky allsky.sh[1350]: Sleeping 1.0 seconds...
Apr 28 02:33:06 Zerosky allsky.sh[1350]: STARTING EXPOSURE at: 2022-04-28 02:33:06   @ 1.0 sec
Apr 28 02:33:07 Zerosky allsky.sh[1350]:   > Capture command: raspistill --output '/home/pi/allsky/tmp/image-20220428023306.jpg' --thumb none --burst -st --timeout 1 --nopreview --mode 3 --exposure off --shutter 1000000 --analoggain 4 --exif IFD0.Artist=li_1000000_4 --awb off --awbgains 2.8,2 --vflip --saturation 1 --brightness 0 --quality 95
Apr 28 02:33:36 Zerosky allsky.sh[1350]:   > Saving NIGHT image 'image-20220428023306.jpg'
Apr 28 02:33:37 Zerosky allsky.sh[1350]: Sleeping 1.0 seconds...
Apr 28 02:33:39 Zerosky allsky.sh[1350]: STARTING EXPOSURE at: 2022-04-28 02:33:39   @ 1.0 sec
Apr 28 02:33:41 Zerosky allsky.sh[1350]:   > Capture command: raspistill --output '/home/pi/allsky/tmp/image-20220428023339.jpg' --thumb none --burst -st --timeout 1 --nopreview --mode 3 --exposure off --shutter 1000000 --analoggain 4 --exif IFD0.Artist=li_1000000_4 --awb off --awbgains 2.8,2 --vflip --saturation 1 --brightness 0 --quality 95

[image]

I was hoping to make sure I could get a good shot or two before putting it all together in my 3d case.  Am I overlooking something?

EricClaeys, 2022-04-28:
@pjdonovan84   Hi, did you look at your settings in the WebUI?  At 2:30 AM it's dark out so a 1 second manual (?) shot at gain of 4 is going to give you pure black, which is what you got.  I suggest turning on auto exposure and auto gain, and setting a "Target Mean" of 0.5 during the day and 0.2 at night, and then adjust accordingly.  Also suggest setting `REMOVE_BAD_IMAGES="true"` once you have good images so your timelapse don't get "polluted" with an occasional very dark or very bright image.
Not to sound snotty, but did you make sure the lens cap is off?  I've ruined many pictures like that...

The `Overlay` section of the WebUI determines what text is added to the pictures, in what color, and where.  The RPiHQ camera has a high resolution so you'll need to set the "Text Y" to 50 - 100 (play with it) to get the text to not get cropped off the top of the picture.  What you have displayed is a personal preference so there's no right or wrong, but most people add the date and exposure; some add a lot more.

Going forward, when you have a question would you please post it to the Discussions link above?  I'm trying to keep the Issues list to actual problems with the software.  I realize it's not always obvious if something is a software problem or a question though.  Thanks.  Please post when you have it working.  If you have the problem I'm sure others have as well, and I like to update the Wiki with info like this.

pjdonovan84, 2022-04-28:
Not to sound snotty, but did you make sure the lens cap is off? I've ruined many pictures like that...
As someone legally required to ask people if they have rebooted their computer everyday, I didn't take it as snotty at all! No lens, but at least I know the camera is working well enough to put it together.  I've been able to move the text to where I can read it (Y was 200), so I know the software is working, this will just take a bit of time to get the right settings.  

This is SUCH a cool project - thank you for your hard work.  This will be my first attempt really getting into photography, so I have a lot to learn about the settings.  Also my first time using Github - so my apologies for posting here vs the discussion.  

I'll circle back as soon as I get the right settings, but you can close/move this ticket as I've copied your reply and I don't want to clutter the software part of your github!  If I need to move it or delete it myself, please let me know!

EricClaeys, 2022-04-28:
@pjdonovan84  Good luck.  Once you get the right settings I think you'll enjoy Allsky.  Auto mode should make it much easier.

The Wiki has a lot of good info so I suggest becoming familiar with it.  And if you ever have any suggestions on how to make things easier on newbie's like yourself, please let me know.  I'm pretty technical so sometimes forget most people aren't and I could use reminders.

To close an Issue, there should be a "Close with comment" button below next to the green "Comment" button.  As far as I can tell there's no way to "close" a Discussion though.  When you have your settings done, please post a Discussion.  There's a subgroup to let others know what your setup is - that might be a good place.
Eric

closed 2022-04-28 by EricClaeys