AllskyTeam/allsky #1182: AutoExposure doesn't seem to work

zapbest, 2022-04-29:
After a long time I resumed the allsky project and I am using a raspberry pi 4 with RPiHQ connected. 

In the past everything worked but after updating everything (Raspbian 10 buster and AllSky v2022.03.01) if I activate autoexposure for both day and night it doesn't work and always sets the value that I insert in "Manual Exposure". 

I followed all the guides and tried to do a clean fresh install. I tried several things but during the day I have completely white images and at night completely black regardless of the value imposed on "Mean Target".

I also set CAPTURE_EXTRA_PARAMETERS = "-mode-mean -autoexposure -daymean 0.5 -nightmean 0.2" but no change.

Can you tell me what I could watch?

holligram, 2022-04-29:
I'm having the same issue with AutoExposure. I'm using the RPiHQ camera. Setting debug to 1 shows the raspistill capture command:

> Capture command: raspistill --output '/home/pi/allsky/tmp/image-20220429150850.jpg' --thumb none --burst -st --timeout 1 --nopreview --mode 3 --exposure off --shutter 500 --analoggain 1 --exif IFD0.Artist=li_500_1 --awb off --awbgains 2.5,2 --saturation 50 --brightness 50 --quality 95
 > Saving DAY image 'image-20220429150850.jpg'

which produces a black image. The Camera Settings (Show advanced options), Daytime Settings show Auto-Exposure set to Yes but the generated command line has --exposure off. The Saving DAY line shows the systems is detecting daytime mode.

If I run raspistill manually from the command line with the same parameters but removing the exposure, shutter and analoggain the images look great:

raspistill --output '/home/pi/allsky/tmp/testimage.jpg' --thumb none --burst -st --timeout 1 --nopreview --mode 3 --exif IFD0.Artist=li_27_1 --awb off --awbgains 2.8,2 --vflip --saturation 50 --brightness 50 --quality 95

The same good result if I remove shutter and analoggain and set exposure explicitly to auto: 

raspistill --output '/home/pi/allsky/tmp/testimage.jpg' --thumb none --burst -st --timeout 1 --nopreview --mode 3 --exposure auto --exif IFD0.Artist=li_27_1 --awb off --awbgains 2.8,2 --vflip --saturation 50 --brightness 50 --quality 95

Is there something I'm missing. I tried this on both buster and bullseye thinking that may be the underlying cause but both fail.

EricClaeys, 2022-04-30:
@zapbest, @holligram,
Please follow the instructions on the Wiki page for Reporting Issues and then post your log files.  I believe other people are using that release successfully so it may be something in your settings. 
Before doing what the Wiki says, clear out the `CAPTURE_EXTRA_PARAMETERS` setting.  The Target Mean settings supercedes `CAPTURE_EXTRA_PARAMETERS`.

holligram, 2022-04-30:
@EricClaeys
Sorry I may have highjacked @zapbest's issue but I think they are the same.
Followed instruction on Wiki. Also returned camera settings to default before setting level 4 debug.
Out of box experience is resulting images are completely white (over exposed). Had to set high and low bad image removal to 0 to see images else the live view just shows 'software is starting up' in green.

[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8597429/allsky.log.txt)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/8597430/config.sh.txt)

commit 3c3a42f7272a2a608e7444fe86ef1047c14e7390
Merge: 83615f0 d7226e6
Author: EricClaeys <83164203+EricClaeys@users.noreply.github.com>
Date:   Thu Apr 14 19:34:54 2022 -0600

    Merge pull request #1160 from thomasjacquin/README.md-scripts-bug-fix-1

    README.md scripts bug fix

commit d7226e6949d26a0617fc8ac2ac843df6dd2b505f

holligram, 2022-04-30:
Through some experimentation I think I have solved this.

The default camera settings for Manual Exposure and Brightness are 0.5 and 100 respectively. 
Auto Exposure seems to use the Manual Exposure as a starting point in some type of descent algorithm. 
Setting Manual Exposure to  1000 (1 sec) is then a better starting point but this still descends to 1us.
Setting the Brightness from the default of 100 to 50 and Manual Exposure from the default of 0.5 to 1000 allows the descent to a stable value and perfect Auto Exposed images.

From my allsky.log posted above the shutter value starts at 500 (0.5) where in the good exposure log segment below the shutter is ~800000.

Apr 30 12:07:39 pisky allsky.sh[25294]: DEBUG: saveImage.sh DAY /home/pi/allsky/tmp/image-20220430120733.jpg EXPOSURE_US=908218 BRIGHTNESS=50 MEAN=0.515083 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.500000 WBB=2 GAIN=0 GAINDB=1.00 BIN=1 FLIP=0 BIT_DEPTH=8
Apr 30 12:07:40 pisky allsky.sh[25294]: STARTING EXPOSURE at: 2022-04-30 12:07:40   @ 809.13 ms (0.81 sec)
Apr 30 12:07:40 pisky allsky.sh[25294]:  > Kill command: pgrep 'raspistill' | xargs kill -9 2> /dev/null
Apr 30 12:07:41 pisky allsky.sh[25294]:   > Capture command: raspistill --output '/home/pi/allsky/tmp/image-20220430120740.jpg' --thumb none --burst -st --timeout 1 --nopreview --mode 3 --exposure off --shutter 809130 --analoggain 1 --exif IFD0.Artist=li_809130_1 --awb off --awbgains 2.5,2 --saturation 50 --brightness 50 --quality 95
Apr 30 12:07:45 pisky allsky.sh[25294]:   > exposure: 0.809 sec, mean: 0.460, mean_value: 0.500, diff: 0.040
Apr 30 12:07:45 pisky allsky.sh[25294]:   > MeanCnt: 3, historySize: 3
Apr 30 12:07:45 pisky allsky.sh[25294]:   > i=1: mean_history[1]=0.487 exp_history[1]=-10
Apr 30 12:07:45 pisky allsky.sh[25294]:   > i=2: mean_history[2]=0.515 exp_history[2]=-5
Apr 30 12:07:45 pisky allsky.sh[25294]:   > i=3: mean_history[0]=0.460 exp_history[0]=-11
Apr 30 12:07:45 pisky allsky.sh[25294]:   > mean_forecast: 0.405, values: 9, mean_diff: 0.043
Apr 30 12:07:45 pisky allsky.sh[25294]:   > ExposureChange: 9 (3)
Apr 30 12:07:45 pisky allsky.sh[25294]:   > Mean: 0.457, diff: 0.043, Exposure level:-2 (9), Exposure time:962.22 ms (0.96 sec), analoggain:1.00
Apr 30 12:07:45 pisky allsky.sh[25294]:   > Got exposure: 1.0 sec, shutter: 962.2 ms, quickstart: 6, mean=0.460
Apr 30 12:07:46 pisky allsky.sh[25294]:   > Saving DAY image 'image-20220430120740.jpg'
Apr 30 12:07:47 pisky allsky.sh[25294]: Sleeping 1.0 seconds...

My conclusion is that the default Camera Setting values, at least for the RPiHQ camera, are too low to allow it to work.
The setting of Manual Exposure of 1000 and Brightness of 50, for my system anyway, allow the algorithm to stabilize to a good exposure.

marco050969, 2022-05-01:
Hello

I have the same problems here.
Autogain and autoexposure only work reliably at night if I enter the same values in both settings, otherwise the automatic control does not work.

marco050969, 2022-05-01:
[allsky.log](https://github.com/thomasjacquin/allsky/files/8599456/allsky.log)

EricClaeys, 2022-05-03:
@marco050969   Is there something in the log file you wanted me to look for?  It looks fine to me.  It took less than a minute to get to the right daytime exposure when it started at 1 second.  Once it got to the correct exposure it stayed there.  If you set the daytime exposure to `.5` it will get to the correct exposure in a few seconds next time.

All the pictures in the log file are daytime, even though it was 8 pm.  Is it still light then?

> Autogain and autoexposure only work reliably at night if I enter the same values in both settings, ...

Do you mean if you set both nighttime Auto Gain and nighttime Auto Exposure to "Yes"?

EricClaeys, 2022-05-03:
@holligram   It's hard to come up with default exposure/gain settings that work for everyone.  The software used (Buster/raspistill versus Bullseye/libcamera-still) seems to make a difference, as well as the lens and the person's location.
We're still tweaking our auto exposure/gain algorithm so it should get better over time.
I'm glad you found settings that work well for you.

EricClaeys, 2022-05-03:
@zapbest, any update?

marco050969, 2022-05-03:
Can i send you Here a Screenshot that yoo see, what i mean? When i set day exposure on 0.5 , so autoexposure don’t work and stay still at o.5. At 8pm Here is still light.

Cheers

Marco

Von meinem iPhone gesendet

> Am 03.05.2022 um 07:14 schrieb EricClaeys ***@***.***>:
> 
> @marco050969 Is there something in the log file you wanted me to look for? It looks fine to me. It took less than a minute to get to the right daytime exposure when it started at 1 second. Once it got to the correct exposure it stayed there. If you set the daytime exposure to .5 it will get to the correct exposure in a few seconds next time.
> 
> All the pictures in the log file are daytime, even though it was 8 pm. Is it still light then?
> 
> Autogain and autoexposure only work reliably at night if I enter the same values in both settings, ...
> 
> Do you mean if you set both nighttime Auto Gain and nighttime Auto Exposure to "Yes"?

holligram, 2022-05-03:
I think there is some confusion about the values entered in camera settings. The entry needs to be in milliseconds and not seconds as the current default of 0.5 implies. This default value represents 5 microseconds and will never work.
I had to enter 1s as 1000 in the camera settings manual exposure and @marco050969 said he had to do the same.
The suggestion to him to try 0.5s would need to be entered as 500 in the camera settings manual exposure field.
The default (0.5) 5 microseconds out-the-box value will not work with Bullseye's and the RPiHQ. The manual exposure value sets a max auto exposure value as the algorithm only descends from there. A better name for it would be "manual/max auto exposure"

zapbest, 2022-05-03:
I have done several tests but I don't understand how you reason the automatic exposure.

To get just decent shots with automatic exposure and gain I have to set the same value as "Max Auto-Exposure" and "Manual Exposure" but so it is as if I set it to manual and it does not adapt to the sky conditions.

For the exposures of the day I had to set "Brightness" to 50 otherwise all the images are white.

I am attaching the log files and configurations that I am using now but the night results are very poor, it is hard to see the stars.
---
commit 3c3a42f7272a2a608e7444fe86ef1047c14e7390
Merge: 83615f0 d7226e6
Author: EricClaeys <83164203+EricClaeys@users.noreply.github.com>
Date:   Thu Apr 14 19:34:54 2022 -0600

    Merge pull request #1160 from thomasjacquin/README.md-scripts-bug-fix-1
    
    README.md scripts bug fix

commit d7226e6949d26a0617fc8ac2ac843df6dd2b505f

[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8609433/allsky.log.txt)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/8609434/config.sh.txt)

EricClaeys, 2022-05-03:
@marco050969    You can attach a screenshot.  You'll first need to save it to a file, then simply drag it to the rectangle at the bottom of your reply where it says "Attach files by ...".

Your daytime target mean is 0.5, which tells the exposure algorithm to change the exposure and gain to get pictures that have a mean brightness of 0.5 during the day.  In your log file from a couple days ago the images were around 0.5 plus or minus a little.  The exposure algorithm looks at a circle in the center of the image to determine the exposure, so it's possible that's not working for you, for example, if your images are offset.

Danke

EricClaeys, 2022-05-03:
@holligram, @zapbest    I apologize.  We distributed a `camera_options_RPiHQ.json` file which determines what options to display that was newer than the program, so there are some options you can set that aren't actually supported yet, like "Max Exposure".  That was my fault.

As you discovered, the **Manual Exposure** setting is really the **maximum** exposure, so I suggest setting it accordingly.  The next version of Allsky will have a separate **Maximum Auto Exposure** setting that is the maximum exposure when using Auto Exposure, and the **Manual Exposure** will be the _starting point_ when using auto exposure.  Ditto for gain.

Despite the confusion you should be able to get well exposed images day and night.

@zapbest   Disregard the "Max Auto Exposure" setting and set the **Manual Exposure** to be the maximum, and set **Auto Exposure** on and set an appropriate **Target Mean** value.  Also, clear out the `CAPTURE_EXTRA_PARAMETERS` setting in config/config.sh.

I am considering getting rid of the **Brightness** setting since it isn't what most people think.  It simply makes all pixels brighter, which turns black into gray.

zapbest, 2022-05-03:
@EricClaeys thank you very much! Tonight I try the new settings.
As for Day Brightness and Night Brightness how much do you recommend setting them?

EricClaeys, 2022-05-03:
@zapbest   You are on Buster so I recommend using the default of 50.  It ranges from 0 to 100.  0 is pure black and 100 is pure white.
Once you get nice looking pictures you can play around with Brightness and see if it helps.

zapbest, 2022-05-03:
I think that the Gain also reasons such as exposure and the set value is the maximum value it can reach automatically. Correct?

EricClaeys, 2022-05-03:
@zapbest    Correct.  The **Manual Exposure** and **Gain** numbers are actually the maximum values.

If this resolves your problem would you please close the Issue?

zapbest, 2022-05-03:
yes, everything perfect.

closed 2022-05-03 by zapbest

holligram, 2022-05-04:
> I am considering getting rid of the **Brightness** setting since it isn't what most people think. It simply makes all pixels brighter, which turns black into gray.

Just FYI, I could never get it to work by changing manual exposure alone to say 100 or 500. I had to also reduce brightness from the default of 100 to 50. Without doing that, the exposure algorithm would continue to descend to 1us without stabilizing the mean.

marco050969, 2022-10-11:
Here my settings so that the Automatik runs.
Im confused…

Von meinem iPhone gesendet

> Am 03.05.2022 um 23:38 schrieb EricClaeys ***@***.***>:
> 
> Correct. The Manual Exposure and Gain numbers are actually the maximum values.
> 
> If this resolves your problem would you please close the Issue?
>

marco050969, 2022-10-12:
> @marco050969 Marco,
> When replying via email, GitHub removes any attachments so if you attached your settings I can't see them.  You'l need to log into GitHub to attach them.
> What exactly are you confused about?

Hi Eric,It’s all ok here.😊Von meinem iPhone gesendet