AllskyTeam/allsky #1186: variables.sh has wrong $WEBSITE_DIR

HPWever, 2022-05-03:
When installing the latest version of Allsky (GREAT PROGRAM; I AM A LIFELONG FAN) the file variables.sh has a wrong website_dir when using the allsky-website as your default local website.
This causes the web inteface to produce the error:
"WARNING: data.json is " + duration.days() + " days old. Check Allsky log file if 'postData.sh' has been running successfully at the end of nighttime."

Default in variables.sh is:
WEBSITE_DIR=${WEBSITE_DIR:-${PORTAL_DIR}/allsky}

In my opinion it should be:
WEBSITE_DIR=${WEBSITE_DIR:-${PORTAL_DIR}/allsky-website}

Agreed?

EricClaeys, 2022-05-03:
@HPWever. It's correct as is.  The full path to the website directory is `/var/www/html/allsky`.  There is no `allsky-website` directory. 
Did you check in the log file to ensure `postata.sh` has been copying `data.json` to the website?
You can also run it manually to make sure it's working correctly:
~~~
cd ~/allsky/scripts
./postData.sh
~~~

HPWever, 2022-05-03:
@EricClaeys
For me there IS a directory /var/www/html/allsky-website
After my correction everything runs perfectly okay (before it did not).
How did I get the allsky-website directory? I just installed it from: https://github.com/thomasjacquin/allsky-website
Maybe you installed it differently or customized....?

EricClaeys, 2022-05-03:
@HPWever   Ah, I see. The "allsky" repository is the only one that's installed by entering the URL to GitHub.  The WebUI is installed by executing
~~~
cd ~/allsky
sudo gui/install.sh
~~~
and the website by executing
~~~
cd ~/allsky
website/install.sh
~~~

To aid in any future debugging, you may want to rename `allsky-website` to `allsky`, then run the `website/install.sh`.  It should recognize that you have an older version of the website, and restore your images/videos and configuration files.
And then don't forget to restore `variables.sh`.

If you installed the website this way, I'm sure others have or will, so I'll need to think of a way to detect that and issue a warning.

EricClaeys, 2022-05-03:
@HPWever   If this resolves the problem, would you please close this Issue?
Thanks - Eric

HPWever, 2022-05-05:
thanks for looking into it Eric (and adding the check feature in future versions)

closed 2022-05-05 by HPWever