AllskyTeam/allsky #1188: SSL - how to https

rodrigoluk, 2022-05-04:
I hope you can guess from the title the issue I'm having. So, I'm trying to install and get SSL cert to work on the web portal and the allsky website.  I've used the following procedure to install certs: https://docs.raspap.com/ssl-quick/ 
I've got a domain in godaddy with an A record pointing at my public IP, which my router ports forwards to my PI4 running AllSky 0.8.3.3
My lighttpd.conf has this:
```
$SERVER["socket"] == ":443" {
server.modules += ("mod_openssl")
ssl.engine = "enable"
ssl.pemfile = "/etc/lighttpd/ssl/www.lopezserrano.me.pem"
ssl.ca-file = "/home/pi/.local/share/mkcert/rootCA.pem"
server.name = "www.lopezserrano.me"
server.document-root = "/var/www/html"
}
```
Any ideas, please?

EricClaeys, 2022-05-04:
@linuxkidd   Michael,  Any ideas?

@rodrigoluk   We don't support certificates in Allsky, but when you get it working I'd like to know what was needed so I can add it to the Wiki.

rodrigoluk, 2022-05-04:
Thanks :-) I'll close and reopen if I figure it out.

rodrigoluk, 2022-05-06:
Ignore the previous instructions. My server is now running on https://lopezserrano.me/allsky after following the instructions here:
https://www.itzgeek.com/how-tos/linux/how-to-configure-lets-encrypt-ssl-in-lighttpd-server.html
Note that I had to change the permissions on /etc/letsencrypt/live in order to be able to read and write and complete the concatenation step. I then set the permissions to 0640.

I hope this helps.

rodrigoluk, 2022-05-06:
> @rodrigoluk <https://github.com/rodrigoluk> Thanks. Once I add the info
> to the Wiki I'll re-close this Issue.

No problem. Glad to be able to help.

Kind regards,

R😊

EricClaeys, 2022-06-29:
@rodrigoluk   I added a page for [configuring SSL](https://github.com/thomasjacquin/allsky/wiki/Using-SSL) to the Wiki.  Would you please take a look and provide feedback?  I appreciate it.
Eric

rodrigoluk, 2022-06-30:
Hi  Eric,

Looks good to me 😊
Many thanks for the update.

R.

EricClaeys, 2022-06-30:
@rodrigoluk   Thanks for looking at it.
Closing this issue since it's documented in the Wiki.

closed 2022-06-30 by EricClaeys