AllskyTeam/allsky #1196: Trying to lessen the load on micro SD- which approach makes sense?

ScottTheHippy, 2022-05-08:
Howdy all,

I've had to go back to the drawing board once already due to a microSD card giving up the ghost.  When that happened, I simply grabbed another and repeated everything.  Only afterwards did I think to consider it can and likely will happen again.  So I added a USB of considerable speed and capacity, and am wondering if I'm being dumb.  The ask:  Should I just image allsky on to the new USB and yank the microSD or ignore it?   Or should I split the load, where heavy read/write happens on USB and boot on microSD?  I didn't see any prior discussions on this, and hope I'm not being a blind nuisance.  Cheers, and thanks.

ScottTheHippy, 2022-05-08:
As an update, I did see this: https://github.com/thomasjacquin/allsky/wiki/Miscellaneous-Tips which has a section on moving the /tmp to a memory space, but I'm not sure it's enough of what I'm after.  I'm still wondering about the initial question- would this be best handled by moving off microSD to USB exclusively, or is a split load better?   Cheers and thanks again.

EricClaeys, 2022-05-08:
@ScottTheHippy,
This question might be better for a Raspberry Pi forum.  They would have more experience with how often SD cards go bad.

If you make a memory filesystem per the Wiki, then Allsky only performs a few writes to the SD card per image, plus writes to the log file (which can be significantly decreased by lowering the **Debug Level**).  Putting the ~/allsky directory on a USB disk is pretty easy and would eliminate most of the Allsky writes to SD card except the log file and the Allsky Website if you have it installed.  In the future the Website will move to ~/allsky so those writes to SD card would also be eliminated.

I don't think putting Allsky on a USB disk will do any harm so if it makes you more comfortable, go for it.  If you do, and your willing to write a paragraph or two on how you did it, I can include it in the Wiki so others can benefit from your experience.

In the future, when you have a question or request for new feature, would you please create a Discussion item for it rather than an Issue?  I'm trying to keep the Issue list dedicated to actual problems.  Thanks much.

If this answers your question, would you please close the Issue?  Thanks.

ScottTheHippy, 2022-05-08:
Howdy Eric,

Ahh, and ahh again.  :)  Yes- this isn't an issue in the way you mean.  

Cheeers!

closed 2022-05-12 by EricClaeys