AllskyTeam/allsky #1197: Thumbnail creation failed

MatiasLM, 2022-05-10:
Hi,

I recently deployed the allsky software (including portal and website) on a Raspberry Pi 4B+ 4GB running Raspberry Pi Os 11.  After first night ended, timelapse, keograms and startrails were generated and uploaded to website, however, following errors related to thumbnail are showed.

**Timelapse**
_Failed to make thumbnail for ./allsky-20220508.mp4 after 2 attempts.
Last command: ffmpeg -loglevel warning -ss 00:00:00 -i './allsky-20220508.mp4' -filter:v scale='100:-1' -frames:v 1 './thumbnails/allsky-20220508.jpg' 2>&1
Output from command: [swscaler @ 0xd86150] deprecated pixel format used, make sure you did set range correctly_
[image]

**Keograms** 
_Unable to create thumbnail for './keogram-20220508.jpg': imagejpeg(./thumbnails/keogram-20220508.jpg): failed to open stream: Permission denied_
[image]

**Startrails** 
_Unable to create thumbnail for './startrails-20220508.jpg': imagejpeg(./thumbnails/startrails-20220508.jpg): failed to open stream: Permission denied_
[image]

After that I ran endOfNight.sh and uploadForDay.sh and get the same result.

A little help with this would be great. 

Best regards.

Matías

EricClaeys, 2022-05-10:
@MatiasLM. How did you install the Website?

Would you please do the following:
~~~
cd /var/www/html/allsky
ls -ld */thumbnails 
~~~

MatiasLM, 2022-05-10:
Hi @EricClaeys,

I installed website following these links:
https://github.com/thomasjacquin/allsky#on-the-pi
https://github.com/thomasjacquin/allsky/wiki/allsky-website-Settings

[image]

Thank you

EricClaeys, 2022-05-10:
@MatiasLM    Is allsky installed in `/home/pi/allsky`?
Did you install the website by running `website/install.sh` ?  If so, it should have created the directories with the correct owner and permissions.

To fix the permissions, run the following:
~~~shell
cd /var/www/html/allsky
sudo chown -R pi:www-data .
sudo find ./ -type f -exec chmod 644 {} \;
sudo find ./ -type d -exec chmod 775 {} \;
~~~

EricClaeys, 2022-05-10:
@MatiasLM    For some reason, the login on your Pi is "allsky" and not "pi".  Are you on Buster or Bullseye?

AndreHellmuth, 2022-05-10:
Hi Eric , i have the same problem. I don't know, what the PW is for "Allsky" directory. I only have the PW for "Pi".

MatiasLM, 2022-05-10:
@EricClaeys 

> @MatiasLM For some reason, the login on your Pi is "allsky" and not "pi". Are you on Buster or Bullseye?

When mounted Raspberry Pi OS image in SD card I specified the Hostname "allsky" instead of default "pi". I didn't found any restriction in the Readme and Wiki, so in order to better organize several RPI that I have at home I chose this host name. Sadly not even suspected this could cause some troubles.

> @MatiasLM Is allsky installed in `/home/pi/allsky`? 
/home/allsky/allsky

> Did you install the website by running `website/install.sh` ? 
I confirm yes, I looked for in the command history.

MatiasLM, 2022-05-10:
I execute the following commands

> ```shell
> cd /var/www/html/allsky
> sudo find ./ -type f -exec chmod 644 {} \;
> sudo find ./ -type d -exec chmod 775 {} \;
> ```

I didnt execute chown due to pi user does not exist in my Raspberry.

To check, I generated and upluad keogram, and get the same error in web.

I have seen the same error in @thomasjacquin web only for videos: https://www.thomasjacquin.com/allsky/videos/

EricClaeys, 2022-05-11:
@MatiasLM 
The `hostname` is the name of the Pi hardware.  If you have more than one Pi you should give them different names otherwise there can be network problems.
On each Pi there is a login that the Allsky software is installed in.  The software assumes the name of the login is `pi`Pi.  On your system, it is `allsky`.  We heard rumors that the Raspberry Pi team was going to remove the `pi` login, but didn't know the specifics.

We will need to make changes to Allsky to not assume the `pi` login exists.

Are you running Buster or Bullseye?
When did you install the operating system?

EricClaeys, 2022-05-11:
@MatiasLM.  Until we can update Allsky, please run the following to fix your permissions:
~~~shell
cd /var/www/html/allsky
sudo chown -R allsky:www-data .
~~~
The key is that the `www-data` group (that runs the web server)  needs write access to the directories so it can create the thumbnails.

MatiasLM, 2022-05-11:
@EricClaeys 

> Are you running Buster or Bullseye? 
Sorry, I don't know what they are. I only installed Raspberry Pi OS (32-bit) using Raspberry Pi Imager v1.7.2. And, I  as previously mentioned, I change the default username, see screenshots below.

[image]

[image]

[image]

> When did you install the operating system?

Last Satuday, 07/05/2022.

MatiasLM, 2022-05-11:
> @MatiasLM. Until we can update Allsky, please run the following to fix your permissions:
> 
> ```shell
> cd /var/www/html/allsky
> sudo chown -R allsky:www-data .
> ```
> 
> The key is that the `www-data` group (that runs the web server) needs write access to the directories so it can create the thumbnails.

This solve the problem. Thancs @EricClaeys

EricClaeys, 2022-05-11:
@MatiasLM     Is your Pi running Buster or Bullseye?

EricClaeys, 2022-05-11:
@MatiasLM    I'm curious why you changed the default username from "pi" to "allsky"?

MatiasLM, 2022-05-11:
> @MatiasLM I'm curious why you changed the default username from "pi" to "allsky"?

To have the same as hostname. Bad choice...

EricClaeys, 2022-05-11:
@MatiasLM    Well, your choice forced me to submit a couple changes  to determine what the user name is, rather than assume it's "pi".  And I thank you for that, because if/when the Raspberry Pi team decides to get rid of the "pi" login, we'll be ready.

If your original problem is fixed, would you please close this issue?  Thanks much.

AndreHellmuth, 2022-05-11:
@EricClaeys i have the same problems as Matias. (See his first post)

[image]

I do the following steps:

cd /var/www/html/allsky
ls -ld */thumbnails 

The result:

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue May 10 21:02:09 2022
pi@allsky:~ $ cd /var/www/html/allsky
ls -ld */thumbnails
drwxrwxr-x 2 pi www-data 4096 15. Apr 21:17 keograms/thumbnails
drwxrwxr-x 2 pi www-data 4096 15. Apr 21:17 startrails/thumbnails
drwxrwxr-x 2 pi www-data 4096 15. Apr 21:17 videos/thumbnails
pi@allsky:/var/www/html/allsky $ ^C

Then i do step 2:
My directories are:  /home/pi/allsky   website/install.sh

Then i type:

cd /var/www/html/allsky
sudo find ./ -type f -exec chmod 644 {} \;
sudo find ./ -type d -exec chmod 775 {} \;

The problem is not fixed. 
PI OS is Bullseye

Greetings: André

EricClaeys, 2022-05-11:
@AndreHellmuth,   André, your symptoms are the same as Matias, but the underlying problem is different.

Your directory permissions are correct.  Please install the latest Website,:
~~~shell
cd ~/allsky
website/install.sh
~~~

Then:
~~~shell
cd /var/www/html/allsky
mv functions.php functions-ORIGINAL.php
~~~

Then copy the attached file to `/var/www/html/allsky`, removing the trailing ".txt".
See if your website will create timelapse and/or startrails thumbnails.  If not, then press F12 and click on `>>` then `Console` and let me know what messages are there.  I'm hoping that will explain why it can't produce thumbnails.

[functions.php.txt](https://github.com/thomasjacquin/allsky/files/8669863/functions.php.txt)

AndreHellmuth, 2022-05-12:
@EricClaeys, Eric, I do all steps and this is my result this morning. 
[image]

EricClaeys, 2022-05-12:
@AndreHellmuth   What messages do you see when you press F12, then `>>` then `Console`?

AndreHellmuth, 2022-05-12:
@EricClaeys , no Message

[image]

EricClaeys, 2022-05-13:
@AndreHellmuth   You're running an older version of the Website, which doesn't log any messages.  Would you please upgrade to the latest website software?
It likely won't fix the problem, but will tell us what the problem is so we can fix it.

I suspect the server your website is on doesn't have the `ffmpeg` command that's used to create the thumbnails.

AndreHellmuth, 2022-05-13:
@EricClaeys, i upgrade the latest website. 
With first Allskycam Version I don't have Problems. 
This is my result with Version 0.8.3.3:

[image]

EricClaeys, 2022-05-17:
@AndreHellmuth   Thanks for pointing that out.  It's a bug and I just submitted a fix.
I'll probably take a few days to be uploaded, so if you want to fix it, change line 268 in `functions.php` to
~~~
$thumbnail = "../NoThumbnail.png";
~~~

EricClaeys, 2022-05-17:
@AndreHellmuth   Forgot to say, the server you're running the Allsky Website on doesn't have the `ffmpeg` command installed, so Allsky can't make the thumbnails.  You can ask your server provider if they can install it, but that will also require them to allow the `exec()` call to be executable, which is a security risk so they are unlikely to do it.
You'll need to wait until I implement the ability for the Pi to generate the thumbnails and upload them to the server.

BTW, Thomas just approved the fix, so you can grab the latest `functions.php` from [here](https://github.com/thomasjacquin/allsky-website.)

EricClaeys, 2022-05-17:
@MatiasLM   Hi, is your original problem resolved?  Is so, would you please close this Issue?
Thanks - Eric

closed 2022-05-17 by MatiasLM