AllskyTeam/allsky #1211: No timelapses being created since

AstroHighlander, 2022-05-16:
My timelapses haven't been created for some considerable time now (last successful one was the night of 20220419/20) and I, at last, had a look at what was happening when the timelapse script was being run.

Theres a bright yellow message appears saying 

[swscaler @ 0x230c320] deprecated pixel format used, make sure you did set range correctly

I've no idea what this means, or how to fix it.  Any help would be appreciated.

EricClaeys, 2022-05-27:
@AstroHighlander    Any update?

AstroHighlander, 2022-05-27:
Haven’t tried anything.  Just feeling it’s all way beyond me now.  Sorry.

Eric

EricClaeys, 2022-05-27:
@AstroHighlander.  I understand Linux and the Pi can be daunting for people not familiar with them.  I am slowly trying to make instructions easier for inexperienced people, bur need help determining what needs to change.

If you can follow the instructions in the "Reporting Issues" Wiki page and attach the log file, I will take a look.

Timelapse problems are almost always caused by one of two different things, which are both fairly straightforward to resolve.

AstroHighlander, 2022-05-27:
Thanks for the encouragement Eric.  I’ll have a look - Wiki is also too difficult for me & I avoid it like the plague - and see if I can do what’s needed.

To be honest though, I think I’m close to ‘meeting my match’

Eric

EricClaeys, 2022-05-27:
@AstroHighlander  I am more than willing to help so you can enjoy your camera and Allsky software.

I could use your help, though, to let me know exactly what is "too difficult" and any suggestions you have to make it easier for the many people like yourself who are not experienced computer people.  Can you give me some examples of difficult things in the Wiki?
Eric

AstroHighlander, 2022-05-30:
Do you have a link to the Wiki Eric?  I can't work out how to find it.

Thanks

Eric

EricClaeys, 2022-05-30:
@AstroHighlander   Click on the [Wiki](https://github.com/thomasjacquin/allsky/wiki) link near the top of any of the Allsky GitHub pages.

AmiSMB, 2022-06-15:
@AstroHighlander

I have just gone through the process of updating my Allsky camera which uses a Raspberry Pi 2 Zero W along with the Arducam IR-CUT IMX477 camera (Same as RPI-HQ but with automatic IR-CUT shutter) and found the following:-

**Make sure that your Pi is not starting in GUI (X11) mode to save as much RAM as possible.**

Use sudo raspi-config

And go into System Options->Boot/Auto login and make sure that Console is selected and then use tab to move to the OK button and press enter/return and that should make sure that the GUI does not start.

**The default swap file is only 100Mb and this needs to be increased to 1024Mb**
See [https://pimylifeup.com/raspberry-pi-swap-file/](url) for details on what commands you need to do this

I was then finding that if my settings were set to save images through the day then the timelapse would just take so long as it would hold up the image resizing processes so I looked into what I could do to fix this.

**Check the Video codec being used and change it**

Check that ffmpeg supports hardware encoding

ffmpeg | grep h264_omx

If you see a line with:-

V..... h264_omx             OpenMAX IL H.264 video encoder (codec h264)

Then set the VCODEC in config.sh to h264_omx and this will enable hardware encoding which most of the Raspberry Pi's support.

[image]

If you do not have that available then you might want to try the libx265 codec as this does not try and go over the limits like the libx264 codec does but does take longer than the hardware decoder.

libx265
[image]

EricClaeys, 2022-06-15:
@AstroHighlander Are you ok with closing this Issue?

@AmiSMB Other people use the GUI on their Pi Zero.  It might require a 2048K swap file.  Have you tried that?
The Wiki page on timelapse problems mentions using `libx265` codec, but it takes about 3 hours to run.

AstroHighlander, 2022-06-15:
I’m ok with you closing it Eric.  It’s just a bit too much for me to grasp.

Thanks very much

Eric

EricClaeys, 2022-06-16:
Closing with user agreement. Allsky and the Pi still require some technical knowledge that not all users have.  Although we have slowly been making Allsky less dependent on those skills, we still have a ways to go.

closed 2022-06-16 by EricClaeys