AllskyTeam/allsky #1212: Error on Longitude Lattitude Change

topazkuntadi, 2022-05-17:
Settings saved and service restarted/home/pi/allsky/scripts/postData.sh: line 6: /variables.sh: No such file or directory
/home/pi/allsky/scripts/postData.sh: line 7: /config.sh: No such file or directory
/home/pi/allsky/scripts/postData.sh: line 8: /ftp-settings.sh: No such file or directory
jq: error: Could not open file : No such file or directory
jq: error: Could not open file : No such file or directory
/home/pi/allsky/scripts/postData.sh: line 47: /data.json: Permission denied
ERROR posting twilight data: ***
postData.sh: WARNING: No local or remote website specified so 'data.json' not copied anywhere.
***.

my location is. in 7.75S -110.37E the error occured everytime i change the lat & longitude value. thanks.

EricClaeys, 2022-05-17:
@topazkuntadi   Please add this line to `~/allsky/scripts/postData.sh` as the 2nd line.  Then try again.  It should give you more messages, and will help me debug the issue.  This happened to another user once, and never again.
~~~
set -x
~~~

Thanks - Eric

topazkuntadi, 2022-05-17:
@EricClaeys , thanks for your reply. i follow your instruction and these are the messages :

+ source /variables.sh
/home/pi/allsky/scripts/postData.sh: line 6: /variables.sh: No such file or directory
+ source /config.sh
/home/pi/allsky/scripts/postData.sh: line 7: /config.sh: No such file or directory
+ source /ftp-settings.sh
/home/pi/allsky/scripts/postData.sh: line 8: /ftp-settings.sh: No such file or directory
++ basename /home/pi/allsky/scripts/postData.sh
+ ME=postData.sh
++ jq -r .angle ''
jq: error: Could not open file : No such file or directory
+ angle=
++ jq -r .latitude,.longitude ''
jq: error: Could not open file : No such file or directory
+ latlong=
++ date +%z
+ timezone=+0700
++ sunwait list rise angle
+ sunrise=17:34
+ sunrise_hhmm=17:34
++ sunwait list set angle
+ sunset=08:04
+ sunset_hhmm=08:04
+ '[' 17:34 = --:-- -o 08:04 = --:-- ']'
++ date +%Y-%m-%d
+ today=2022-05-18
+ FILE=data.json
+ OUTPUT_FILE=/data.json
/home/pi/allsky/scripts/postData.sh: line 47: /data.json: Permission denied
+ COPIED=false
+ typeset -i RETCODE=0
+ '[' -d '' ']'
+ '[' '' '!=' '' ']'
+ '[' false = false ']'
+ echo '***'
+ echo -e 'postData.sh: WARNING: No local or remote website specified so '\''data.json'\'' not copied anywhere.'
+ echo '***'
+ exit 1
ERROR posting twilight data: ***
postData.sh: WARNING: No local or remote website specified so 'data.json' not copied anywhere.
***.

topazkuntadi, 2022-05-18:
@EricClaeys I done fresh installation today. it seems the error don't repeat.

unfortunately the WebUI have some errors (not like yesterday version - error on latitude entry, WebUI work perfectly)
the WebUI don't show the captured image, and the editor tab is not working. i capture the SS and save it in my dropbox :

https://www.dropbox.com/sh/wdllmeqop9ssf0r/AACNBmKpic2r_MD_RxWl5dYRa?dl=0

Thanks a lot.

EricClaeys, 2022-05-18:
@topazkuntadi   Hi, please use the attached file which should fix the issue with error messages.  To use it:
~~~shell
cd ~/allsky/scripts
mv postData.sh postData.sh-ORIGINAL
# move the attached file to this directory, removing the ending ".txt" from the name
~~~
They try making a change again in the WebUI.

[postData.sh.txt](https://github.com/thomasjacquin/allsky/files/8713772/postData.sh.txt)

Did the WebUI show the image yesterday?  And did the Editor tab work?  If so, what's changed?
If not, let's work on the image problem first, although I suspect they are related and are due to a configuration mismatch.

When in the browser's F12 mode and the image doesn't appear, click on the "image.jpg?ts=xxxxx" entry - it should display the full URL it's looking for.  On my Pi, which is called `allsky`, it has `http://allsky/current/tmp/image.jpg?_ts=1652856201275`.  The "_ts" (time stamp) number will change every picture, so ignore that.

Execute the following:
~~~shell
grep IMG_DIR ~/allsky/config/config.sh
~~~
and let me know what it says.

Eric

topazkuntadi, 2022-05-18:
thanks for fixing the error message, i'll replace the postdata tonight.

and for the WebUI. it work yesterday on both my RPi 4B + 224MC. and RPi Zero 2W + NoIR. 
after RPi Zero get RETCODE=137 and try fixing it resize swapfile to 1024. the AllSky never start again. it just stuck on "AllSky is restarting", get frustated. So I do fresh installation this morning. and come error on the WebUI + script editor tab .

Pardon for my bad English. I already use AllSky and this RPi 4B for 3 years. this time I try to build RPi Zero as my eyes (cctv style) when i run my AP rigs on my rooftop. because my cctv (ezvis) died last week. Now i realized RPi Zero is not the best choice. should go for 3B+ / 4B instead.

yes it find : http://192.168.2.36/current/tmp/image.jpg?_ts=1652856201275

pi@allsky:~ $ grep IMG_DIR ~/allsky/config/config.sh
# The websites look in IMG_DIR for the current image.
# If you use the default IMG_DIR and have the Allsky website, set "imageName" in
# Only change IMG_DIR if you know what you are doing.
IMG_DIR="current/tmp"
if [ "${IMG_DIR}" = "current/tmp" ]; then
        CAPTURE_SAVE_DIR="${IMG_DIR}"

topazkuntadi, 2022-05-18:
Here's the message after I replace postData.sh :

>>>

/home/pi/allsky/scripts/postData.sh: line 6: /variables.sh: No such file or directory
/home/pi/allsky/scripts/postData.sh: line 7: /config.sh: No such file or directory
/home/pi/allsky/scripts/postData.sh: line 8: /ftp-settings.sh: No such file or directory
ERROR posting twilight data: ***
postData.sh: WARNING: No local or remote website specified so 'data.json' not copied anywhere.
***.

>>>

Here's after I add "set -x" : 

>>>

Settings saved and service NOT restarted+ source /variables.sh
/home/pi/allsky/scripts/postData.sh: line 6: /variables.sh: No such file or directory
+ source /config.sh
/home/pi/allsky/scripts/postData.sh: line 7: /config.sh: No such file or directory
+ source /ftp-settings.sh
/home/pi/allsky/scripts/postData.sh: line 8: /ftp-settings.sh: No such file or directory
+ '[' -z '' ']'
+++ dirname /home/pi/allsky/scripts/postData.sh
++ realpath /home/pi/allsky/scripts/..
+ ALLSKY_HOME=/home/pi/allsky
+ export ALLSKY_HOME
++ basename /home/pi/allsky/scripts/postData.sh
+ ME=postData.sh
+ source /home/pi/allsky/variables.sh
++ '[' '' = '' ']'
++ set -a
++ ALLSKY_VARIABLE_SET=true
+++ basename /home/pi/allsky/variables.sh
++ ME2=variables.sh
++ tty --silent
++ ON_TTY=0
++ RED=
++ GREEN=
++ YELLOW=
++ NC=
++ '[' /home/pi/allsky = '' ']'
++ ALLSKY_TMP=/home/pi/allsky/tmp
++ ALLSKY_CONFIG=/home/pi/allsky/config
++ ALLSKY_SCRIPTS=/home/pi/allsky/scripts
++ ALLSKY_IMAGES=/home/pi/allsky/images
++ ALLSKY_NOTIFICATION_IMAGES=/home/pi/allsky/notification_images
++ ALLSKY_NOTIFICATION_LOG=/home/pi/allsky/tmp/notification_log.txt
++ ALLSKY_DARKS=/home/pi/allsky/darks
++ PORTAL_DIR=/var/www/html
++ WEBSITE_DIR=/var/www/html/allsky
+ '[' 0 -ne 0 ']'
+ source /home/pi/allsky/config/config.sh
++ CAMERA=ZWO
++ IMG_UPLOAD=false
++ IMG_UPLOAD_FREQUENCY=1
++ IMG_DIR=current/tmp
++ IMG_RESIZE=false
++ IMG_WIDTH=2028
++ IMG_HEIGHT=1520
++ CROP_IMAGE=false
++ CROP_WIDTH=640
++ CROP_HEIGHT=480
++ CROP_OFFSET_X=0
++ CROP_OFFSET_Y=0
++ AUTO_STRETCH=false
++ AUTO_STRETCH_AMOUNT=10
++ AUTO_STRETCH_MID_POINT=10%
++ RESIZE_UPLOADS=false
++ RESIZE_UPLOADS_SIZE=962x720
++ IMG_CREATE_THUMBNAILS=true
++ REMOVE_BAD_IMAGES=true
++ REMOVE_BAD_IMAGES_THRESHOLD_LOW=1
++ REMOVE_BAD_IMAGES_THRESHOLD_HIGH=90
++ CAPTURE_EXTRA_PARAMETERS=
++ DAYTIME_CAPTURE=true
++ DAYTIME_SAVE=false
++ TIMELAPSE=true
++ TIMELAPSEWIDTH=0
++ TIMELAPSEHEIGHT=0
++ TIMELAPSE_BITRATE=2000k
++ FPS=25
++ VCODEC=libx264
++ PIX_FMT=yuv420p
++ FFLOG=warning
++ KEEP_SEQUENCE=false
++ TIMELAPSE_EXTRA_PARAMETERS=
++ UPLOAD_VIDEO=true
++ KEOGRAM=true
++ KEOGRAM_EXTRA_PARAMETERS='--font-size 1.0 --font-line 1 --font-color '\''255 255 255'\'''
++ UPLOAD_KEOGRAM=false
++ STARTRAILS=true
++ BRIGHTNESS_THRESHOLD=0.1
++ STARTRAILS_EXTRA_PARAMETERS=
++ UPLOAD_STARTRAILS=false
++ THUMBNAIL_SIZE_X=100
++ THUMBNAIL_SIZE_Y=75
++ DAYS_TO_KEEP=5
++ WEB_DAYS_TO_KEEP=
++ POST_END_OF_NIGHT_DATA=true
++ DARK_FRAME_SUBTRACTION=false
++ WEBUI_DATA_FILES=
++ UHUBCTL_PATH=
++ UHUBCTL_PORT=2
++ CAMERA_SETTINGS_DIR=/etc/raspap
++ END_OF_USER_SETTINGS=true
++ '[' ZWO = '' ']'
++ '[' -z 5 ']'
++ AUTO_DELETE=true
++ NIGHTS_TO_KEEP=5
++ '[' current/tmp = current/tmp ']'
++ CAPTURE_SAVE_DIR=/home/pi/allsky/tmp
++ CAMERA_SETTINGS=/etc/raspap/settings_ZWO.json
+++ jq -r .filename /etc/raspap/settings_ZWO.json
++ FULL_FILENAME=image.jpg
++ EXTENSION=jpg
++ FILENAME=image
+++ jq -r .debuglevel /etc/raspap/settings_ZWO.json
++ ALLSKY_DEBUG_LEVEL=1
+ source /home/pi/allsky/config/ftp-settings.sh
++ PROTOCOL=local
++ IMAGE_DIR=
++ WEB_IMAGE_DIR=
++ VIDEOS_DIR=/home/pi/dropbox
++ VIDEOS_DESTINATION_NAME=
++ WEB_VIDEOS_DIR=
++ KEOGRAM_DIR=
++ KEOGRAM_DESTINATION_NAME=
++ WEB_KEOGRAM_DIR=
++ STARTRAILS_DIR=
++ STARTRAILS_DESTINATION_NAME=
++ WEB_STARTRAILS_DIR=
++ REMOTE_HOST=
++ REMOTE_USER=
++ REMOTE_PASSWORD=
++ LFTP_COMMANDS=
++ AWS_CLI_DIR=/home/pi/.local/bin
++ S3_BUCKET=allskybucket
++ S3_ACL=private
++ jq -r .angle /etc/raspap/settings_ZWO.json
+ angle=-6
++ jq -r .latitude,.longitude /etc/raspap/settings_ZWO.json
+ latlong=' 7.75S
110.37E'
++ date +%z
+ timezone=+0700
++ sunwait list rise angle -6 7.75S 110.37E
+ sunrise=05:21
+ sunrise_hhmm=05:21
++ sunwait list set angle -6 7.75S 110.37E
+ sunset=17:49
+ sunset_hhmm=17:49
+ '[' 05:21 = --:-- -o 17:49 = --:-- ']'
++ date +%Y-%m-%d
+ today=2022-05-18
+ FILE=data.json
+ OUTPUT_FILE=/home/pi/allsky/tmp/data.json
+ echo '{'
+ echo '"sunrise":' '"2022-05-18T05:21:00.000+0700",'
+ echo '"sunset":' '"2022-05-18T17:49:00.000+0700",'
+ echo '"streamDaytime":' '"true"'
+ echo '}'
+ COPIED=false
+ typeset -i RETCODE=0
+ '[' -d /var/www/html/allsky ']'
+ '[' '' '!=' '' ']'
+ '[' false = false ']'
+ echo '***'
+ echo -e 'postData.sh: WARNING: No local or remote website specified so '\''data.json'\'' not copied anywhere.'
+ echo '***'
+ exit 1
ERROR posting twilight data: ***
postData.sh: WARNING: No local or remote website specified so 'data.json' not copied anywhere.
***.

EricClaeys, 2022-05-27:
@topazkuntadi    The errors with the allsky website and WebUI have been fixed.  If you re-install them per the instructions in the README.md file on the main GitHib page, that should fix those problems.
On the Pi Zero you may need to use 2048 for the swap size file, especially if you are getting RETCODE=137 which is due to not enough swap space.

If you ever get the "Allsky is Restarting" message for more than a few minutes, please follow the instructions on the "Reporting Issues" Wiki page.

EricClaeys, 2022-06-03:
@topazkuntadi    I just noticed that your `ftp-settings.sh` file doesn't specify a server (REMOTE_HOST) and you don't appear to have the Allsky Website installed (there is no `/var/www/html/allsky`) on your Pi, so there is no where to upload `data.json` to.
If that is correct, you should set `POST_END_OF_NIGHT_DATA="false"` in config/config.sh.

topazkuntadi, 2022-06-04:
hi. i turn it on because i need to use PROTOCOL="local" to copy my end of night videos locally, then use rclone to upload to dropbox.

EricClaeys, 2022-06-09:
@topazkuntadi   What did you just turn on?  POST_END_OF_NIGHT_DATA >

EricClaeys, 2022-06-09:
@topazkuntadi   Are things working now?

EricClaeys, 2022-06-19:
Closing - no response from user in a couple weeks, and it appears things are working.

closed 2022-06-19 by EricClaeys