AllskyTeam/allsky #1225: Failed to install (3 times) - no live image view or config editor loading

digitalurban, 2022-05-19:
Evening all,

I do have a successful version of allsky running but I thought I would install it on another pi. I have now checked all things and have failed to get it running 3 times.

With a PiHQ cam - I don't seem to get a live view and when I go to to the Editor, nothing loads

[image]
[image]

I have installed via the info on the GitHub page - am I missing something really stupid?

Any help would be great,

Andy

topazkuntadi, 2022-05-19:
i also experience with this error. it happen since yesterday (new install).

i have old working files from previous install (3 days ago), copy 3 files from ~/allsky/gui and variables.sh from ~/allsky before install the webUI. and it works. 

maybe the last update of webUI have bug.

Bi7, 2022-05-19:
Same here, paths just don't work.

New install, tried changing paths in config.sh and lighttpd.conf  but no success.

EricClaeys, 2022-05-20:
Folks, please attach your /etc/lighttpd/lighttpd.conf file and /var/www/html/includes/functions.php.
You will need to add append a `.txt` to their names.

EricClaeys, 2022-05-20:
Is everything else working, just not the Live View and Editor pages?

ramsessk, 2022-05-20:
It happened to me.
It worked before but somehow it started. Fresh start from os installation didn't help.
My solution was to add '/' at /etc/lighttpd/lighttpd.conf where it defines actual locations such as '/home/pi/images'
My best guess is that some updates affect this.

topazkuntadi, 2022-05-20:
@EricClaeys i found other bug.

i use raspi NoIR. if i choose manual exposure & manual gain. no problem.

but i assure if i choose auto exposure & auto gain,  the value of auto exposure & value of auto gain is ignored.

it get value of manual exposure & manual gain as the maximum exposure & gain. 

i already check and re check, it happen on both day and night. thanks.

ON2ON-HAM, 2022-05-20:
@EricClaeys , as far as I found, only the Editor, and the live view. But on the Allsky Webpage, the camera details ,, even filled in correctly, are not shown.
I will wait for a new release, and reinstall then.

digitalurban, 2022-05-20:
Just looked at my lighttpd.cong as per @ramsessk but my urls seem to be ok:

alias.url                   = ("/current/" => "/home/pi/allsky")
alias.url                   += ("/images/" => "/home/pi/allsky/images")
alias.url                   += ("/website/" => "/var/www/html/allsky")

So still no joy...

Andy

ramsessk, 2022-05-20:
My one was same like this:
alias.url = ("/current/" => "/home/pi/allsky")
alias.url += ("/images/" => "/home/pi/allsky/images")
alias.url += ("/website/" => "/var/www/html/allsky")

This made me confused at first because based on info here, these were recommended ones.
I just tried to add '/' at the end (e.g. /home/pi/allsky/") and it solved it.
I think that this is just workaround only. I don't know the root cause of it.
Anyway I have still one error message like the below.

Thanks.
Steven
[image]

ON2ON-HAM, 2022-05-20:
@digitalurban  you need to put a / behind all 3 aliases,, save and reboot,
It does work,  :

alias.url = ("/current/" => "/home/pi/allsky/")
alias.url += ("/images/" => "/home/pi/allsky/images/")
alias.url += ("/website/" => "/var/www/html/allsky/")

digitalurban, 2022-05-20:
Thanks for the replies - the / at the back worked (sort of) the Editor is back but now I get 'Error see var/log/allsky.log for more information' on the Live View (but there is no allsky.log file in var/log)...

So closer but might have to wait until the general issue gets a fix as I assume this happens to all current installs...

Andy

ON2ON-HAM, 2022-05-20:
@digitalurban 
This is what should be in line 250 of allsky.sh =

 "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"           # run the main programTake

Take a look in yours if it's the same but do not change or save anything yet :  

sudo nano -l /home/pi/allsky/allsky.sh

digitalurban, 2022-05-20:
Yep mine is the same (thanks for the help btw)...

"${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"           # run the main program

ON2ON-HAM, 2022-05-20:
@digitalurban , That can always work. I re-installed after my first try without the /'s,, then but the '/s and it worked. Also to avoid any other changes i might have put in ;)

EricClaeys, 2022-05-20:
All,
Sorry for the hassle.
A recentchage removed the trailing `/` from the three alias lines in the lighttpd.conf file.  I will submit a fix soon, but it will just add the missing slashes.
After you add them, you don't need to reboot; you can just execute:
~~~
systemctl restart lighttpd
~~~

EricClaeys, 2022-05-20:
@topazkuntadi. We got ahead of ourselves a little.  The WebUI's Camera Settings page has a few options that aren't supported yet, like `Max Exposure`.  The manual exposure values are currently used as the max exposure values.
This will be fixed in the next release.

EricClaeys, 2022-05-21:
All,  We just posted a fix, but in order to use it you'll need to reinstall allsky.  I suggest simply adding the missing trailing "/", so the lines in `/etc/lighttpd/lighttpd.conf` look like:
~~~
alias.url = ("/current/" => "/home/pi/allsky/")
alias.url += ("/images/" => "/home/pi/allsky/images/")
alias.url += ("/website/" => "/var/www/html/allsky/")
~~~
Then execute:
~~~
systemctl restart lighttpd
~~~

This should fix the issues with the WebUI where it wouldn't display the live view image and the Editor page didn't display the files.  If you have a different problem, please follow the instructions at the Wiki Page on [Reporting Issues](https://github.com/thomasjacquin/allsky/wiki/Reporting-Issues).

ON2ON-HAM, 2022-05-21:
@EricClaeys The fix works fine, just re-installed and now it is ok, however ..
The Post on MAP function at the end of the Camera settings does not do anything any more.
When changing from Show on MAP 'YES' to 'NO', nothing happens,
The normal info on top of the Camera settings page is also not shown, it just mentions restart of allsky, but whatever change i make on the post on map info, it is not taken into account./
And my 3rd camera is still not visible on the map

EricClaeys, 2022-05-21:
@ON2ON-HAM 
> The normal info on top of the Camera settings page is also not shown, it just mentions restart of allsky, but whatever change i make on the post on map info, it is not taken into account.

Can you attach a screen shot of this please?

ON2ON-HAM, 2022-05-21:
@EricClaeys  here you go , i hope it is helpfull.
Like I mentioned only a manual map update works for me.

[image]

[image]

EricClaeys, 2022-05-22:
@ON2ON-HAM   If I'm interpreting your screen shots correctly, what's not working with the new release is the map data isn't being updated (or at least the message isn't being displayed) when you change one of the map data fields and then click on the "Save Settings" button at the bottom of the page.  based on the message "Settings saved and service restarted", I assume the other settings on the page are working.  Is that correct?

Would you please do the following?
~~~
sudo truncate -s 0 /var/log/lighttpd/error.log
~~~

Then update some map data and press F12 in the browser, then click on `>>` then `Console` and let me know what messages are there.  Also, let me know the contents of `/var/log/lighttpd/error.log`, if any.

Thanks.

ON2ON-HAM, 2022-05-22:
[image]

When i change some fields in the Map Data section. They do stay after in the page, also after a reboot, but do not get pushed to the server from Thomas. It's like the Map Data section has no way to communicate to the outside .

The only thing that pushes a change in one of the fields is a 'manual Map Data' upload from console.

The error log remains empty @EricClaeys  , I include here the older logs .

[errorlog.txt](https://github.com/thomasjacquin/allsky/files/8748783/errorlog.txt)
[error.log.2.gz](https://github.com/thomasjacquin/allsky/files/8748785/error.log.2.gz)

[image]

One error in Console (Dutch) :  
[image]

ON2ON-HAM, 2022-05-22:
@EricClaeys , 2 different camera's, 192.168.1.62 is with older Allsky soft, 192.168.1.43 is with latest Allsky soft :  

[image]

[image]

EricClaeys, 2022-05-25:
@ON2ON-HAM, @digitalurban, @WirthmU, @topazkuntadi, @Bi7, @ramsessk

I just found a bug in the `gui/install.sh` script that may be causing some of your problems.
Please make sure line 12 of `/var/www/html/includes/functions.php` is:
~~~
define('ALLSKY_SCRIPTS', '/home/pi/allsky/scripts');
~~~
### Let me know if this fixes the remaining problems in this Issue.

@ramsessk, Steven, don't worry about that error with "monokia.min.css.map".  None of the .map files are present.

ON2ON-HAM, 2022-05-25:
@EricClaeys  ok, just changed it :  (It had _XX in mine)

[image]

After reboot/restart : 

[image]

Eureka !! Thanks a zillion @EricClaeys 
No clue how the XX got there, i sure did not change it ;)

EricClaeys, 2022-05-25:
@ON2ON-HAM   Thanks for the feedback.  The "_XX" got there because of the bug - it should have removed it.
The fix was just uploaded so next time you or anyone else updates the WebUI it'll work.

EricClaeys, 2022-05-27:
@digitalurban    Andy,  Did the change above fix the problem?  If so, would you please close the Issue?
Thanks - Eric

digitalurban, 2022-05-31:
It did :) closing...