AllskyTeam/allsky #1228: GUI not working on Ubuntu 22.04

JosWennmacker, 2022-05-22:
Actually I solved the issue already but I want to bring it to your attention. 
I installed Allsky on Ubuntu 22.04 and found the gui not working. The problem was caused by changed permissions settings.
Since Ubuntu 21.04 the default permissions for the home directory are 0750 in stead of 0755. So the gui is not allowed to read config.sh, resulting in '' CAMERA type not defined. Please update 'home/pi/allsky/config.config.sh'.
Changing the permissions for /home/pi to 0755 resolved the issue.

EricClaeys, 2022-05-25:
Thanks for that information.  I will make sure that is fixed in the next release.
Eric

EricClaeys, 2022-05-25:
Closing issue.  I just made the fix in the next version of the installer.

closed 2022-05-25 by EricClaeys