AllskyTeam/allsky #1230: Loading image only - no live images.

pczjrh, 2022-05-22:
http://allsky.local/allsky/index.php is just showing "Loading" on a completely fresh install (fresh clone today - 22/5/22).
I'm running on Rpi-Buster.

No settings have been changed, images are being saved in /home/pi/allsky/images/* and /home/pi/allsky/images/*/thumbnails just fine.
The temp image is being saved in /home/pi/allsky/tmp as image.jpg.
I've checked the lighttpd.conf for the extra /'s, and they are there.
I installed the website package only, and not the admin-website, as I'm conformatble with command-line (no need to).
 
As far as I can tell from the docs, this is the right configuration and should be working, but alas, I'm stuck with no live images.
Suggestions?

Thanks,
Jason.

pczjrh, 2022-05-23:
is this right in the ftp-settings.config file:

PROTOCOL="local"
IMAGE_DIR="/home/pi/allsky/images/"
WEB_IMAGE_DIR="/var/www/html/allsky/images/"
VIDEOS_DIR="/home/pi/allsky/images/"
VIDEOS_DESTINATION_NAME=""
WEB_VIDEOS_DIR="/var/www/html/allsky/videos/"
KEOGRAM_DIR="/home/pi/allsky/images/"
KEOGRAM_DESTINATION_NAME=""
WEB_KEOGRAM_DIR="/var/www/html/allsky/keograms/"
STARTRAILS_DIR="/home/pi/allsky/images/"
STARTRAILS_DESTINATION_NAME=""
WEB_STARTRAILS_DIR="/var/www/html/allsky/startrails/"

the image file is being saved in the web folder so that it is accessible to the webpage now.
However, I had to modify the index files on the web-admin and on the /allsky/index.php to point at the file. 

The reason I've been getting "Loading" page is that it is hard coded in the index.php file. I manually pointed it at /images/image.jpg  and it loads the live-view fine. 

That's one thing down, but to see the folders of images, keograms etc..... 
You can't point to files outside of the web-root (/var/www/html/) folders; how is this meant to work?

Clearly, I'm not getting something right in the configx, as no-one else seems to be having these sorts of issues.
Thanks,
Jason.

EricClaeys, 2022-05-25:
Jason,
If you didn't install the WebUI and didn't install a different web server, the Allsky Website won't work.  It requires a web server.  I suggest installing the WebUI.

Eric

EricClaeys, 2022-05-25:
@pczjrh    Jason,  The `ftp-settings.sh` file has comments that look like `If you have the "allsky-website" package installed ...` that tell you what to use if you have the website on your Pi, which you do.
Since you only have the website on the Pi, you don't need to use the WEB_*_DIR variables - they should be left blank.
Despite what the comment says about IMAGE_DIR, you can also leave it blank because the website looks for the file in ~/allsky/tmp, so there's no reason to copy it anywhere.

The index.php files don't need to be modified.  I suggest installing the WebUI (allsky-portal) then reinstalling the allsky-website to undo the changes to the files you made.
Instructions for installing them are in the README.md file on the main GitHub page.

pczjrh, 2022-05-28:
Completely clean install of rpi-bullseye, clean repo clone and it worked out of the box.
All resolved.
Thanks.

closed 2022-05-28 by pczjrh

EricClaeys, 2022-05-28:
@pczjrh    Jason, I'm glad it's working now.  I suggest saving the configuration files on a PC or USB disk in case you need to start over you'll have the settings.