AllskyTeam/allsky #1231: WARNING: data.json is 4 days old. Check Allsky log file if 'postData.sh' has been running successfully at the end of nighttime.

cboivin11, 2022-05-23:
Hi

here is my warning message on the web/allsky window
WARNING: data.json is 4 days old.
Check Allsky log file if 'postData.sh' has been running successfully at the end of nighttime.

I have not found the postData.sh file 

What can i do to fix this issue ?

Thank you
Claude

EricClaeys, 2022-05-25:
@cboivin11   Claude,
The Allsky scripts like `postData.sh` are in ~/allsky/scripts.  However, what you want to do is check the log file (/var/log/allsky.log) to see if `postData.sh` has been running - you can do that by searching for "postData.sh" in the log file and checking if there are any errors.

But first, try refreshing your web browser.  If the message does NOT go away, do the above.  If the message DOES go away, then you're probably seeing the message because you've had the browser window running for several days.  The code only checks for updates to the file in the morning if that browser window is the open window, which if you're like me, it likely wasn't.  This is a bug that I'll fix at some point, so I'll leave this Issue open until I fix it.

themagisterCH, 2022-05-29:
If I may hijack this: My data.json is uploaded to tmp - the website is looking for it in the root though. How can I either change the upload path to root, or find the function that takes it from tmp and makes a copy in root?

EricClaeys, 2022-05-29:
@themagisterCH,  Except for the configuration files, you should never need to change any `*.sh` or `*.php` files to get Allsky to work.   In your case, the `IMAGE_DIR` variable in `config/ftp-settings.sh` should be blank, and in `config/config.sh` the `UPLOAD_DATA` variable should be `true`.

The `postData.sh` script always copies data.json to the website's root directory if it's on the Pi.

If this is not happening there is either a bug or a misconfiguration.

Stephen2615, 2022-06-03:
I also have this issue but it might be due to me using sftp instead of ftp and the associated changes that I made to the scripts as directed. I have this in the ftp-settings.sh:
IMAGE_DIR="domains/<some directory>/public_html/allsky/upload"
I can easily put the data.json file into the remote directory with my own scripts but I was wondering if it is supposed to go at a specific time like sunrise as the uploads seem to be done then.

EricClaeys, 2022-06-03:
@cboivin11    Any update?

EricClaeys, 2022-06-03:
@Stephen2615   It doesn't need to be done at a specific time.  It's done at the end of night because that's the only time Allsky runs daily scripts.

`data.json` must go in the root of the allsky website where the other website files are, which is usually the `allsky` folder.  Are the rest of your files in `upload`?  If not, and that's where you are copying the image.jpg files to, then that's a problem.  The images and `data.json` file put their files in IMAGE_DIR.

If you modified `upload.sh` I suggest you un-modify it. If things aren't working it's likely due to a mis-configuration or you are using a non-standard directory structure (which I suggest not doing - it's only asking for trouble).

Stephen2615, 2022-06-03:
Thanks for the response. The other files are going up properly as they have the appropriate path in the ftp-settings.sh.. As you suggested, I added a -o port to the upload script. I think that the data.json might need something like the individual paths set in the ftp-settings.sh script as it is the only problem file that goes to my web server.

Stephen2615, 2022-06-03:
I think I need to pay a bit more attention to the posts. I just realised the data.json file is in the upload directory. I made a change in the ftp-settings.sh to:
IMAGE_DIR="domains/<some directory>/public_html/allsky"
I expect it will work properly now.

EricClaeys, 2022-06-03:
@Stephen2615 if you don't want to wait until morning you can run `postData.sh` manually if you have the most current one.

Stephen2615, 2022-06-25:
For some reason since my new fresh install, the data.json is not getting uploaded automagically. I can run it manually without any issues and with a -x after bash, I can see it works properly and goes into the right directory. The message on the website disappears immediately. What calls this script to run?

EricClaeys, 2022-06-25:
@Stephen2615 The end of night script calls `postData.sh`, but ONLY if **POST_END_OF_NIGHT_DATA=true** in config/config.sh.
The very last thing the website installation script does is tell the user to update that setting.  Did you see that line?

The next release will set POST_END_OF_NIGHT_DATA automatically which will eliminate this problem.

Stephen2615, 2022-06-25:
Need I say anymore than DOH!!

EricClaeys, 2022-06-25:
With the new release you won't even have to say that!!!!!

EricClaeys, 2022-07-30:
Closing issue.  It's fixed in the next release.
The bug was due to the website only refreshing the image, and checking for an old data.json file, if the browser window was active.  In many cases, users open a different window, then go to bed, and in the morning the Allsky Website missed the time it was supposed to re-read the file.  The same thing happens the next day, etc., until the file is "too old".
The fix is to have the website check every 5 intervals, even if the window is NOT active.

closed 2022-07-30 by EricClaeys

cboivin11, 2022-10-11:
Hi Eric

 

Thank you for your help, but sorry I did have the time to really try to fix
this problem until today.

 

The postData.sh does not seem to be running because did not find  it in the
allsky.log

 

I dont think it s my browser because I rebooted a few times since and
nothing changes.

 

I put my last allsky.log with this email

 

Thanks

Merci et bonne journée

 

Claude Boivin

nighttime. (Issue #1231)

 

<https://github.com/thomasjacquin/allsky/issues/1231#issuecomment-1136780997
XBN3ANCNFSM5WXJHQ7A> .
<https://github.com/notifications/beacon/ALDUPFJ4UQ6RVOEEIRX3O23VLXBN3A5CNFS
M5WXJHQ7KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOIPA6NRI