AllskyTeam/allsky #1233: PiHQ Camera Not Found - Works on Pi4, not Pi 3A+/Zero

digitalurban, 2022-05-24:
Morning - I have been trying to install allsky onto a Pi 3A+ to reduce power usage (I currently have it running on a 4). With a fresh new install I can't seem to get allsky to recognise the PiHQ Camera. I have checked and double checked the cables, even changed the cables but no joy. 

Any ideas would be great ...

[image]

digitalurban, 2022-05-25:
Thanks for the reply, on Bullseye there is no longer the camera enable option - still no joy but it seems to be a Pi issue - so noted

-  On Raspberry Pi 3 and earlier devices running Bullseye you need to re-enable Glamor in order to make the X-Windows hardware accelerated preview window work. To do this enter sudo raspi-config at a terminal window and then choose Advanced Options, Glamor and Yes. Finally quit raspi-config and let it reboot your Raspberry Pi (from https://support.thepihut.com/hc/en-us/articles/360008782198-Camera-Module-isn-t-working)

So everything works out of the box on Pi 4's but so no joy with a Pi3 A+

Andy

digitalurban, 2022-05-26:
Will do thanks ! (sorry I should have noticed it in the wiki before)... only using a Pi3 to save a bit of energy as I'm running my allsky from solar power...

EricClaeys, 2022-05-26:
@digitalurban Andy,
If the Wiki solves your problem would you please close this Issue?
Thanks

digitalurban, 2022-05-28:
It did :) thanks

closed 2022-05-28 by digitalurban