AllskyTeam/allsky #1264: Problems after changing the Config

alrebi, 2022-06-02:
Hello, I have already tested with several Raspberry Pis but I just can't get it to work. When first started, the camera still displays a normal image. If I then change something in the config, the live image remains either completely white or black, depending on the setting of Brightnes/Auto Gain or something else. I can't get an image to load after the first start. I even have to set "Remove bad images" to false to even get an image. I would be grateful for help.
Hardware: Raspberry Pi 3/4 HQ cam . Software Bullseye or Buster
[Log file.txt](https://github.com/thomasjacquin/allsky/files/8823413/Log.file.txt)
[Settings.txt](https://github.com/thomasjacquin/allsky/files/8823414/Settings.txt)

Pedritoprint, 2022-06-02:
What means "a normal image"? 
Did you made the correct installation see RAEDME.md and configuration? see WIKI. 
Correct camera set during installation?
Where did you have the settings.txt from? I can't find a file corresponding to your "settings.txt in my installation. I have also Rapsi 4B with Bullseye inuse and an RPiHQ

alrebi, 2022-06-02:
Hello, the picture is normally sharp at first. The only thing different there is that the picture has a red cast. No matter what setting I make after the restart, the picture is extremely dark. I still have a cam running but with an old Bullseye version and an older Allsky version. It runs without any problems. I got the settings from the wiki instructions. I installed the webui

EricClaeys, 2022-06-03:
@alrebi   Hi,  Please follow the instructions on the Wike page on "Reporting Issues", then attach the log file.  The one you attached is abbreviated so doesn't give sufficient information.  The Settings.txt file is also truncated.  Please also include the full `config/config.sh` file, appending ".txt" to its name.  It helps if you don't rename the files.
Thanks

@Pedritoprint   Peter, the "Settings.txt" file is the last portions of alebi's config.sh file.

alrebi, 2022-06-05:
Hello, sorry I was traveling. (Unfortunately my English is very bad, I work with the Google translator) Today I dealt with it again. In my opinion, it doesn't give me an error even if I change values ​​in the config. Since I was a little frustrated, I copied the etc/raspap from my old working system to the new one and replaced it with it. Lo and behold it works now. The error would have to be in the new config for the Rhp

EricClaeys, 2022-06-05:
@alrebi I would really like to determine what the problem is so I can fix it and you and others won't run into it again.  Can you revert to the bad configuration file, then follow the instructions on the "Reporting Issues" Wiki page, then attach the log file and the bad configuration file. 

Thanks much

EricClaeys, 2022-06-07:
@alrebi   Reine, do you have the good (old) and bad `settingsRPiHQ.json` files you can attach?  If I understand correctly, with the "bad" version all your images are very dark. I assume with a 500 us exposure time which will be black at night, but might be ok during the day.

alrebi, 2022-06-08:
Yes all are very Dark
[settingsRPiHQ-json.txt](https://github.com/thomasjacquin/allsky/files/8858781/settingsRPiHQ-json.txt)
.

EricClaeys, 2022-06-09:
@alrebi   Is that the "good" or "bad" settings file?

alrebi, 2022-06-09:
@EricClaeys  the good

EricClaeys, 2022-06-09:
@alrebi   And do you have the "bad" settings file so I can compare?

alrebi, 2022-06-10:
[settingsRPiHQ-bad.txt](https://github.com/thomasjacquin/allsky/files/8876387/settingsRPiHQ-bad.txt)

EricClaeys, 2022-06-10:
@alrebi thanks, I will look at the files in a bit.  In the meantime, your Font Color in the bad file has only 1 number and should have 3. I don't think it would cause the problem you are seeing but suggest fixing it to see.

EricClaeys, 2022-06-15:
@alrebi   Are the "good" and "bad" settings files for the same camera?  Quite a few of the settings were different, including longitude, latitude, and auto exposure/gain.

Save your current `/etc/raspap/settings_RPiHQ.json` file, then use the attached file, removing ".txt" from the file name.  It has all the settings from your "good" file, plus the new settings for the newest Allsky.

[settings_RPiHQ.json.txt](https://github.com/thomasjacquin/allsky/files/8905907/settings_RPiHQ.json.txt)

alrebi, 2022-06-15:
Thank you I will test it. To the differences. Yes, the settings belong to the same camera. I actually wanted to show that I only have to change any value to create the problem. It didn't matter if I changed identical values ​​or just a single value. The error starts immediately.
Greetings Reiner

EricClaeys, 2022-06-19:
@alrebi Have you had a chance to test with the settings_RPiHQ.json file I uploaded?

alrebi, 2022-06-22:
Soory I haven't been home for a few days. Yes so far it works. Thank you for your work.

EricClaeys, 2022-06-23:
@alrebi   Reiner,   If it continues to work for a few more days, would you please close this Issue?
Danke

closed 2022-06-24 by alrebi