AllskyTeam/allsky #1278: No picture taken after upgrade

blacc01, 2022-06-06:
Hi,

I upgraded my system, did everything as described, but there is no picture taken. WebUI is running. have following error:
pi@allsky:~/allsky $ sudo systemctl status allsky
● allsky.service - All Sky Camera
   Loaded: loaded (/etc/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2022-06-06 12:38:45 CEST; 3min 54s ago
 Main PID: 760 (allsky.sh)
    Tasks: 7 (limit: 4035)
   Memory: 97.2M
   CGroup: /system.slice/allsky.service
           ├─ 760 /bin/bash /home/pi/allsky/allsky.sh
           └─1018 /home/pi/allsky/capture -debuglevel 0 -tty 0 -aggression 75 -alwaysshowadvanced 0 -angle -6 -autousb 1 -awb 0 -camera -computer -coolerEnabled

jún 06 12:42:18 allsky allsky.sh[760]:  >xxx mean was 119 and went from 3 below min of 122 to 2 above max of 134, is now at 136; OK to set temp max to current_e
jún 06 12:42:19 allsky allsky.sh[760]: === next exposure=424 (multiply by 0.801) [last_exposure_us=530, maxAcceptable=134, mean=152]
jún 06 12:42:19 allsky allsky.sh[760]:   > Sleeping 15.0 sec from histogram exposure
jún 06 12:42:19 allsky allsky.sh[760]: removeBadImages.sh: 'image-20220606124218.jpg' not found in '/home/pi/allsky/images/current'
jún 06 12:42:34 allsky allsky.sh[760]: STARTING EXPOSURE at: 2022-06-06 12:42:34   @ 530 us (0.530 ms)
jún 06 12:42:34 allsky allsky.sh[760]: === next exposure=474 (multiply by 0.896) [last_exposure_us=530, maxAcceptable=134, mean=136]
jún 06 12:42:34 allsky allsky.sh[760]: === next exposure=535 (multiply by 1.128) [last_exposure_us=475, minAcceptable=122, mean=119]
jún 06 12:42:34 allsky allsky.sh[760]: === next exposure=597 (multiply by 1.128) [last_exposure_us=530, minAcceptable=122, mean=119]
jún 06 12:42:34 allsky allsky.sh[760]:   > Sleeping 15.0 sec from histogram exposure
jún 06 12:42:35 allsky allsky.sh[760]: removeBadImages.sh: 'image-20220606124234.jpg' not found in '/home/pi/allsky/images/current'

The folder /home/pi/allsky/images/current is not set in any config file, so i don't understand whats going here.

Please help

EricClaeys, 2022-06-07:
@blacc01 Please follow the instructions on the Wiki page for Reporting Issues, then upload the log file.
Thanks

blacc01, 2022-06-07:
I decided to start everything from scratch, as the Raspberry OS is old. So i deleted everything, new OS, and everything installed. Now its seems to be working now.