AllskyTeam/allsky #1279: aurora levels

cboivin11, 2022-06-06:
Hi

in the web allsky it gives me the level of the aurora for 3 days
Im not sure that these levels are accurate because when i look at the nooa website and evaluate the KP for me location the levels are way lower that the ones given by the allsky screen.

So I would like to understand how this evaluation is done and how to modify to show the real levels for my location...

Claude

EricClaeys, 2022-06-09:
@cboivin11    Claude, we use [this page](https://services.swpc.noaa.gov/text/3-day-forecast.txt) to get the aurora forecast data.  We do not take the location into account, other than northern and southern hemisphere.  If you have a URL where the latitude and longitude can be specified to give more location-based results, let me know.

cboivin11, 2022-06-09:
Eric

 

Does it use this scale to show the levels of aurora ?  if yes then I do
agree with the shown levels 

 <https://auroraforecast.is/kp-index/> Kp-index | Aurora Forecast

 

This web site gives the levels with the Kp but it is not function of the
latitude and its ok

 

 

Merci et bonne journée

 

Claude Boivin

 

<https://github.com/thomasjacquin/allsky/issues/1279#issuecomment-1150593800
FHFLANCNFSM5X7IYBTQ> .
<https://github.com/notifications/beacon/ALDUPFMVTLSGKKFF2PXQOV3VOFHFLA5CNFS
M5X7IYBT2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOISKKWCA

EricClaeys, 2022-06-10:
@cboivin11 Claude,  I am not sure what it uses; I have never looked into it.  The code uses this mapping:
~~~
0: "Low",	
1: "Low",	
2: "Low",	
3: "Active",
4: "High",
5: "Extreme",
6: "Extreme",
7: "Extreme",
8: "Extreme",
9: "Extreme"
~~~
Which looks very similar to the page you posted.

cboivin11, 2022-06-10:
Yes indeed but could we propose something better like: 0: "Low",	
1: "Low",	
2: "Low",	
3: "medium",
4: "medium high",
5: "high",
6: "very high",
7: "Extreme",
8: "Extreme",
9: "Extreme"The gradation between level is more realistic for most viewersIs this possible or acceptable?Claude Téléchargez Outlook pour Android

EricClaeys, 2022-06-15:
@cboivin11    The next release of the website will use the  names that were in the link from Iceland, with a couple minor changes per NOAA.

Are you ok with closing this Issue?

cboivin11, 2022-06-15:
Eric

 

Yes and thank you for your time

 

 

Merci et bonne journée

 

Claude Boivin

 

<https://github.com/thomasjacquin/allsky/issues/1279#issuecomment-1156364372
G6IFANCNFSM5X7IYBTQ> .
<https://github.com/notifications/beacon/ALDUPFLN3MK3CIZUS66NCVLVPG6IFA5CNFS
M5X7IYBT2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOITWLQVA

EricClaeys, 2022-06-15:
Closing Issue with user agreement.  Feature requested is implemented in next release.

closed 2022-06-15 by EricClaeys