AllskyTeam/allsky #1285: starmap is 8h late in website

drakula1981, 2022-06-10:
Hi Everyone !

5 days ago, I've made a brand new configuration in my AllSky camera with raspbian bullseye and the last version of this wonderful software. Everything works well but the starmap drawn in the website is always precisely 8h late!
I'm sure for the geographical coordinates and the raspberry gives me the correct current hour (with the date command).
I'm pretty sure I've misconfigured some thing but I don't know what. Can you help me?

EricClaeys, 2022-06-10:
@drakula1981 The constellation overlay uses the latitude and longitude you specify in the `config.js` file, along with the current time.  In the `virtualsky.json` file you can have it display the time and location via the `showdate` and `showlocation` settings.  They display in the upper right of the overlay and aren't very bright.

You can use them to make sure the correct values are there.

Also make sure your `az` in `config.js` is correct.

drakula1981, 2022-06-10:
@EricClaeys Thanks for your fast answer! 
After the activation of the "showlocation" parameter, I saw that the geographical coordinates weren't updated and pointing far far west, not corresponding with the coordinates I configured in the config.js file!
I clicked on the location label and modified the coordinates et everything is OK now! I wonder were does come from the original coordinates after installation?

drakula1981, 2022-06-10:
In addition for my previous comment, I would add that i found this in the virtualsky.js (L269 to 271) :
	// Set location on the Earth
	this.setLongitude(-119.86286);
	this.setLatitude(34.4326);
And this is precisely the coordinates that are set by default until I modify by clicking on the location label. HTH

EricClaeys, 2022-06-14:
@drakula1981   Those are the default coordinates if nothing else is specified.
The default latitude/longitude in `config.js` are empty, so if you didn't update them after installing the website, then you'd get the behavior you saw.  If the "Title" field isn't updated a message should be displayed telling you to update the config.js file.  However, if you updated that field but not the latitude/longitude you wouldn't see the message but would get the behavior you saw.

If your problem is resolved, would you please close this Issue?  Thanks.

EricClaeys, 2022-06-19:
Closing Issue.  It appears the user didn't have the latitude and longitude set, or set correctly.

EricClaeys, 2022-06-23:
@drakula1981   Can you post the `config.js` file?  Is your website accessible via the Internet?

EricClaeys, 2022-07-01:
@drakula1981     Any update?

drakula1981, 2022-07-05:
Eric,
I unsterstood what happened, I wrote right in the config.js the good coordinates, while not enter the coordinates in the camera settings page. Later, I corrected the coordinates in the camera settings page (with the N and E cardinal points, as required) and then, it wrote the coordinates in the config.js file also but with double cardinal points (example : "48.123456NN" and "1.23456EE") but, if I understand the "latitude_longitude_comment" property in this file, "latitude and longitude must be decimal numbers with OUT the trailing N, S, E, and W.". This could be my source of problems ?

EricClaeys, 2022-07-05:
@drakula1981 The coordinates in config.js are only used for the Allsky Website to determine where to put the constellation overlay and what to enter into the right-side pop-out.

The coordinates in the camera settings determine when the daytime/nighttime transitions are, and the coordinates for the map.  If you change them in the WebUI they should get propagated to config.js.   Having the wrong format in that file will cause issues but nothing major.

I will take a look at the code to check for a bug when updating the coordinates from the WebUI.

EricClaeys, 2022-07-10:
@drakula1981   I didn't see anything in the code that would cause "NN" or "EE" to be added to the Website's `config.js` file.  However, I changed that code somewhat in the next version of Allsky so if there IS a bug hidden, hopefully it'll be fixed.

I believe your original problem is resolved, so are you ok with closing this Issue?

EricClaeys, 2022-07-15:
Closing issue - no response from user in two weeks and I believe the issue is resolved.

closed 2022-07-15 by EricClaeys