AllskyTeam/allsky #1291: Cannot view files uploaded to a remote website

hlsautter, 2022-06-17:
### Discussed in https://github.com/thomasjacquin/allsky/discussions/1290

<div type='discussions-op-text'>

<sup>Originally posted by **hlsautter** June 17, 2022</sup>
Selecting any of the "archive" buttons shows the following page:

Please run the following from the 'allsky' directory before using the Website:
website/install.sh --update

I installed the website "remotely" on my NAS and uploading files works correctly,
The current image is displayed, the camera information is displayed, the overlay works was expected.
However attempting to view any of the archived files suggests running the install script... but the site is not installed on the PI
So this script file does not exist. 

the check in functions.php seems to be the issue...

// If on a Pi, check that the placholder was replaced.
$isarm = preg_match("/(arm|aarch)/",php_uname());
if ($isarm && ALLSKY_CONFIG == "XX_ALLSKY_CONFIG" . "_XX") {
	// This file hasn't been updated yet after installation.
	echo "<div style='font-size: 200%;'>";
	echo "<span style='color: red'>";
	echo "Please run the following from the 'allsky' directory before using the Website:";
	echo "</span>";
	echo "<br><br><code>   website/install.sh --update</code>";
	echo "</div>";
	exit;
}

Any suggestions about how to fix this?  
</div>

I bypassed the check with this edit...
define('ALLSKY_CONFIG', 'XX_ALLSKY_CONFIG_X');
(Not a real solution but I got further in the code)

Next there were issues with the thumbnail directory.... I did create it manually and then discovered this "bug"

$thumb_dir = "$dir/thumbnails";
if (! is_dir($thum_dir)) {
	if (! mkdir($thum_dir, 0775))
		echo "<p>Unable to make '$thum_dir' directory. You will need to create it manually.</p>";
		print_r(error_get_last());
} 
fixed by this edit.....

$thum_dir = "$dir/thumbnails";

EricClaeys, 2022-06-19:
@hlsautter    Henry, It's been a pain trying to determine in a generic way when the website is running on a Pi.  Everything we've tried runs into one-offs that break our algorithm, just like your NAS does (it's not your fault).
The next version of the website will have a configuration variable that states whether or not it's on a Pi.  The default will be "true" but people running the website on remote servers will need to set it to "false".  Other configuration variables also need to be set prior to running the website, so having to set this one shouldn't be an issue.
The message you receive about running the update script will also be changed to mention the variable.

I've already implemented this on my test machine.  Are you ok with closing this Issue since it'll be fixed in the next release?

hlsautter, 2022-06-19:
That sounds like a good solution.  Setting a variable seems reasonable and saves a lot of coding effort that as you say keeps breaking as things change.  Thanks for the help with this and fixing the bug. with the variable name (these things happen!).   I will close this issue.

closed 2022-06-19 by hlsautter