AllskyTeam/allsky #1292: Saving Images

maserowik, 2022-06-18:
So everything was working a few days ago.  I was just check my pi and noticed that I am no NOT saving my day imaged.  I have the save images option selected in the config settings.
[image]
but looking at the file explore there are no files
[image]
The allsky log shows the files being written the \tmp folder.
[image]
but nothing is saved under the allsky\images folder
[image]
view from ssh session
[image]
View from Admin gui
[image]
 Allsky log attached.
[allsky.log](https://github.com/thomasjacquin/allsky/files/8933794/allsky.log)

maserowik, 2022-06-18:
user error

closed 2022-06-18 by maserowik

EricClaeys, 2022-06-19:
DAYTIME_CAPTURE determines whether images are taken (and saved in tmp).  DAYTIME_SAVE determines if those captured images are saved in allsky/images/DATE.