AllskyTeam/allsky #1306: How do i save Images on Raspberry Pi4 8 GB ALl Sky Camera

ssgaurav2000, 2022-06-22:
Hello I just set up my all sky camera , i can see live view and sky with stars  but not sure how to save images for time lapse- Images folder is empty and there is no option to save them on micro SD card 

See attached screenshot
[image]

EricClaeys, 2022-06-22:
@ssgaurav2000 welcome to the Allsky community!
Have you looked at the Wiki?  It has answers for most questions and problems.  Take a look at the "Settings" page.

When you are done, please close this Issue.

In the future,  if you have a question or want to discuss something,  please post in the Discussions link above.  I am trying to keep the Issues list just for problems so the developers have a feel for how much effort is needed. 

Thanks - Eric

ssgaurav2000, 2022-06-23:
I tried after all night of running the camera i still don't
[image]
 see any images saved or time lapse created here is my setting 

#!/bin/bash

# NOTES:
#  - Variable names must start at the beginning of a line.
#  - There should be nothing after a variable's value, i.e., no comments on the same line.
#  - Values can optionally be quoted with double quotes, NOT single quotes.
#  - If a variable occurs multiple times, only the last value is used.

# Choose between "ZWO" or "RPiHQ".
CAMERA="ZWO"

########## Images
# Set to "true" to upload the current image to your website.
IMG_UPLOAD="true"

# If IMG_UPLOAD is "true", upload images every IMG_UPLOAD_FREQUENCY frames, e.g., every 5 frames.
# 1 uploades every frame.
IMG_UPLOAD_FREQUENCY=1

# The websites look in IMG_DIR for the current image.
# When using the Allsky website from the "allsky-website" packge,
# "current" is an alias for "/home/pi/allsky".
# If you use the default IMG_DIR and have the Allsky website, set "imageName" in
# /var/www/html/allsky/config.js to:
#     imageName: "/current/tmp/image.jpg",
# This avoids copying the image to the website.
# Only change IMG_DIR if you know what you are doing.
IMG_DIR="current/tmp"

# Resize images before cropping, stretching, and saving.
# Adjust IMG_WIDTH and IMG_HEIGHT according to your camera's sensor ratio.
IMG_RESIZE="false"
IMG_WIDTH=2028
IMG_HEIGHT=1520

# Crop images before stretching and saving.
# This is useful to remove some of the black border when using a fisheye lens.
# If you crop an image you may need to change the "Text X" and/or "Text Y" settings in the WebUI.
CROP_IMAGE="false"
CROP_WIDTH=640
CROP_HEIGHT=480
CROP_OFFSET_X=0
CROP_OFFSET_Y=0

# Auto stretch images saved at night.
AUTO_STRETCH="false"
AUTO_STRETCH_AMOUNT=10
AUTO_STRETCH_MID_POINT="10%"

# Resize uploaded images.  Change the size to fit your sensor.
RESIZE_UPLOADS="false"
RESIZE_UPLOADS_SIZE="962x720"

# Create thumbnails of images.  If you are not running the WebUI consider changing this to "false".
IMG_CREATE_THUMBNAILS="true"

# Remove corrupt images before generating keograms, startrails, and timelapse videos.
# This only adds a few seconds per image as it's being processed.
REMOVE_BAD_IMAGES="true"

# If REMOVE_BAD_IMAGES is "true", images whose mean brightness is
# less than THRESHOLD_LOW or greater than THRESHOLD_HIGH percent (max: 100) will be removed.
# Set to 0 to disable the brightness checks.
REMOVE_BAD_IMAGES_THRESHOLD_LOW=1
REMOVE_BAD_IMAGES_THRESHOLD_HIGH=90

[... 118 lines cut ...]

# COMPATIBILITY CHECK
if [ -z "${DAYS_TO_KEEP}" ]; then		# Old version had AUTO_DELETE and NIGHTS_TO_KEEP
	AUTO_DELETE="false"
else
	AUTO_DELETE="true"
	NIGHTS_TO_KEEP="${DAYS_TO_KEEP}"
fi

# This is needed in case the user changed the default location the current image is saved to.
if [ "${IMG_DIR}" = "current/tmp" ]; then
	CAPTURE_SAVE_DIR="${ALLSKY_TMP}"
else
	CAPTURE_SAVE_DIR="${IMG_DIR}"
fi

CAMERA_SETTINGS="${CAMERA_SETTINGS_DIR}/settings_${CAMERA}.json"

# Get the name of the file the websites will look for, and split into name and extension.
FULL_FILENAME=$(jq -r '.filename' "$CAMERA_SETTINGS")
EXTENSION="${FULL_FILENAME##*.}"
FILENAME="${FULL_FILENAME%.*}"
 
# So scripts can conditionally ouput messages; DO NOT CHANGE NEXT LINES.
ALLSKY_DEBUG_LEVEL=$(jq -r '.debuglevel' "${CAMERA_SETTINGS}")

EricClaeys, 2022-06-23:
@ssgaurav2000   Ok, let's first make sure the images are being saved.  Do you see the images in the `~/allsky/images` directory?
If so, then try this:
~~~
sudo truncate -s 0 /var/log/lighttpd/error.log
~~~
then try to view the images in the WebUI.  It'll probably fail.  See if there are any messages in `/var/log/lighttpd/error.log`.  If so, post them.  Put three tildas (~~~) on a line by themselves, then paste the contents of error.log, then put 3 more tildas on a line by themselves.  The tildas cause whatever's between them to be formatted properly.

EricClaeys, 2022-07-01:
@ssgaurav2000   Any update?

EricClaeys, 2022-07-16:
Closing Issue - no response from user in 3 weeks.  The problem is most likely a configuration issue.

closed 2022-07-16 by EricClaeys

ssgaurav2000, 2022-07-16:
> Closed  #1306 as completed.

Sorry about the delay was traveling 
Hope this is closed now . 
--
Sent from myMail for Android Saturday,

ssgaurav2000, 2022-10-11:
Hello

Time lapses are automatically saved now , I didn't do nything except
changing TL script   to" true"

Thanks