AllskyTeam/allsky #1309: Versions

calberts, 2022-06-23:
Please can some tell me how versioning is working with allsky

The version in the version file is different than in the readme.md.
will the version and the readme.md be updated by every merge to master

It looks like it's not in sync. It would be handy if the version is shown in the WebGUI.

I have no idea where I am now, so mostly I look to the modification date of the files

Chris

EricClaeys, 2022-06-23:
@calberts   Chris,

With the current software we switched to releases called "vYYYY-MM-DD", which is what's in the `version` file.  We didn't fully implement this, but the next release will - the `version` file will match the README.md file and what appears in the log file.
We are only updating the current software with major bug fixes.  Everything else is being done in our "dev" branch.  When it's ready, we'll ask for testers, then merge the "dev" branch into the "master" branch and give it a new version number.
We'll then create a new "dev" branch.
I'm hoping that when we update the "master" branch with major bug fixes we also update the `version` file, probably by appending a point release of some kind.

I agree with putting the version number on the WebUI.  We also want to have it check if there's a new version, and if so, display a message.

When you created this Issue, did it present a green "Get Started" button?  Creating new Issues is supposed to display a template, but apparently didn't in your case.

Going forward, when you have a question, please add it to the Discussions link above rather than creating an Issue; we're trying to keep the Issue list to just problems so we know how much work we have to do.  Thanks.

calberts, 2022-06-23:
Eric

Thanks for the answer, 
Below the bug report, there is the possibility of opening a black request.
By the way clear answer
you are right in this in relation to. it was a time ago that I was here 

Thanks for all the excellent work for this community

Chris

EricClaeys, 2022-06-23:
@calberts   Chris,  thanks for the "Thanks ...".  I enjoy doing it, especially when people say "thanks".
I hope I can find a way to get rid of the "blank request" on the new Issue page.  I really want people to use the template.