AllskyTeam/allsky #1315: Generated timelapse is missing night images

zapatodefuego, 2022-06-26:
**Describe the bug** 

The timelapse generated at the end of the night contains roughly 1/3rd of the total captured images. Over the night, 2450 images are captured and the timelapse is configured to run at 25fps and so I would expect a total video length of 101 seconds yet the generated video is only 28seconds. Spot checking the images for the night and I can confirm that the video does contain only about 1/3rd; roughl7 the first 700 images.

The keogram _does_ contain data from every image.

I have tried running `./timelase.sh 20220625` manually and it "succeeds" without error yet the produced mp4 video remains the same length every time. The content of `timelapseTMP.txt` in the morning and after manually running the timelapse is the following every time:

```
Processed 2450 images
```

No errors are printed to the console (or from what I can tell the log either) while manually running the timelapse.

**Desktop:** 

 - OS: Bullseye 
 - Allsky version: v2022.03.01

**Additional context** 

Using RPiHQ with the following configuration settings to reduce the processing load and file size:
```
### Images
IMG_WIDTH=2028
IMG_HEIGHT=1520

### Timelapse
TIMELAPSEWIDTH=1920
TIMELAPSEHEIGHT=1080

# Bitrate of the timelapse video.  Higher numbers will produce higher quality but larger files.
TIMELAPSE_BITRATE="2000k"

[allsky.log](https://github.com/thomasjacquin/allsky/files/8987366/allsky.log)

**Log file** 

Attached. No errors other than startup messages (I restarted a number of time during that day getting things tested).

zapatodefuego, 2022-06-26:
I noticed `IMG_RESIZE="false"` and so I've set that to true just now after realizing the images were full resolution on disk.

EricClaeys, 2022-06-27:
@zapatodefuego Does it do this every night, or just June 25?  Did you change anything that impacts the image size on June 25?  The keogram and startrails programs don't like it when images are different resolutions; I'm not sure if the timelapse program is the same way.
If you did not make any changes, set `FFLOG="info"` and `KEEP_SEQUENCE="true"` in config.sh, then run `timeapse.sh` again.  Look at the `timelapseTMP.txt` file for any additional information and look in allsky/tmp/sequence-20220625 to see if there are 2450 files.
Another thing to try: determine the last frame added to the video (or close to the last), then `mkdir ~/allsky/images/test` and move the last image plus a thousand or more from 20220625 to `test` then `timelapse.sh test` and see what happens.  That will tell us if it was stopping due to a corrupt image.

Since you can reproduce this we'll be able to fix it, it may just take some effort.

zapatodefuego, 2022-06-27:
Thanks for the response. There are indeed 2450 files in the sequence directory but looking the files closer with `ls -alh` reveals that there's a file of size 0 exactly where the timelapse stops. Removing this file and running `timelapse.sh` again results in success.

I'm closing this though perhaps there's a bug/feature request such that `timelapseTMP.txt` doesn't print the number of files the script intended to process but rather captures how many it did? Not sure if that information is possible to extract from ffmpeg.

Thank you for your help!.

closed 2022-06-27 by zapatodefuego

EricClaeys, 2022-06-27:
@zapatodefuego I don't think that info is available, or it it is, that it's easily parseable.  I would have expected it to issue a warning message but instead it just stopped.

Did you have `REMOVE_BAD_IMAGES="true"`  I assume not since it would have caught the corrupt file.  If you don't like it removing too dark or too light files, adjust the thresholds instead of turning it off.

It would be useful to search for the name of the empty file in the log to (hopefully) see what happened.

zapatodefuego, 2022-06-27:
I do have `REMOVE_BAD_IMAGES="true"` actually. I was messing with settings yesterday have have managed to reproduce this; that is:

1. Be an idiot and toggle settings on and off with enough frequency such that you interrupt an image (either in acquisition or during saving I would think), or add an invalid image somewhere into the directory
2. Observe the generated timelapse stopping at that image
3. Remove the invalid image and re-run the timelapse

While I do have the invalid image with filesize 0 I can't attach it as is - github reports "The file is empty. Try again with a file that's not empty".  I was able to add it to a zip and upload that however:

[image-20220626145604.zip](https://github.com/thomasjacquin/allsky/files/8993243/image-20220626145604.zip)

You'll note that it really is just an empty file. `hd` and `cat` on the file prints nothing.

zapatodefuego, 2022-06-27:
`grep 20220626145604 allsky*` doesn't return anything, so my log level must not have been high enough.

EricClaeys, 2022-06-28:
@zapatodefuego can you attach your log file?  I'd like to see what happened before and after that.