AllskyTeam/allsky #1320: RPi HQ Cam installed, but no images captured

blacc01, 2022-06-30:
Hi,

Today i switched from my old ZWO ASI120MC camera to a new RPiHQ Camera. On RPi the camera is working i got image on the `libcamera-hello` command. Also changed the settings to `CAMERA="RPiHQ"`  Also changed som parameters in `Camera settings`, like Longitude, Altitude, etc. After restarting everyring, it says that "AllSky Software is starting up", and thats all. 

Here the log in debug level 4:
[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/9019069/allsky.log.txt)

And the config file:
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/9019073/config.sh.txt)

And the git log:
~~~
pi@allsky:/etc/raspap $ cd ~/allsky; git log | head
commit fea3f2e03992b79e95b0845532482ce11af285ff
Merge: 861e464 a6da0cf
Author: EricClaeys <83164203+EricClaeys@users.noreply.github.com>
Date:   Sun May 29 20:48:51 2022 -0700

    Merge pull request #1214 from thomasjacquin/postData.sh-bug-fix

    postData.sh: Bug fix: Allow to be run by WebUI

commit 861e464c237a1a8667ed8ad95b29d4fc1f19f6e6
~~~

blacc01, 2022-06-30:
I found the following line in the log:
~~~
Jun 30 12:38:09 allsky allsky.sh[10137]: removeBadImages.sh: File is bad: removed 'image-20220630123804.jpg' (below threshold: MEAN=.42577400, threshold = 1)
~~~

I changed the following value from 1 to 0.2:
`REMOVE_BAD_IMAGES_THRESHOLD_LOW=0.2 `

Now image is captured, but its extremely dark (In the middle its a lamp, because camera is now in my living room, but the light inside is ok):
[image]

Here the settings:
[image]

[image]

What should i change, or what is the problem? I think because the images are too dark, this could be the reason that those images were removed by `REMOVE_BAD_IMAGES_THRESHOLD_LOW=1` setting.

EricClaeys, 2022-06-30:
@blacc01   What is your daytime Manual Exposure set to?  You may need to enable "Advanced settings" to see it.  The "Max Auto-Exposure" is actually ignored (that's a bug) and the "Manual Exposure" is the max.  I suspect you have "Manual Exposure" set to 0.5, because all your images are 500 us (0.5 ms).  Try increasing that to 10000 or whatever.

With a 500 us exposure indoors, the images will be very dark as you noticed, and are getting deleted (which you would normally want).  Since you are indoors you'll need a longer exposure than outdoors (500 us might be a good outdoor number).

EricClaeys, 2022-06-30:
@blacc01   The next version of Allsky fixes the issue with Max Auto-Gain being ignored and Manual Gain being the max.

blacc01, 2022-06-30:
@EricClaeys here the actual settings:
[image]

Ok i didn't know that Auto-Exposure on Daytime is not working, only manual. :)

Now its better, but the colors.... 
[image]

When i switch on White balance, i get only full white images

EricClaeys, 2022-07-01:
@blacc01   Indoors with an incandescent light you'll get an orange image because those types of bulbs give off orange/yellow light.  The TV's colors looks pretty good which tells me the camera is working fine.  Your **Red Balance** is also higher than your **Blue Balance** making the image even redder.  Unless you're going to keep the camera indoors, I wouldn't do anything until you have it outside.

What happens if you turn on Auto White Balance and wait a while?  The auto-exposure/gain algorithm should eventually bring the brightness to 0.5 (your daytime **Mean Target** value).  I'm surprised the image went pure white and yet the exposure time was only 3% of the Auto White Balance off picture above it.

blacc01, 2022-07-01:
@EricClaeys thanks, than for now i'll not touch those settings. I hope tomorrow i'll have the time and i can install the camera to its final position. After that we will see. ;)

blacc01, 2022-07-02:
@EricClaeys I installed the camera on its final position. But the image is quite blue now. :(
[image]

blacc01, 2022-07-02:
@EricClaeys okay i could get a better daytime picture now, please check, if its ok now:
[image]

Here the settings:
[settings_RPiHQ.json.txt](https://github.com/thomasjacquin/allsky/files/9033042/settings_RPiHQ.json.txt)

I needed to change:
Daytime settings:
Mean target to 0.2
Brightness to 10

Both daytime and nighttime settings:
Saturation settings was changed from 50 to 1 as my system is Bullseye, that was the reason i got full white image. :)
Auto White balance is now on
Red and Blue Balance was set to 0.1 

I don't know if those settings will be ok or should change to different values.
Also i hope the camera focus is set correctly, will see at night.

EricClaeys, 2022-07-02:
@blacc01   That picture looks pretty good to me, but it's more important what you think.  If you like it that's really all that matters.

The next version of Allsky will have defaults based on Buster versus Bullseye so situations like you had with Saturation shouldn't happen.

Some people use clouds to focus during the day, but if you have stars at night that's even better.  Try for the highest "Focus" value you can get, but looking at the stars zoomed in is even better.  After you've achieved focus you may want to disable the **Focus Metric** in the WebUI so it doesn't appear in the images.

blacc01, 2022-07-02:
@EricClaeys thanks! As we do not have here a single cloud all the day, i have to rely at the stars at the night. What value is good for focus? What should i achieve?

blacc01, 2022-07-02:
@EricClaeys untill waiting for the nightsky, another question.
I filled my website datas, and swithced on "Show on Map" function.
Here how i filled it:
[image]

But on the map, it does not show the jpeg image, only the folder. which i wrote to the "Website URL" line:
[image]

If i write the same to the "Website URL" as in "Image URL", did not help. Is it a bug, or what should i do?
The link http://mcllserver.gotdns.com/allsky/image.jpg is ok, the updated image is there.

blacc01, 2022-07-02:
@EricClaeys ok, so as i see something has to be change in the config, as the stars are not shown correctly. As i see focus is ok, but something is not okay. Could you help me what i shoud change?

Here a sample image:
[image]

My settings:
[settings_RPiHQ.json_2.txt](https://github.com/thomasjacquin/allsky/files/9033737/settings_RPiHQ.json_2.txt)

blacc01, 2022-07-02:
I covered the light at the bottom, a little bit better, but not ok. :(
[image]

EricClaeys, 2022-07-02:
@blacc01.   The focus number is different depending on what camera, lens, and dome you have, as well as brightness, and may change night-to-night.  Your best number will probably be quite different than mine, so just try to get the highest number you can.  I think the software is just reporting the contrast, so it's nothing intelligent. 

The map server uses https and your site uses the less secure http, and many (most ?) browsers don't display http images on an https site for security reasons.  It's a browser issue,  not an allsky issue.  Thomas is seeing if there is anything he can do on the server, but I don't know the status of that.  If you can use https on your server that will fix it.
I just looked on the map and don't see your pin.  Did you turn it off in the WebUI?

What is not ok in your star picture?  There might be dirt or remains of water drops on the dome (the out of focus spots), and in the first one especially at the top,  you are getting reflections from the camera.  Try painting everything under the dome flat black, or put black felt there.  It's pretty much trial and error.  I got rid of most of my reflections, but when I removed the camera a while ago and put it back, I didn't put it all the way is so there is some reflection again.

blacc01, 2022-07-03:
@EricClaeys the dome was clear, all what there was a reflection from the lamp bottom of the picture. I blocked that region with positioning brick on the camera housing. It was much better, but still not enough ti get strartrails or even to see clearly the stars. I don't know what parameters should i change. :(
[image]

EricClaeys, 2022-07-03:
@blacc01 your stars look pretty good to me.  With libcamera, the brightness goes from -1.0 to 1.0 with 0.0 as the default.  Tr` page in the Wiki for the settingy playing with the brightness.  You can also try stretching the image via the option in config.sh.  I have never used that option yet so don't have any advice.

For startrails look in the `allsky settings` Wiki page.  It describes how to adjust sensitivity.

blacc01, 2022-07-03:
@EricClaeys thanks. So then i should change Brightness to 0? Now its 10. You said "With libcamera, the brightness goes from -1.0 to 1.0 with 0.0 as the default".

EricClaeys, 2022-07-03:
@blacc01 just the night brightness. Play with it.

blacc01, 2022-07-03:
Thanks!! Tried cropping in config.txt, i think the image size is now much more better!
[image]

closed 2022-07-04 by blacc01