AllskyTeam/allsky #1332: Startails image does not appear on the website

horatti, 2022-07-08:
Hi everyone!

I updated to the latest "allsky" version (07/02/2022) and everything works perfectly, except for one.
The current starttails image does not appear on my website after the end of the night. Keogram and timelapse video are OK.
The image is generated, but there is a problem with uploading, so I tested it manually, here is the result:

"pi@allsky:~/allsky/scripts $ ./uploadForDay.sh  -s 20220706
cd ok, cwd=/                                                 
1535750 bytes transferred                                   
cd ok, cwd=/                                                 
1535750 bytes transferred                                   
lftp OUTPUT from '/home/pi/allsky/images/20220706/startrails/startrails-20220706.jpg:
   rm of /allsky/startrails/ failed!  Trying again...
rm: Access failed: 550 Could not delete allsky/startrails/: Invalid argument (/allsky/startrails/)"

I have no idea what is causing this error, since everything is unchanged on the hosting, where my website is.

I welcome any ideas: Ati

EricClaeys, 2022-07-09:
@horatti  Did you look at the [Troubleshooting -> Uploads](https://github.com/thomasjacquin/allsky/wiki/Troubleshooting-uploads) Wiki page?  It describes what to do.

When you created this Issue, did you see a green **Get started** button?  Clicking it brings up a template, which asks that if you have a question to enter a Discussion entry, not an Issue.

horatti, 2022-07-09:
Hi Eric!
You're right, I got a little carried away!
:-(
I haven't exhausted all possibilities yet, I'm still looking for the cause of the phenomenon.

Thanks, Ati

closed 2022-07-09 by horatti

EricClaeys, 2022-07-09:
@horatti If you have any suggestions on how to improve the Wiki, let me know.  For example, if the problem you are seeing isn't listed, or is listed but isn't clear.

Eric