AllskyTeam/allsky #1345: CPU to hot 78+ Celcius

cboivin11, 2022-07-11:
Hi
my CPU is always in the 77+ celcius and it is only 21C outside
is this normal, 
if not what can i do to cool this cpu

I,m using a POE to  power the zwo178 cam

Thank you

Claude

EricClaeys, 2022-07-12:
@cboivin11.  Claude,
When you created this issue was there a template that asked some questions, and requested that questions be entered via the `Discussions` link?

cboivin11, 2022-07-14:
No,  I didnt know where to put this question ?

EricClaeys, 2022-07-14:
@cboivin11    Interesting.  The code I added when entering a new Issue must not have worked.
Anyway, questions and feature requests should go in the `Discussions` link at the top of the page.  I'm trying to limit the Issues list to true bugs so we know how much work is needed.

Is your CPU hot (you can look at the `System` page of the WebUI to determine CPU temperature), or is the camera sensor hot (look at an image)?

Take a look at the `Can the end user get the camera sensor by a script?` Discussion.  You can also try the Allsky Facebook group to see if anyone there has any other suggestions.

I'm closing this Issue.  If your CPU is hot, please add a new Discussion item.  If the Sensor is hot, add to the Discussion above.

closed 2022-07-14 by EricClaeys

cboivin11, 2022-07-14:
Eric

 

Yes i look in the system of the allsky page and it is over 80C and the image
temp goes up to more the 60C

Just want to know if this is too hot ?  because I think it is

 

Claude 

 

<https://github.com/thomasjacquin/allsky/issues/1345#issuecomment-1184547754
AS25ANCNFSM53JGBYPA> .
<https://github.com/notifications/beacon/ALDUPFOBCMXDZY66CG35ZADVUAS25A5CNFS
M53JGBYPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOI2NMHKQ

EricClaeys, 2022-07-15:
@cboivin11    A CPU temp of 80 is pretty hot but the Pi will throttle itself when it gets too hot.  Look on the `System` page - it will say if it's is currently throttling or has since the Pi was booted.
My CPU temp is currently 56 and the camera temp is 26.  It's 10 pm and cooler than during the day.

Looks like it's time to get a fan.