AllskyTeam/allsky #1347: Update from 0.6 to 0.8.3, ASI camera not working

brianboru82, 2022-07-13:
I just updated my setup from an old 0.6 version that was working okay, to the current version. I followed the instructions for upgrading from an old version, and everything seemed to work fine for the install process, however now the system is not taking photos at all and is erroring out. Not quite sure what is wrong. The log file shows 

> Jul 12 21:48:32 allsky allsky.sh[3088]: WARNING: ASISetControlCaps() for control 14, value=0 failed: ASI_ERROR_GENERAL_ERROR (16)
> Jul 12 21:48:32 allsky allsky.sh[3088]: NOTICE: Camera does not support ControlCap # 2; not setting to 50.
> Jul 12 21:48:32 allsky allsky.sh[3088]: Will adjust gain at transitions
> Jul 12 21:48:32 allsky allsky.sh[3088]: *** Stop the allsky service to end this process. ***
> Jul 12 21:48:32 allsky allsky.sh[3088]: *** ERROR: it's not DAY or NIGHT, it's ''
> Jul 12 21:48:33 allsky allsky.sh[3088]: copy_notification_image.sh: Uploading NotRunning.jpg
> Jul 12 21:48:34 allsky allsky.sh[3088]: upload.sh: FTP '/home/pi/allsky/tmp/image.jpg' finished
> Jul 12 21:48:35 allsky allsky.sh[3088]:      ***** Stopping AllSky *****
> Jul 12 21:48:35 allsky allsky.sh[3088]: 'capture' exited with RETCODE=100
> Jul 12 21:48:35 allsky allsky.sh[3088]: ***
> Jul 12 21:48:35 allsky allsky.sh[3088]: *** After fixing, restart the allsky service. ***

Here's the header output, the log, and config files. Thank you for the help!

> commit fea3f2e03992b79e95b0845532482ce11af285ff
> Merge: 861e464 a6da0cf
> Author: EricClaeys <83164203+EricClaeys@users.noreply.github.com>
> Date:   Sun May 29 20:48:51 2022 -0700
> 
>     Merge pull request #1214 from thomasjacquin/postData.sh-bug-fix
>     
>     postData.sh: Bug fix: Allow to be run by WebUI
> 
> commit 861e464c237a1a8667ed8ad95b29d4fc1f19f6e6
[allsky.log.txt.log](https://github.com/thomasjacquin/allsky/files/9099021/allsky.log.txt.log)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/9099022/config.sh.txt)

brianboru82, 2022-07-13:
Log file re-attached without extra .log on the end.

[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/9099031/allsky.log.txt)

EricClaeys, 2022-07-13:
@brianboru82   Some version of the Operating System changed the version of `sunwait` and that impacted Allsky.
Do the following
~~~
cd
which sunwait
# copy src/sunwait to the location returned from the above command, e.g.,
cp src/sunwait /xxxxxxx
~~~

That should fix the issue.

When you created this Issue, did GitHub display some instructions on what to enter, and other information?

brianboru82, 2022-07-13:
Thanks @EricClaeys that makes sense, I will try that out this evening.

I was just trying to follow the instructions at https://github.com/thomasjacquin/allsky/wiki/Reporting-Issues when I created the issue.

EricClaeys, 2022-07-13:
@claudejaspart.   Claude, what solution did you try?  The one I posted above?
If so, would you please do an `ls -l ~/allsky/src/sunwait` and another `ls -l` on the sunwait returned by the `which` command?

Thanks for following the instructions on the Reporting Issues Wiki page.  I recently changed the Issues page to display some instructions and request certain information, but that's not in your Issue so I am not sure if the Issue change worked or not.
Do you recall seeing the items I mentioned?

Eric

brianboru82, 2022-07-14:
Thanks @EricClaeys that did work. I had to slightly change the command you gave as follows, but minor issue.
`pi@allsky:~/allsky $ sudo cp src/sunwait /usr/local/bin/sunwait`

When I created the issue I didn't see any extra instructions. I just tested creating a new issue, and I see those instructions now, perhaps they hadn't been pushed when I created this one. Anyway, thank you very much!

closed 2022-07-14 by brianboru82

EricClaeys, 2022-07-14:
@brianboru82    Sorry about that.  I should have remembered it needed `sudo`, but glad you figured it out on your own.
The sunwait issue is fixed in the next Allsky release so users won't run into the problem.

Glad you have things working.  If you ever have a suggestion for improving Allsky or the Wiki, please enter a Discussion item.

brianboru82, 2022-07-14:
Wasn't meaning to complain, I really appreciate all your help! Just wanted to write it out in case someone else had the same issue.

Cheers!

claudejaspart, 2022-07-14:
Thanks for the help :)

EricClaeys, 2022-07-15:
@brianboru82   It never crossed my mind that you may have been complaining.  I took it the way you meant it, and appreciate it.

@claudejaspart  Claude, do you have it working now as well?

claudejaspart, 2022-07-15:
@EricClaeys Yes, thank you !