AllskyTeam/allsky #1460: ERROR: it's not DAY or NIGHT, it's ''

garrington, 2022-08-02:
### Environment

* Camera:  ZWO ASI120MM-S
* OS: Buster
* Allsky version: v2022.03.01
* Pi 4, 2GB RAM

### Bug Description

This system was updated, following the procedure from the upgrade page.  This resulted in the below error.
Following this issue, allsky was uninstalled using the uninstall.sh script and the /home/pi/allsky dir, /var/www/html/* and /etc/raspap/* were all deleted.  It was then installed from a fresh clone of the repo, the result was the same.   The error seems to suggest that sunwait is not returning a valid/expected response, but when attempted manually at the CLI it seems to work fine.

Error is as follows (excerpt from allsky.sh output):
~~~
ZWO ASI120MM-S Information:
  - Native Resolution: 1280x960
  - Pixel Size: 3.8microns
  - Supported Bins: 1 2
  - Mono camera

  - Camera ID: [none]
  - Camera Serial Number: 2034d30111090900
- Sensor temperature: 29.70
  - Current camera mode: NORMAL (0)

Capture Settings:
 Image Type: RAW8
 Resolution (before any binning): 1280x960
 Quality: 95
 Daytime capture: Yes
 Exposure (day): 0.500ms, Auto: Yes
 Exposure (night): 10,000ms, Auto: Yes
 Max Auto-Exposure (day): 60,000ms (60.0s)
 Max Auto-Exposure (night): 20,000ms (20.0s)
 Delay (day): 5,000ms
 Delay (night): 10ms
 Gain (night only): 50, Auto: No, max: 200
 Gain Transition Time: 15.0 minutes
 Brightness (day): 50
 Brightness (night): 50
 Skip Frames (day): 5
 Skip Frames (night): 1
 Aggression: 75%
 Gamma: 50
 Binning (day): 1
 Binning (night): 1
 USB Speed: 80, auto: Yes
 Text Overlay: [none]
 Text Extra File: [none], Age: 600 seconds
 Text Line Height 60px
 Text Position: 15px from left, 30px from top
 Font Name: SIMPLEX (0)
 Font Color: 255, 255, 255
 Small Font Color: 0, 0, 255
 Font Line Type: 16
 Font Size: 7.0
 Font Line Width: 1
 Outline Font : No
 Flip Image: 0
 Filename: image.jpg
 Filename Save Directory: /home/pi/allsky/tmp
 Latitude: 60.7N, Longitude: 135.05W
 Sun Elevation: -6
 Locale: en_US.UTF-8
 Notification Images: Yes
 Histogram Box: 500 500 50 50, center: 640x480, upper left: 390x230, lower right: 890x730
 Show Histogram Box: No
 Show Histogram Mean: No
 Show Time: Yes (format: %Y%m%d %H:%M:%S)
 Show Temperature: Yes, type: C
 Show Exposure: Yes
 Show Gain: Yes
 Show Brightness: No
 Preview: No
 Taking Dark Frames: No
 Debug Level: 1
 On TTY: Yes
 Video OFF Between Images: Yes
 ZWO SDK version 1, 20

WARNING: ASISetControlCaps() for control 14, value=0 failed: ASI_ERROR_GENERAL_ERROR (16)
*** Press Ctrl+C to stop ***

*** ERROR: it's not DAY or NIGHT, it's ''
     ***** Stopping AllSky *****
'capture' exited with RETCODE=100
***
*** After fixing, restart allsky.sh. ***
***
     ***** AllSky Stopped *****
~~~

### Log / configuration files

configs are all default at this point, matches the repo

EricClaeys, 2022-08-02:
@garrington   Garey,
Please do the following:
~~~shell
cd ~/allsky
which sunwait
sudo cp ./sunwait THE_LOCATION_which_RETURNED
~~~

"sunwait" changed in Bullseye and no longer returns "DAY" or "NIGHT".
This is fixed in the next version of Allsky.

garrington, 2022-08-03:
There is no sunwait binary in ~/allsky/

garrington, 2022-08-03:
Based on this I got it working by installing sunwait via snapd and adding it to the path.  Version 0.6 was installed and did not work, but version 0.8 seems to be working fine.  I had to remove the older sunwait binary.

garrington, 2022-08-03:
Well... it was until I ran the gui install script, now it is well and thoroughly hosed again.

EricClaeys, 2022-08-03:
@garrington   Sorry, the sunwait binary you want to copy from is in `~/allsky/src/sunwait` so the corrected instructions are:
~~~shell
cd ~/allsky/src
which sunwait
sudo cp ./sunwait THE_LOCATION_which_RETURNED
~~~

BTW, what did `which` return?

garrington, 2022-08-03:
~~~
$ which sunwait
/usr/local/bin/sunwait
~~~

This corrected the issue.

EricClaeys, 2022-08-04:
Closing Issue as the suggested workaround fixed the problem.

closed 2022-08-04 by EricClaeys