AllskyTeam/allsky #1513: [BUG] WARNING: Value of 100 greater than max value allowed (20) for control 'Offset' (#5)     found in allsky.log

RabensteinadP, 2022-08-08:
### Environment

* Camera:  ZWO (including model)  ASI120 Firmware on a T7C
* OS: Buster or Bullseye (Sorry Guys, I´m very new at Raspberry and do not know this
* Allsky version:
     * The newest software includes a file called  ~/allsky/version v0.8.3.3
     * If it does not exist, run:       grep Software /var/log/allsky.log
* Pi model and amount of memory (512 MB, 1, 2, 4, or 8 GB) Pi 4B 4GB

### Bug Description

* I could not find the 'Offset' to set it to 20 or lower. This is suddenly my Problem before ASI_Error_Timout occours.
* Can it be reproduced?  If so, how? Yes, every time Nighttime shooting starts. After Rebooting I have a wrong picture sometimes
* Did anything change?  Any settings?  Most issues occur after a change was made. I have it absolutly new. Prevous version (1 year) worked good bevore my SD-Card was killed.
* Please include the exact messages or a screenshot. Exact message is the title of the issue. It ocours in allsky.log just bevore ASI_Error_Timeout message.
* If this is a configuration-related issue (or you are not sure), attach these files, appending ".txt" to their names first:
 I´m sure the only problem is to find the entry for offset.

### Log / configuration files

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.
Sorry, just learning Pi. I don´t know how to get a file to my PC to paste it here.

EricClaeys, 2022-08-08:
@RabensteinadP,  You can safely ignore this message.  It's only a warning and the offset was set to the maximum.  This issue is also fixed in the next release.
In the WebUI, click on the "Camera Settings" link on the left side.  At the bottom of the page, click on "Show Advanced Options".  This will add a bunch of "advanced" options that aren't used much.  `Offset` is one of them.  You can change the value there.  I suggest setting it to 0.  At 20, your pictures might have a washed-out look, lacking in contrast.

Until you figure out how to copy files to/from the Pi to another machine, the easiest way to upload the log file is to open the web browser on the Pi and log into GitHub, then you can drag/drop the file into the bottom of your reply.

If this solves the problem, please close the Issue in GitHub.

EricClaeys, 2022-08-14:
@RabensteinadP    Any update?

RabensteinadP, 2022-08-17:
Since I looked for the Offset that wasn´t in the advanced Options with the ZWO-Camera Software the Cam is running near perfekt. The only issue is actually a "Pause" from about 30s between the Pictures with 20s Exposure. So I kicked some settings and the website but i missed many Meteors right now.

closed 2022-08-17 by RabensteinadP