AllskyTeam/allsky #1573: 0.8 Exposure has a 21-22s delay between Pics. 0.7 Exposure runs, but sometimes skips to a split Picture until reboot

RabensteinadP, 2022-08-21:
### Environment

* Camera:  T7C with ZWO-Firmware
* OS: Bullseye
* Allsky version:
     * The newest software includes a file called  ~/allsky/version v2022.03.01
     
* Pi model and amount of memory Pi 4B 4GB

### Bug Description

* Include a clear and concise description of what the bug is. A Massiv delay in Night-Mode with Exposure 0.8 occurs. for a 20s Photo there is a delay of 21 to 22s. The old Version don´t give me a correct Pic with 50% Chance to get it correctly after reboot. (Seems better if the Temperatures are medium instead of nearly 60°C
* Can it be reproduced?  If so, how? Nightmode with 0.8 exosure changes the delay between Photos very extreme.
* Did anything change?  Any settings?  Most issues occur after a change was made. Older Version worked. Actual Version is a Problem.
* Please include the exact messages or a screenshot. No Message in expsure 0.8. Splittet Pics with 0.7 
[image]

* If this is a configuration-related issue (or you are not sure), attach these files, appending ".txt" to their names first:
  * Allsky or WebUI problems
    * ~/allsky/config/config.sh I dont´t think so.
    * /etc/raspap/settings_ZWO.json: Atached
    * ~/allsky/config/ftp-settings.sh (upload-related problems): ftp turned off for testing Performance of delay.
  * Allsky Website problems: Website kicked for Performancetesting of delay
    * /var/www/html/allsky/config.js (Allsky Website problems)

If you have program output or multi-line messages to include,
add it like this so it formats correctly:

~~~
output/message line 1
output/message line 2
~~~

### Log / configuration files

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.
[settings_ZWO.json.txt](https://github.com/thomasjacquin/allsky/files/9389852/settings_ZWO.json.txt)

EricClaeys, 2022-08-21:
@RabensteinadP Is the delay IN ADDITION TO the delay you specified (19,970 day and 0 sec at night)?

Do you have an example of a "split" image you sometimes get with **Version 0.8 Exposure Mode** off?

Your settings look fine so I'll need a copy of the log file.  Please follow the instructions in the Wiki page for Reporting Issues.

RabensteinadP, 2022-08-23:
No! The delay is not what i specified. I have split-Pics only in 0.7 and massive delay only in 0.8. The cam is running the last 2,5 days without problems in exposure 0.7 (I don´t want to do a reboot actually because its a game to get the full pic unsplitted) The log is here. So I´m sorry for this long version. On Top you can see the 0.8 exposure Method with Warnings in the file. At bottom you can see the 0.7 exposure (actually running without porblems)
[allsky.log](https://github.com/thomasjacquin/allsky/files/9409427/allsky.log)

RabensteinadP, 2022-08-24:
Today i had the "Too many ASI Errors"-Error and so a Reboot was necesary. The CPU was Throttling and had over 60°C. After 3 or 4 reboots i had the splittet Pic in exposure 0.7 and also the massive delay in exposure 0.8.   I wait until the CPU-Temp goes down. I think my Problems occurs with high Temperatures. So the exposure 0.8 isn´t helping to cool down the system because of delay. The exposure 0.7 now i think leaks with high temperatures. So i start to cool the system an get a look at the Problems again with cooler Temps.  Now i was able to generate a short allsky.log to and so here it is
[allsky with ASI Error Timeout exposure07.txt](https://github.com/thomasjacquin/allsky/files/9419415/allsky.with.ASI.Error.Timeout.exposure07.txt)
[image]
.

EricClaeys, 2022-08-25:
@RabensteinadP    Is the picture above a "split" picture?  I can't tell what it is (a wall with a doorway?)  What's wrong with it?  There are a lot of hot pixels, which is very common with the 120 camera.
Can you post a good picture and a bad picture so I can see what you mean by "split"?

Please try setting the **USB Bandwidth** to 50.  For _some_ people it solved the ASI_ERROR_TIMEOUT messages and banding in images.  Also try setting Auto USB off.
The problems you are seeing in Version 0.8 mode are that the software is asking for a 20 second exposure but the camera is taking about 40 seconds to take the exposure.  That's the cause of the WARNING messages in the log file and the long delay you see.  This can be caused by the USB being the wrong value.  Unfortunately it's trial and error to get the right value.

When there are delays between exposures, Version 0.8 exposure mode typically results in a 10 degree drop in temperature compared to pre-version 0.8.

FYI, Allsky doesn't currently accept "," in a number, so your daytime exposure of **0,2** is interpreted as **0**.  Change it to **0.2**

RabensteinadP, 2022-08-25:
The roof of the hous marks the left end normaly like here:
[image]
.

The thing I mean with split is often it looks like you take a scissor an puzzle a bit with two parts. So i called it split-Pic
[image]
.

I just turned now the USB-Bandwith to 50 and thanks for the FIY with "," and "."

RabensteinadP, 2022-08-25:
The really Interesting Thing: In Exposure 0.8 everytime the image is okay. In Exposure 0.7 It´s really try and error until i get a correct picture. I´m switching the exposure until i have a correct 0.7 Picture.

EricClaeys, 2022-08-26:
@RabensteinadP   Have you tried a lower USB value yet?  The split image can also be caused by USB problems.
Have you ever had this camera working with any version of Allsky?

Other things you can try is to switch USB ports on the Pi - try USB 2 and USB 3 ports.

A fair number of people with the ASI 120 have replaced it with a better camera - not just because of the hot pixels, but also because of problems like you're seeing.

At the bottom of the [Wiki page](https://github.com/thomasjacquin/allsky/wiki/Troubleshoot-ASI_ERROR_TIMEOUTs) on troubleshooting ASI_ERROR_TIMEOUTs area couple other things to try.  See the items after the "IGNORE THESE NOTES FOR NOW" heading.

RabensteinadP, 2022-08-26:
The Version I used in 2021 was running all the time with this cam. Yes I want a better cam but I don´t want the problems with a better one. What cam I should prefer? I try the Bandwith 50 yet. It seems better for the last 1h.  I´m reading the Wiki page for now.

Off Topic: Darks are not made yet, and I think a badPixel-map is a good idea. I use this in a Stacking-Software (AstroPixelProcessor) and it is a wonderful thing. I just try this with the allsky pics. :-)

EricClaeys, 2022-08-29:
@RabensteinadP I have been suggesting the RPi HQ camera to people who either don't have a camera or want to upgrade.  It's less expensive than the ZWO cameras and has good quality and resolution.  And it doesn't use USB so avoids all those problems.

Darks and bad pixel maps service the same purpose.  I think for an allsky camera, darks are probably better.

Let me know how it goes with changing USB.

EricClaeys, 2022-10-06:
@RabensteinadP    Any update?

EricClaeys, 2022-10-31:
Closing - no response from user since end of August.

closed 2022-10-31 by EricClaeys