AllskyTeam/allsky #1596: Timelapse generation failure, nothing useful in log

keenerb, 2022-08-29:
root@allsky:/var/log# /home/pi/allsky/scripts/timelapse.sh 20220828

*** : ERROR: ffmpeg failed.
Error log is in '/home/pi/allsky/tmp/timelapseTMP.txt'.

Links in '/home/pi/allsky/tmp/sequence-20220828' left for debugging.
Remove them when the problem is fixed.

timelapseTMP.txt:

Processed 1989 images
/home/pi/allsky/tmp/timelapseTMP.txt (END)

keenerb, 2022-08-29:
I grabbed the ffmpeg command out of the script:

pi@allsky:~/allsky/tmp $ ffmpeg -y -f image2     -loglevel warning       -r 25   -i /home/pi/allsky/tmp/sequence-20220828/%04d.jpg       -vcodec libx264     -b:v 2000k       -pix_fmt yuv420p        -movflags +faststart                    /home/pi/allsky/images/20220828/allsky-20220828.mp4
[swscaler @ 0x19bed60] deprecated pixel format used, make sure you did set range correctly
Killed

Not sure what "Killed" the ffmpeg process???

keenerb, 2022-08-29:
I think this is memory related, only have a 2gb rpi4..

keenerb, 2022-08-29:
It was two issues:  I needed to increase swap file size.  Once I increased swap file size, I encountered the following error:

https://github.com/thomasjacquin/allsky/issues/249

Updating the timelapse photo file size in the config fixed this it seems.

closed 2022-08-29 by keenerb

EricClaeys, 2022-08-29:
@keenerb   I'm glad you got it fixed.
In the future, please look in the Wiki before entering an Issue.  In this case, the "Troubleshooting -> Timelapse" page describes this error and what to do to fix it.

When you created this Issue, did you see a page with several questions on it?