AllskyTeam/allsky #1717: [BUG]

ccdanieldb, 2022-10-07:
~/allsky/version

### Environment

* Camera:  RPi
* OS: Bullseye
* Allsky version: v2022.03.01
* Pi model and amount of memory (rpi4, 8GB)

### Bug Description

* Time-lapse files not being completed and remain in temp file
* If I run `./timelapse.sh 20221006` I get the below error

```
pi@localhost:~/allsky/scripts $ ./timelapse.sh 20221006

*** : ERROR: ffmpeg failed.
Error log is in '/home/pi/allsky/tmp/timelapseTMP.txt'.

Links in '/home/pi/allsky/tmp/sequence-20221006' left for debugging.
Remove them when the problem is fixed.
```

* This is a new install about 3 days old.

* ~/allsky/config/config.sh

```

#!/bin/bash

# NOTES:
#  - Variable names must start at the beginning of a line.
#  - There should be nothing after a variable's value, i.e., no comments on the same line.
#  - Values can optionally be quoted with double quotes, NOT single quotes.
#  - If a variable occurs multiple times, only the last value is used.

# Choose between "ZWO" or "RPiHQ".
CAMERA="RPiHQ"

########## Images
# Set to "true" to upload the current image to your website.
IMG_UPLOAD="false"

# If IMG_UPLOAD is "true", upload images every IMG_UPLOAD_FREQUENCY frames, e.g., every 5 frames.
# 1 uploades every frame.
IMG_UPLOAD_FREQUENCY=1

# The websites look in IMG_DIR for the current image.
# When using the Allsky website from the "allsky-website" packge,
# "current" is an alias for "/home/pi/allsky".
# If you use the default IMG_DIR and have the Allsky website, set "imageName" in
# /var/www/html/allsky/config.js to:
#     imageName: "/current/tmp/image.jpg",
# This avoids copying the image to the website.
# Only change IMG_DIR if you know what you are doing.
IMG_DIR="current/tmp"

# Resize images before cropping, stretching, and saving.
# Adjust IMG_WIDTH and IMG_HEIGHT according to your camera's sensor ratio.
IMG_RESIZE="false"
IMG_WIDTH=2028
IMG_HEIGHT=1520

# Crop images before stretching and saving.
# This is useful to remove some of the black border when using a fisheye lens.
# If you crop an image you may need to change the "Text X" and/or "Text Y" settings in the WebUI.
CROP_IMAGE="false"
CROP_WIDTH=640
[... 346 lines cut ...]
	# You may want to turn off or limit bucket versioning to avoid consuming lots of
	# space with multiple versions of the "image.jpg" files.
S3_BUCKET="allskybucket"

	# S3_ACL is set to private by default.
	# If you want to serve your uploaded files vis http(s), change S3_ACL to "public-read".  
	# You will need to ensure the S3 bucket policy is configured to allow public access to
	# objects with a public-read ACL.
	# You may need to set a CORS policy in S3 if the files are to be accessed by
	# Javascript from a different domain.
S3_ACL="private"

```
If you have program output or multi-line messages to include,
add it like this so it formats correctly:

~~~
output/message line 1
output/message line 2
~~~

### Log / configuration files

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.

EricClaeys, 2022-10-10:
@ccdanieldb   Did you check out the Wiki page on Troubleshooting Timelapse?

EricClaeys, 2022-10-31:
@ccdanieldb   Any update?

ccdanieldb, 2022-10-31:
Its still not working.  Do you have a link to the section that describes how to get the Timelapse up.  It is saving the photos just not creating the time-lapse.

EricClaeys, 2022-10-31:
Did you check out the Wiki page on Troubleshooting Timelapse?  It's helped almost everyone.

ccdanieldb, 2022-10-31:
Your right this dose look helpful.  

https://github.com/thomasjacquin/allsky/wiki/Troubleshooting-timelapse

I will take another look at this tonight and see if I have any useful questions.

EricClaeys, 2022-12-16:
Closing - no response in about 6 weeks.  As far as I know, all timelapse creation-related issues have been one of the two problems described in the Wiki page so I'm confident following the instructions there will solve the problem.

closed 2022-12-16 by EricClaeys