AllskyTeam/allsky #1879: ERROR: See log file for information /var/log/allsky.log

AlexXu12256, 2022-11-21:
Device: Raspberry pi 3b+
Camera: ZWO ASI 224MC, 120MC-S, 385MC, 662MC
Issue: The Allsky Web GUI is used, and it works with 224MC, 120MC-S, and 385MC, but when the 662MC is plugged in, it displays "ERROR: See log file for information /var/log/allsky.log". 
I have checked the allsky.log, and it says *** ERROR: No connected Camera... when the 662MC is connected. 
Please scroll down to the very bottom of the log file to see the error message. Sorry for causing any inconvenience.
Thank you very much

[allsky.log](https://github.com/thomasjacquin/allsky/files/10061073/allsky.log)

EricClaeys, 2022-11-22:
@AlexXu12256
The version of the ZWO SDK in the current release I believe is 1.20.0 and doesn't support all the new cameras.  Search for `ZWO SDK version` in the log file.
The next version of Allsky has version 1.26.0 which includes the file.  I just checked ZWO's developer download page, and it says the latest is from June and is 1.20.0, but when downloading it, the file name is 1.24.0.

If you need support for the 662MC now, unzip the attached file, then:
~~~
cd ~/allsky/src/lib/armv7
mv libASICamera2.a libASICamera2.a-OLD
~~~
then copy the unzipped file to this directory and restart Allsky.

[libASICamera2.zip](https://github.com/thomasjacquin/allsky/files/10063449/libASICamera2.zip)

AlexXu12256, 2022-11-22:
@EricClaeys 
I have tried adding the new file, but it still gives the same error. The ZWO SDK Version shown in the log file when the old cameras are connected is 1.20.0 (only shows when it works). Is the file you attached in version 1.26.0 or 1.24.0?

Thanks

AlexXu12256, 2022-11-23:
@EricClaeys 
I have tried to replace the file you attached, but it is still outputting the same error. I've tried to replace the libASICamera2.a in armv6, armv8, x64 and x86 with the files found in this link: https://github.com/OpenPHDGuiding/phd2/pull/1005/commits/3db5d555576ca862e7840976b6a661b3973d9e8a#diff-87de52c36d986481a65bc0a52f420997dbcb3abc7a46c2c5fb53e5c0cc45e521

But it still gives the same error.

AlexXu12256, 2022-11-23:
@EricClaeys 
I have tried to replace the file you attached, but it is still outputting the same error. I've tried to replace the libASICamera2.a in armv6, armv8, x64 and x86 with the files found in this link: https://github.com/OpenPHDGuiding/phd2/commit/3db5d555576ca862e7840976b6a661b3973d9e8a#diff-87de52c36d986481a65bc0a52f420997dbcb3abc7a46c2c5fb53e5c0cc45e521
But it still gives the same error.
I have checked the SDK version in the log file, and it still appears to be "1, 20" after copying those files to the directory. Does this mean I am not updating the SDK correctly?

EricClaeys, 2022-11-23:
What operating system are you on?  Buster or Bullseye?  And on 32 bit or 64 bit?
Try replacing the armv6 and armv8 files from PHP, then recompile.  I may have been wrong saying armv7 needed to be updated.  I think it depends on which operating system and 32 / 64 bits.

AlexXu12256, 2022-11-24:
@EricClaeys I'm on Bullseye, 32 bit. Did you mean replacing the armv6 and 8 files with the file you have attached? Thanks.

EricClaeys, 2022-11-24:
@AlexXu12256. No, replace the armv6 and armv8 files with the corresponding files from the PHP link you posted.

AlexXu12256, 2022-11-24:
@EricClaeys Hi there, I have tried replacing the files and the SDK version remains at 1, 20. Thanks.

EricClaeys, 2022-11-26:
@AlexXu12256    Please do the following and let me know what you get:
```
cd ~/allsky/src
make -n capture
```

This will tell me if you have armv6, armv7, or armv8.

AlexXu12256, 2022-11-27:
@EricClaeys The output says "make: 'capture' is up to date."

EricClaeys, 2022-11-28:
@AlexXu12256   do the following:
```
cd ~/allsky/src
rm capture
make -n capture
```

That should show what it will do to compile a new capture.

AlexXu12256, 2022-11-28:
@EricClaeys 
The outputs is shown below:
arm-linux-gnueabihf-g++  capture.cpp -o capture -Werror -Wall -Wno-psabi -Wno-unused-result -g -O2 -lpthread -march=armv7 -mthumb -D_LIN -D_DEBUG -DGLIBC_20 -Llib/armv7 -I./include -I/usr/include/opencv4 -lopencv_stitching -lopencv_alphamat -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_dnn_objdetect -lopencv_dnn_superres -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_hfs -lopencv_img_hash -lopencv_intensity_transform -lopencv_line_descriptor -lopencv_mcc -lopencv_quality -lopencv_rapid -lopencv_reg -lopencv_rgbd -lopencv_saliency -lopencv_shape -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_superres -lopencv_optflow -lopencv_surface_matching -lopencv_tracking -lopencv_highgui -lopencv_datasets -lopencv_text -lopencv_plot -lopencv_ml -lopencv_videostab -lopencv_videoio -lopencv_viz -lopencv_ximgproc -lopencv_video -lopencv_dnn -lopencv_xobjdetect -lopencv_objdetect -lopencv_calib3d -lopencv_imgcodecs -lopencv_features2d -lopencv_flann -lopencv_xphoto -lopencv_photo -lopencv_imgproc -lopencv_core -lASICamera2 -I/usr/include/libusb-1.0 -lusb-1.0 allsky_common.o

EricClaeys, 2022-11-28:
@AlexXu12256   Thanks.  You have armv7 so updating that file in ~/allsky/src/lib/armv7, then recompiling should work.
As a test, rename ~/allsky/src/lib/armv7/ASICamera2.a to anything. then remove ~/allsky/src/capture, then try to compile again.  It should give an error that it can't find the ASICamera2.a file.  Then restore the file name and try to compile again.

When searching for "ZWO SDK" in the log file, make sure you look at the LAST entry, which is the newest.

AlexXu12256, 2022-11-28:
@EricClaeys It is working now, thank you so much for the time you have spent on this.

closed 2022-11-28 by AlexXu12256

EricClaeys, 2022-11-29:
@AlexXu12256    My pleasure.  What did you do to get it working?

AlexXu12256, 2022-11-29:
@EricClaeys 
I run the following commands:

systemctl stop allsky
cd ~/allsky/src
rm capture
(switch the file in armv7)
make capture
cp capture ..
systemctl start allsky

I think the reason the reason it wasn't working before could be that I didn't remove the previous capture.

EricClaeys, 2022-11-29:
@AlexXu12256    Good point.  That's what the `cp capture ..` did.  Allsky uses the `capture` program in ~/allsky, not ~/allsky/src.