AllskyTeam/allsky #1917: [BUG]  Just loaded PI firmware  can not ssh into it

BobW55, 2022-11-27:
C:\Users\xxxxx>ssh pi@raspberrypi.local
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ED25519 key sent by the remote host is
SHA256:0uxDrvNaxvznuyqYVhGH+VO5s9fVdc/FYOAaxXGPOAU.
Please contact your system administrator.
Add correct host key in C:\\Users\\xxxxx/.ssh/known_hosts to get rid of this message.
Offending ECDSA key in C:\\Users\\xxxxx/.ssh/known_hosts:3
Host key for raspberrypi.local has changed and you have requested strict checking.
Host key verification failed.

Making a NEW load on a 256GB SDcard.  using same instructions from the other night  and this video
https://www.youtube.com/watch?v=7TGpGz5SeVI
My first load on a 32GB card ran without issue.
Sorry Linex is not my thing.

EricClaeys, 2022-11-27:
@BobW55

The key is this line from the error message:
```
Add correct host key in C:\Users\xxxxx/.ssh/known_hosts to get rid of this message.
```
When you first `ssh` into a machine, it prompts for something about wanting to know who you are.  When you enter "yes", it puts an entry in your `~/.ssh/known_hosts` file describing the remote machine.
The next time you `ssh` to the remote machine, you aren't prompted because `ssh` already knows about the machine from its `~/.ssh/known_hosts` entry.

I suspect what happened is when you `ssh`ed into the 32 GB card, an entry was added to `~/.ssh/known_hosts`.  Now when you try to `ssh` to the 8 GB card, it probably has the same IP address as the 32 GB card did, but `ssh` considers it a different machine and the `~/.ssh/known_hosts` entry is no longer valid.

The fix is to remove the "old" entry in `known_hosts`.  The trick will be determining which entry to delete.  If there's only one, simply delete the file.
This line **may** indicate it's the 3rd entry:
```
Offending ECDSA key in C:\Users\xxxxx/.ssh/known_hosts:3
```
Before doing anything, I suggest you make a copy of the `known_hosts` file.

BobW55, 2022-11-27:
Thank you.....
Just deleted the .ssh file and it let me in

BobW55, 2022-11-27:
SOLVED by deleting files in the .ssh folder on my computer

closed 2022-11-27 by BobW55

EricClaeys, 2022-11-27:
@BobW55    Cool!  Glad I was able to help.