AllskyTeam/allsky #1968: Trailing / in upload.sh

ea1ii, 2022-12-04:
My server did not allow for a trailing / in 'cd' commands. Needed to comment out line 131:

<code>[ "${REMOTE_DIR}" != "" -a "${REMOTE_DIR: -1:1}" != "/" ] && REMOTE_DIR="${REMOTE_DIR}/"</code>

ea1ii, 2022-12-04:
Playing with something else, I enabled the line again and seems to work now. I don't understand (baffled)

EricClaeys, 2022-12-04:
Let me know if it happens again.

EricClaeys, 2022-12-05:
@ea1ii    When you created this issue did it have some text on what information to include, etc.?  Or was it a blank page?

ea1ii, 2022-12-05:
Hi Eric,

No it was not blank. I only just overlooked it (sorry).

When that happened I feed manually the commands to lftp and I'm sure I saw it happening. I want to check again, but I'm back on night shift and afdter that off to Spain for a few days. I want to check again in a week or two and I'll forward you the results (I'm curious, you know?).

It think it might be the hosting being to fresh new as I noticed some initial 'latency' acknowleging changes. But I'll check again. Promise.

EricClaeys, 2023-01-06:
@ea1ii, any update?

ea1ii, 2023-01-24:
Forgot about this one. It's fixed.

closed 2023-01-24 by ea1ii