AllskyTeam/allsky #1969: First Time Install Problem - Failed to fetch from /www.mirrorservices.org - Service Unavailable

EdwardonGitHub, 2022-12-04:
I'm doing a first time install (Rpi HQ Camera) with ./install.sh (from within the allsky directory)

Multiple "Failed to fetch from /www.mirrorservices.org/.......... - Service Unavailable
Install terminates with "Installing dependencies failed!

Is this short-term and can be resolved by trying again later, or longer-term?

[221204 1233 Response to install.sh cmd.docx](https://github.com/thomasjacquin/allsky/files/10148493/221204.1233.Response.to.install.sh.cmd.docx)

EricClaeys, 2022-12-04:
@EdwardonGitHub   I was just able to "ping" the server it couldn't contact, so I suggest trying again.  We don't have any control over which server the Linux update gets its files from.
Let me know if it works the second time.

EdwardonGitHub, 2022-12-04:
Eric,

 

I have tried again the result being the same failure reports.

 

I can ping 212.219.56.184 with an immediate successful response.

 

I’ll keep trying, it seems it will resolve given time.

 

Thanks Edward

EdwardonGitHub, 2022-12-05:
Okay, now resolved and the installation completed.

 

The issue can be closed.

 

Thanks for your attention. Edward