AllskyTeam/allsky #1999: *** WARNING: time to take image () differs from requested exposure time[BUG]

JosWennmacker, 2022-12-07:
### Environment

* Camera:  ZWO ASI 178 MC
* OS: Ubuntu 22.04.1
* Allsky version: *** Allsky Camera Software v0.8.3.3 |  2022 ***
     * The newest software includes a file called  ~/allsky/version
     * If it does not exist, run:       grep Software /var/log/allsky.log
* Pi 4B, 4GB)

### Bug Description

*** WARNING: time to take image () differs from requested exposure time

Strange exposure times, sometimes even negative times.  This was also mentioned in Issue #1118 

This happened several times during the night. 

In my case I found the probable cause. It only happens when I boot the Pi from a USB3 attached SSD. When booting from the SD card there is np problem (at least not this;-). 

So thiss is not really an issue, but maybe an explanation of an issue.

Kind regards,
Jos

### Log / configuration files

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.

JosWennmacker, 2022-12-13:
@EricClaeys, no, the images are not ok. Exposure times are really much shorter than expected. In fact they eventualy become so short there is not enough time to save an image before the next is ready. Moreover, several images are taken in the same second, resulting in same image filenames. I attached part of the allsky log.
- My camera is not outside at the moment. I'm rebuilding it, so the images are taken inside. The problem starts either at the moment taking of night images starts or at a moment lights in the room are switched on or off.
- The SSD is an image of the SD  card.
[allsky - error.log](https://github.com/thomasjacquin/allsky/files/10213421/allsky.-.error.log)

EricClaeys, 2022-12-13:
@JosWennmacker   I've have seen this before where we tell the camera to take a 30-second exposure and it returns almost immediately. I have not seen any pattern but suspect it may be that the camera isn't getting enough power.

Your delay is 10 ms which is why it's sometimes taking two pictures in a second.
Try increasing your delay to 2 seconds and turn off auto-gain if it's on.

If that doesn't help, stop Allsky run `sudo truncate -s 0 /var/log/allsky.log`, then change the Debug Level in the WebUI to 4 and press on the "Save changes" button.  About a minute later toggle the lights.  Keep track of what time you toggled the lights, ideally to the second so I can compare it to the log file.

Is it doing this all the time, or just occasionally?

JosWennmacker, 2022-12-14:
@EricClaeys I did the following:
- connected the camera to a powered USB hub to make sure it gets enough power. There was no change in the exposure errors.
- changed the 10 ms delay to 2000 ms. The exposure errors didn't change, the problem with double file names -of course- disappeared.
- checked auto-gain: it was never on
- stopped Allsky, truncated log file, set Debug Level to 4 (actually it was already) and started Allsky with the lens cap on the camera. At Wed Dec 14 02:05:30 CET 2022 I removed the lens cap.
- exposure errors started just after the start of allsky and again later when I removed the cap.
- warnings come every time I toggle the lights.
- log file is attached.
[allsky.log](https://github.com/thomasjacquin/allsky/files/10223441/allsky.log)

EricClaeys, 2022-12-29:
@JosWennmacker, try turning off "Auto USB" and adjust the `USB Bandwidth` in the WebUI up and down.  It goes from 40 - 100.  My only guess right now is the SSD disk is hogging the USB bandwidth and it's impacting the camera's ability to transfer data.  You can also try enabling `Version 0.8 Exposure Mode`.

sanlupkim, 2023-01-15:
After upgrading from 0.8.3 to v0.8.3.3 I've got similar problem also at ASI 178 MC. "WARNING: time to take image" message appear sometimes but when is cloudy (exposure is not max 10s but less) every 5-6 image is much brighter than others. Look at mean:
```
image @ mean 108.  Reported exposure: 6.9 sec,
image @ mean 156.  Reported exposure: 8.4 sec,
image @ mean 130.  Reported exposure: 7.3 sec,
image @ mean 125.  Reported exposure: 7.0 sec,
image @ mean 115.  Reported exposure: 6.7 sec,
image @ mean 108.  Reported exposure: 6.8 sec,
image @ mean 159.  Reported exposure: 8.3 sec,
image @ mean 128.  Reported exposure: 7.2 sec,

```
I've Tried on/off "Version 0.8 Exposure Mode" and changing max speed USB but no luck.
Timelapse of this images looks very "unstable". 
detail log attached
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/10420187/config.sh.txt)
[allsky.log](https://github.com/thomasjacquin/allsky/files/10420188/allsky.log)

JosWennmacker, 2023-01-19:
Turning off Auto USB AND enabling Version 0.8 Exposure Mode seems to do the trick. Since I changed this last night there are no Warnings anymore, even when light conditions change suddenly.
There are also no ASI_ERROR_TIMEOUT errors in the log. However th following message appears in the syslog with EVERY exposure:
Jan 19 00:53:13 allsky allsky.sh[50975]:   > Camera set auto-exposure to 46.84 ms (0.05 sec), timeout: 5,093 ms
**Jan 19 00:53:13 allsky kernel: [87166.382871] usb 2-1.4: reset SuperSpeed USB device number 4 using xhci_hcd**
Jan 19 00:53:14 allsky allsky.sh[50975]:   > Got image @ mean 129.  Reported exposure: 43.09 ms (0.04 sec), auto=yes

EricClaeys, 2023-01-19:
@JosWennmacker Glad you got it working.  The ASI_ERROR_TIMEOUT messages are a pain.  Most people don't see them, but for people that do, it's a hit-or-miss regarding how to resolve it.  The associated Wiki page on it works for most people.  Some people found it useful to stand on their left foot and chant...

See [this Wiki page](https://github.com/thomasjacquin/allsky/wiki/Troubleshooting-other) for info on the USB log message.  It's not an Allsky issue and can be ignored.

If this Issue is resolved, would you please close it?  Thanks - Eric

sanlupkim, 2023-01-19:
In my case I still can see warning in log
Jan 19 04:58:20 allsky allsky.sh[15222]: *** WARNING: time to take image (9,791,006) differs from requested exposure time (6,520,590) by 3,270,416, threshold=3,260,295
Jan 19 06:02:49 allsky allsky.sh[15222]: *** WARNING: time to take image (9,785,549) differs from requested exposure time (6,517,535) by 3,268,014, threshold=3,258,767
Jan 19 06:06:03 allsky allsky.sh[15222]: *** WARNING: time to take image (9,177,195) differs from requested exposure time (6,111,238) by 3,065,957, threshold=3,055,619

look at attched two kerograms -  smooth is from version v0.8.3, rough - v0.8.3.3
[image]

EricClaeys, 2023-01-19:
@JosWennmacker    So two nights ago you didn't see the warnings but last night you did?  Did anything change?

The bands are caused by the brightness slowing decreasing over a few pictures then all of a sudden increasing.  I haven't been able to determine why, and don't plan to spend time on it since the next release is getting close to being released. 
If you want to test with the new release,  let me know.

sanlupkim, 2023-01-19:
I'd like to test new release becasue bands are annoying for me. It's dev branch in code?

EricClaeys, 2023-01-19:
@sanlupkim, please only install the new release if you are willing to provide feedback and help us identify and resolve issues.  You don't need to be a coder, but will need to provide detailed input like logs and screenshots.

If you agree, go to the Wiki page and at the bottom of the table of contents is a link to the new documentation. In there you'll find the instructions for installing allsky.
The only change is to add `--branch dev` to the `git clone` command.  If you want to install the Allsky Website you'll need to execute `website/install.sh --branch dev`.

JosWennmacker, 2023-01-24:
As said: turning off Auto USB AND enabling Version 0.8 Exposure Mode seems to do the trick. 
Thanks  Eric.

closed 2023-01-24 by JosWennmacker