AllskyTeam/allsky #2044: [UPDATE] Update ZWO SDK to newest version to support new cameras

akermania, 2022-12-15:
### Environment

* Camera:  ZWO 678MC
* OS: Bullseye
* Allsky version: v2022.03.01
* Pi model and amount of memory: 4b 4GB

### Bug Description

I bough a new ZWO 678MC which was out a few months ago, AllSky software did not recognize it and output:

*** ERROR: No Connected Camera...
'capture' exited with RETCODE=100

After debugging this was due to the SDK function ASIGetNumOfConnectedCameras()
I then downloaded the latest ZWO SDK (v1.27) and replaced the files under: 
~/allsky/src/include/ASICamera2.h
~/allsky/src/lib/armv7/libASICamera2.a

and ran make clean all - it did not work
What worked was to delete the allsky directory completely, download a fresh repo, BEFORE installing replace the files above, and only them run install.sh 

It is now working !  

### Log / configuration files

EricClaeys, 2022-12-16:
Closing - the next release of Allsky will include the latest ZWO library which supports their newest cameras.

closed 2022-12-16 by EricClaeys

EricClaeys, 2023-04-21:
The workaround is fairly easy if you are somewhat Linux savy:
~~~
cd ~/allsky/src/lib/armv7
mv libASICamera2.a libASICamera2.a-OLD
# Copy the attached file to this directory and remove the trailing ".txt" from its name.
cd ../..
make capture
sudo systemctl stop allsky
cp capture ..
sudo systemctl start allsky
~~~
What you are doing is re-creating the ZWO "capture" program with their latest library.

[libASICamera2.a.txt](https://github.com/thomasjacquin/allsky/files/11298752/libASICamera2.a.txt)

Vejbymats, 2023-04-21:
Thanks, I'l come back after a try.

Vejbymats, 2023-04-21:
Well, it did not work...
First, I was in the middle of trying the akermania way , and had deleted the allsky folder.
Then when installing I also have the problem with armv7 and armv8 (#2525) and so not compiling  
So, I do not have a armv7 any more and therefore put the file in armv8.
But, same result...

Vejbymats, 2023-04-21:
I tryed to start from the beginning, changing from 32 to 64-bit OS.
Switched the file libASICamera2.a before running the install (verifyed after install) but the same result.
Still a big error on live view, and " *** ERROR: No Connected Camera..." in the log.

EricClaeys, 2023-04-22:
Just to be clear, are you running Bullseye 64-bit?  The file I uploaded is for 32-bit.

Vejbymats, 2023-04-22:
No, at first  I used Raspberry Pi OS Lite (32-bit), but when that did not work (with new file from you) I changed to Raspberry Pi OS Lite (64-bit)
(With 64-bit I dont have to rename and switch armv-folders.)
This is a totaly new installation, so changing OS is only a few minutes extra.

Vejbymats, 2023-04-22:
Now it works!

I swapped the "libASICamera2.a"-file in both armv7 and -8 and in x64 folder from matching folders in the downloaded "ASI Camera SDK" before installing (as in OP's post) and now I have a picture.
Thanks to akermania and Eric for help and guidance!

EricClaeys, 2023-04-22:
The February update to Pi OS changed what the 32-bit OS reports for its "architecture" when run on a 64-bit machine like the Pi 4.  It used to report "armv7" which is why Allsky looked in allsky/src/lib/armv7 for the ZWO library file.  The Pi OS now reports "aarch64" so Allsky looked in allsky/src/lib/armv8, which caused Allsky installation to fail since that ZWO library file is only for 64-bit OS.

The next Allsky release fixes this as well as uses the newest ZWO library.

Vejbymats, 2023-04-23:
Thanks for the update.
There is still some issues, but I think I will wait for the new release to
hopefully take care of them.
Camera is up and running so soon it will show on the map.