AllskyTeam/allsky #2135: Segmentation fault in CAPTURE on Raspberry Pi Zero W with ZWO ASI120MC-S

keenerb, 2023-01-20:
Jan 20 18:32:04 allsky400 allsky.sh[842]:      ***** Starting AllSky *****
Jan 20 18:32:09 allsky400 allsky.sh[842]: CAMERA: ZWO
Jan 20 18:32:09 allsky400 allsky.sh[842]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
Jan 20 18:34:51 allsky400 allsky.sh[842]: /home/pi/allsky/allsky.sh: line 250:  1371 Segmentation fault      "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Jan 20 18:34:51 allsky400 allsky.sh[842]: 'capture' exited with RETCODE=139
Jan 20 18:34:51 allsky400 allsky.sh[842]: ***
Jan 20 18:34:51 allsky400 allsky.sh[842]: *** After fixing, restart the allsky service. ***
Jan 20 18:34:51 allsky400 allsky.sh[842]: ***
Jan 20 18:34:56 allsky400 allsky.sh[842]:      ***** AllSky Stopped *****
Jan 20 18:34:57 allsky400 allsky.sh[842]: Terminated

Works on a Raspberry Pi 3, but not on a Zero W.

It's a fresh git pull from this morning.

Any idea what might be going wrong here?

This is what an strace reported:

```clock_gettime64(CLOCK_MONOTONIC, {tv_sec=551, tv_nsec=434333107}) = 0
openat(AT_FDCWD, "/proc/self/auxv", O_RDONLY) = 3
read(3, "\20\0\0\0\326\201\0\0", 8)     = 8
close(3)                                = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_ACCERR, si_addr=0x319aa} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

keenerb, 2023-01-21:
I think you can ignore this, I re-ran install from scratch and all seems well.

EricClaeys, 2023-01-21:
@keenerb, that's not good.  Segmentation fault is usually when the program tries to write to a memory location that doesn't exist, often location 0.  That would be a bug in the `capture` program.  I haven't seen one of those faults in a long time.

If you still have it, would you please ATTACH your /etc/raspap/settings_ZWO.json file that existed when the crash occurred?

keenerb, 2023-01-21:
I think it's because I copied over too much from the Raspberry Pi 3 install.  A fresh install worked fine, and the old install works fine on the Pi 3, so it may be an architectural thing?

EricClaeys, 2023-01-21:
@keenerb what kind of stuff did you copy over?  Allsky?
If you copied the "capture" command, I'm not surprised it crashed right after starting, since that's when it uses the ZWO library, which is architecture dependent.

EricClaeys, 2023-01-30:
Closing.  User likely used Pi 3 executable on Pi 4, which doesn't work.

closed 2023-01-30 by EricClaeys