AllskyTeam/allsky #2191: Problème ZWO Asi 120 mc

Michel85300, 2023-02-07:
J'ai tout installer ça fonctionne bien avec ma 290 MM et impossible de faire fonctionner l'Asi 120 MC
Merci

EricClaeys, 2023-02-07:
@Michel85300   The README.md file on the main GitHub page has this note:

**NOTE: Owners of USB2.0 cameras such as ASI120MC and ASI120MM may need to do a [firmware upgrade](https://astronomy-imaging-camera.com/software-drivers). This changes the camera to use 512 byte packets instead of 1024 which makes it more compatible with most hardware.**

Depending on what the issue is, this may fix your problem.  If not, please give us more information on exactly what problems you are having.  See the "Reporting Issues" Wiki page on how to do that.

EricClaeys, 2023-02-23:
@Michel85300    Any update?

EricClaeys, 2023-03-18:
Closing Issue. User has not responded in about a month.

closed 2023-03-18 by EricClaeys