AllskyTeam/allsky #2213: Excessively long exposure times at Sunrise

havocedge, 2023-02-12:
For some reason the allsky app is completely ignoring my max exposure times and taking 1-1.5 hours to reach an appropiate exposure time every morning. My max night time exposure is 45secs and max daytime exposure is 10secs. I don't udnerstand why it is jumping to 300+ secs and taking a long time to adjust back down.

`Feb 12 06:08:17 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:08:17   @ 45.0 sec
Feb 12 06:09:04 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:09:04   @ 45.0 sec
Feb 12 06:09:50 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:09:50   @ 45.0 sec
Feb 12 06:10:36 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:10:36   @ 45.0 sec
Feb 12 06:11:22 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:11:22   @ 45.0 sec
Feb 12 06:12:09 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:12:09   @ 45.0 sec
Feb 12 06:12:55 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:12:55   @ 357.4 sec
Feb 12 06:19:09 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:19:09   @ 357.4 sec
Feb 12 06:25:22 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:25:22   @ 321.7 sec
Feb 12 06:31:00 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:31:00   @ 289.5 sec
Feb 12 06:36:06 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:36:06   @ 260.6 sec
Feb 12 06:40:43 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:40:43   @ 234.5 sec
Feb 12 06:44:54 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:44:54   @ 211.1 sec
Feb 12 06:48:41 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:48:41   @ 190.0 sec
Feb 12 06:52:08 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:52:08   @ 171.0 sec
Feb 12 06:55:15 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:55:15   @ 153.9 sec
Feb 12 06:58:06 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:58:06   @ 138.5 sec
Feb 12 07:00:41 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:00:41   @ 124.6 sec
Feb 12 07:03:02 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:03:02   @ 112.2 sec
Feb 12 07:05:10 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:05:10   @ 101.0 sec
Feb 12 07:07:07 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:07:07   @ 90.9 sec
Feb 12 07:08:54 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:08:54   @ 81.8 sec
Feb 12 07:10:32 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:10:32   @ 73.6 sec
Feb 12 07:12:02 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:12:02   @ 66.2 sec
Feb 12 07:13:25 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:13:25   @ 59.6 sec
Feb 12 07:14:41 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:14:41   @ 53.7 sec
Feb 12 07:15:51 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:15:51   @ 48.3 sec
Feb 12 07:16:55 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:16:55   @ 43.5 sec
Feb 12 07:17:55 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:17:55   @ 39.1 sec
Feb 12 07:18:50 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:18:50   @ 35.2 sec
Feb 12 07:19:42 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:19:42   @ 31.7 sec
Feb 12 07:20:30 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:20:30   @ 28.5 sec
Feb 12 07:21:15 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:21:15   @ 25.7 sec
Feb 12 07:21:57 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:21:57   @ 23.1 sec
Feb 12 07:22:36 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:22:36   @ 20.8 sec
Feb 12 07:23:13 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:23:13   @ 18.7 sec
Feb 12 07:23:48 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:23:48   @ 16.8 sec
Feb 12 07:24:21 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:24:21   @ 15.2 sec
Feb 12 07:24:53 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:24:53   @ 13.6 sec
Feb 12 07:25:23 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:25:23   @ 12.3 sec
Feb 12 07:25:51 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:25:51   @ 11.0 sec
Feb 12 07:26:19 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:26:19   @ 9.9 sec
Feb 12 07:27:04 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:27:04   @ 3.94 ms (0.00 sec)
Feb 12 07:27:20 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:27:20   @ 3.53 ms (0.00 sec)
Feb 12 07:27:36 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 07:27:36   @ 3.53 ms (0.00 sec)
`

[image]

havocedge, 2023-02-12:
I tried to disable the version 0.8 exposure method and that made things worse where every other frame was vastly overexposed. The exposure time was flipping between 200-250usecs to 3-4ms.

EricClaeys, 2023-02-12:
@havocedge   There's a bug in the current version where "max exposure" isn't really the max exposure.  I think it's the starting exposure instead.  It's fixed in our new release which should be out this month.

havocedge, 2023-02-12:
@EricClaeys Does that explain why the first image after the endofNight script runs is 6 minutes long?

Feb 12 06:16:12 allsky allsky.sh[17767]: endOfNight.sh: ===== Timelapse complete
Feb 12 06:16:12 allsky allsky.sh[17767]: endOfNight.sh: Deleting old directory /home/pi/allsky/images/20230128
Feb 12 06:18:54 allsky allsky.sh[17767]:   > Sleeping 15.0 sec from histogram exposure
Feb 12 06:19:09 allsky allsky.sh[17767]: STARTING EXPOSURE at: 2023-02-12 06:19:09   @ 357.4 sec

The max Daytime exposure in the config is set to 10 seconds.

EricClaeys, 2023-02-12:
@havocedge   It's supposed to never go longer than the maximum auto exposure the camera supports, which for most ZWO cameras is 200 seconds.  I'm pretty sure there's a bug there as well.  If you want to try the new release now, I can let you know how.  It's pretty stable, but if you elect to install it I'd ask you to install the official release when it's out - that will be very easy to do.

havocedge, 2023-02-12:
Ok I'm using an ASI385MC which has a max exposure time of 2000secs. I willing to try the new version. 

Sensor: 1/2″ CMOS IMX385
Diagonal: 8.35mm
Resolution: 2.12Mega Pixels 1936×1096
Pixel Size: 3.75µm
Exposure Range: 32µs-2000s

EricClaeys, 2023-02-12:
@havocedge   Go to the Wiki link on any GitHub page; on the bottom of the table of contents, click on the "New documentation" link.  Look at the Installation-->Allsky page and follow the instructions with one exception, Add `--branch dev` to the `git` line, i.e., `git --dev branch --clone .....`.
If you have the Allsky Website installed on your Pi or on a remote server, look at the Installation-->Website page.  Again, add `--branch dev` to the `git` line.

havocedge, 2023-02-13:
Here's the lines from this morning logs with debug increased.

Feb 13 06:11:56 allsky allsky.sh[30162]: xxx resetGainTransitionVariables(0, 180) called at DAY
Feb 13 06:11:56 allsky allsky.sh[30162]: Processing end of night data
Feb 13 06:11:56 allsky allsky.sh[30162]: ==========
Feb 13 06:11:56 allsky allsky.sh[30162]: === Starting daytime capture ===
Feb 13 06:11:56 allsky allsky.sh[30162]: ==========
Feb 13 06:11:56 allsky allsky.sh[30162]: Using the last night exposure, old and new Gain to calculate new exposure of 343.1 sec
Feb 13 06:11:56 allsky allsky.sh[30162]: Turning off daytime auto-exposure to use histogram exposure.
Feb 13 06:11:56 allsky allsky.sh[30162]: STARTING EXPOSURE at: 2023-02-13 06:11:56   @ 343.1 sec

My Night Time Gain is 180 and my Daytime is set to 1. Auto Gain is disabled for both. I will install the dev version today.

havocedge, 2023-02-16:
@EricClaeys The new dev version seems to have fixed the problem after tweaking the settings. This new version is a pretty significant upgrade. I like it a lot. I did find one odd issue. The REMOTE_PORT var in the ftp-settings.sh file is only used for FTP/SFTP. I switched to uploading via SCP with SSH keys and had to manually edit the upload.sh script to leverage the REMOTE_PORT var as I don't use the default SSH port of 22. I'm not sure why all methods that can specify a non-default port don't use this var.

EricClaeys, 2023-02-16:
@havocedge. Thanks for the feedback. We are pretty excited by this release.  The next one will be even better!
I don't use `scp` so did not know it accepted a port option.  I will fix that, thanks for letting me know. 
Are there any other protocols that also accept it?

EricClaeys, 2023-02-17:
The fix for the "scp" protocol is now in the "dev" release.  Thanks for pointing it out.

closed 2023-02-17 by EricClaeys