AllskyTeam/allsky #2257: AllSky lattitude/longitude

psenness, 2023-02-21:
I have entered my location in AllSky config.js as 48.2 lat and -122.6 long. The sunwait report command shows the locaton as 40.1N and 271.7E so the program switches to NIGHT mid afternoon. I can't find where the error is coming from and need some help.
Thanks.

EricClaeys, 2023-02-21:
@psenness   The Allsky Website should be showing 48.2N latitude and 122.6W longitude so I don't know where it's getting 40.1N and 271.7E.
Is your Website on the Internet?  If so, what's the URL?  If not, please attach the config.js file.
What's your "angle"?  It also impacts when daytime is.

Here's what I get for an angle of -6:
~~~
pi:~ $ sunwait list angle  -6  48.2N  122.6W
08:37, 20:10
~~~

psenness, 2023-02-21:
Eric,

Thanks for your quick response. I have attached the configuration file and it shows the correct lat/long. I use -6 for the angle also. I do not have an online website for this project – yet.

 

I’m using a Pi 4B with 8Gb and a RPiHQ camera.

 

In doing more troubleshooting this morning I did a complete reinstall per the video from Patriot Astro. I got the same results. I’ve also attached the sunwait report I mentioned in my first email. It shows the correct local date, time and time zone. The times for all the twilights are off by two hours. The “Target Information… Location” is what I find confusing. Longitude 272E (which doesn’t exist) would be 92W – which is central standard time and two hours different from PST (my time zone) which would explain the off shift for day imaging to night imaging and back. And like you, I have no idea where those coordinates come from and I surely didn’t enter anything like that. The other item I’m curious about is the “General Information    (no offset)…” comment. Leads me to believe there is an “offset” setting somewhere.

 

If I may speculate – my understanding is that “sunwait” is a program called within AllSky and the data is passed to it. Is it possible the coding for “sunwait” has a default location that accounts for this behavior? I know that that is a longshot because if that were true then lots of other people would be experiencing the same issue.

 

I appreciate your help.

 

Kind regards,

Paul

 

      Current Date and Time: 21-Feb-2023 09:24 PST

Target Information ...

                   Location:  40.108306N, 271.742700E
                       Date: 20-Feb-2023
                   Timezone: PST
   Sun directly north/south: 10:06
             Twilight angle: -0.83 degrees (daylight)
          Day with twilight: 04:38 to 15:34
                      It is: Day (or twilight)

General Information (no offset) ...

 Times ...         Daylight: 04:38 to 15:34
        with Civil twilight: 04:12 to 16:00
     with Nautical twilight: 03:41 to 16:32
 with Astronomical twilight: 03:09 to 17:03

 Duration ...    Day length: 10:55 hours
        with civil twilight: 11:48 hours
     with nautical twilight: 12:51 hours
 with astronomical twilight: 13:54 hours

#!/bin/bash

# NOTES:
#  - Variable names must start at the beginning of a line.
#  - There should be nothing after a variable's value, i.e., no comments on the same line.
#  - Values can optionally be quoted with double quotes, NOT single quotes.
#  - If a variable occurs multiple times, only the last value is used.

# Choose between "ZWO" or "RPiHQ".
CAMERA="RPiHQ"

########## Images
# Set to "true" to upload the current image to your website.
IMG_UPLOAD="true"

# If IMG_UPLOAD is "true", upload images every IMG_UPLOAD_FREQUENCY frames, e.g., every 5 frames.
# 1 uploades every frame.
IMG_UPLOAD_FREQUENCY=1
[... 188 lines cut ...]
 
# So scripts can conditionally ouput messages; DO NOT CHANGE NEXT LINES.
ALLSKY_DEBUG_LEVEL=$(jq -r '.debuglevel' "${CAMERA_SETTINGS}")

var config = {
	comment: "See https://github.com/thomasjacquin/allsky/wiki/allsky-website-Settings for a description of these settings",
	title: "AllSky",
	imageName: "/current/tmp/image.jpg",
	location: "H3T Observatory",
	latitude_longitude_comment: "latitude and longitude must be decimal numbers with OUT the trailing N, S, E, and W.",
    latitude: 48.2,
    longitude: -122.6,
	az: 0,
	camera: "RPiHQ",
	lens: "Fisheye",
    computer: "Raspberry Pi 4 Model B Rev 1.4",
	owner: "Paul",
	overlaySize: 875,
	overlayOffsetLeft: 0,
	overlayOffsetTop: 0,
	showOverlayAtStartup: false,
	auroraForecast: false,
    auroraMap: "north",
	intervalSeconds: 5,
}

EricClaeys, 2023-02-21:
@psenness    Paul,  I learn something new every day.  I wasn't aware of "sunwait report".  I have no idea where it's getting the "Location" data from - mine is way off.  Allsky doesn't use "report" so **ignore it**.  Allsky uses:
~~~
sunwait poll angle  -6  48.2N  122.6W
~~~
Which returns either "DAY" or "NIGHT".
To see when Allsky things your daytime and nighttime start:
~~~
sunwait list angle  -6  48.2N  122.6W
~~~

psenness, 2023-02-21:
Eric,

 

Sunwait list… returns expected times for my location. I’ll watch tonight and tomorrow and see where the switch takes place. This morning it switched from night to day at 4:38, two hours before it should have and correlated with the sunwait report listing. Maybe the reinstall fixed it. I’ll let you know.

 

Paul

EricClaeys, 2023-02-21:
@psenness   Paul, is the timezone on the Pi correct?
Can you post `~/allsky/tmp/data.json` ?

psenness, 2023-02-21:
Eric,

 

Yes, I’ve confirmed the time zone on the Pi is set to America/Los Angeles.

 

I tried some methods to get the file off the Pi and onto my laptop, but none seem to work. If you have a suggestion that will work I’d be happy to try it and send the file requested. (I tried SSH and USB methods I found on the web.)

Paul

EricClaeys, 2023-02-21:
I mount /home/pi from my Pi onto my PC just like any other network drive.  I can then easily copy files.

To do this:
~~~
sudo apt install samba -y
sudo smbpasswd -a pi
sudo cp /etc/samba/smb.conf /etc/samba/smb.bak
# Using a text editor, put the lines below into /etc/samba/smb.conf
sudo /etc/init.d/smbd restart
~~~

I have the following in /etc/samba/smb.conf:
~~~
[global]
   workgroup = WORKGROUP
   log file = /var/log/samba/log.%m
   max log size = 1000
   logging = file
   panic action = /usr/share/samba/panic-action %d
   server role = standalone server
   obey pam restrictions = yes
   map to guest = bad user
   usershare allow guests = no
[pi_home]
   comment=Pi home directory
   path=/home/pi
   veto files = /.*/
   browseable = yes
   read only = no
   create mask = 0664
   directory mask = 0775
~~~

EricClaeys, 2023-02-22:
Closing issue per user request.  The problem is solved.

closed 2023-02-22 by EricClaeys