AllskyTeam/allsky #2525: Fresh install fails on Compile [BUG]

WheresWally, 2023-04-02:
Hi Guys.
I am tying to reinstall AllSky on my Raspberry Pi 1GB, with RPi HQ camera.
You can see it running here with previous install. 
http://brisbanesky.ddns.net:82/allsky/

I gave up and put original SD back in.

I made it to the step
**./install.sh**

it was then I get the following error.

`pi@raspberrypi:~/allsky $ ./install.sh

**********************************************
*** Welcome to the Allsky Camera installer ***
**********************************************

* Dependencies installation

make[1]: Entering directory '/home/pi/allsky/src'
2023-04-02 13:45:01 Installing build dependencies...
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
bc is already the newest version (1.07.1-2+b1).
gawk is already the newest version (1:5.1.0-1).
imagemagick is already the newest version (8:6.9.11.60+dfsg-1.3+deb11u1).
jq is already the newest version (1.6-2.1).
lftp is already the newest version (4.8.4-2+b1).
libopencv-dev is already the newest version (4.5.1+dfsg-5).
libusb-1.0-0-dev is already the newest version (2:1.0.24-3).
libusb-dev is already the newest version (2:0.1.12-32).
ffmpeg is already the newest version (7:4.3.5-0+deb11u1+rpt3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
make[1]: Leaving directory '/home/pi/allsky/src'

* Compile allsky software

make[1]: Entering directory '/home/pi/allsky/src'
2023-04-02 13:45:07 Building capture program...
/usr/bin/ld: lib/armv8/libASICamera2.a: error adding symbols: file format not recognized
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:132: capture] Error 1
make[1]: Leaving directory '/home/pi/allsky/src'
make: *** [Makefile:24: all] Error 2
Compile failed!
pi@raspberrypi:~/allsky $ ./install.sh
`

I found where this sort of error was happening a year ago and the brains-trust concluded it was 64 bit OS.
I ran the suggested tests as per below.
It seems I did install the 32 bit OS but it does seem to say 4 bit as well.
very odd.

`pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ getconf LONG_BIT
32
pi@raspberrypi:~ $ md5sum ~/allsky/src/lib/armv8/libASICamera2.a
5161e951ed924dffe7b4131c5ea87a96  /home/pi/allsky/src/lib/armv8/libASICamera2.a
pi@raspberrypi:~ $ tail -n 4 /proc/cpuinfo
Hardware        : BCM2835
Revision        : a03115
Serial          : 10000000ce4d8a2c
Model           : Raspberry Pi 4 Model B Rev 1.5

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
`

Any suggestions for me to try tomorrow?

The reason I wanted to do a fresh install is to get the timelapse and startrails to work.
Also I noticed the constellation overlay is incorrect.
Maybe it will be faster to fix those.

LOL

EricClaeys, 2023-04-02:
@WheresWally, the February update to Pi OS changed how it reported the architecture, which broke Allsky and other programs.  32-bit Bullseye is being treated as 64-bit Bullseye which causes compiles to fail, as you saw.
The workaround is easy though:
~~~
cd ~/allsky/src/lib
mv armv8 armv8-ORIGINAL
mv armv7 armv8
~~~
Then run the installation again.

EricClaeys, 2023-04-16:
A workaround was provided 2 weeks ago and no response from user.  The issue is fixed in the next release.

closed 2023-04-16 by EricClaeys

WheresWally, 2023-04-18:
Oh ok, Sorry I missed this, I shall look and try and report back shortly.
Many thanks.

WheresWally, 2023-04-18:
That looks simple enough, I will give it a crack in the light of day.
I really want to get timelapses working. If I try to run the command to generate the time lapse, it tells me some command is not found.
I will need to repeat that to report exactly what.

Then I have a problem, if you have a look at the link, the constellations are all wrong.
How do I fix that?
http://brisbanesky.ddns.net:82/allsky/.

EricClaeys, 2023-04-18:
@WheresWally   Take a look at the Wiki page on "Constellation Overlay" for tips on how to align the constellations with your actual skies.  Be warned: it's a pain in the butt and next to impossible to get perfect, but I've been able to get it "good enough".

WheresWally, 2023-04-18:
Thanks Eric, I would be happy just to not see Polaris. As an Australian I have serious doubts it actually exists as I have never seen it.
LOL I love flat earthers ... they say the darnest things.

EricClaeys, 2023-04-19:
@WheresWally   Those are the defaults it uses when it can't find the ones you specify, or yours are "bad", but I just looked at yours and they seem fine.
As a test, try changing the `az` to 90 and see if the constellations rotate 90 degrees.  Let me know.

EricClaeys, 2023-04-19:
Isn't Polaris next to the Southern Cross?

WheresWally, 2023-04-19:
LOL, yeah I had the AZ at 180 to get Polaris to the North instead of right over Crux.
yes setting az to 90 puts Polaris on the side.
I can change the overlayOffsetTop: 0,  and move the overlay up/down.

EricClaeys, 2023-04-19:
@WheresWally,   the virtual sky code appears not to see your latitude and longitude, although it does see your az.  I'm not sure why.  We have a new Allsky and Allsky Website version coming out soon that I know works.
If you need it working prior to then and you're comfortable editing a file, you can replace the default values in `virtualsky.js`.  Then press Shift-F5 to force your browser to re-read the file.

WheresWally, 2023-04-19:
totally a hacker, just tell me what to change.
the lat/lon values are only in the config.js file not the virtualsky.json file.
BTW I am editing these via the Administration panel. "</> Editor" feature.
I assume it is pointing me to the correct files to edit. 
oh, the editor is editing virtualsky.json, you mentioned virtualsky.js.

EricClaeys, 2023-04-19:
@WheresWally   The file you want to edit is `virtualsky/virtualsky.js` (a Javascript file).  Look starting line 269:

~~~
	// Set location on the Earth
	this.setLongitude(-119.86286);
	this.setLatitude(34.4326);
~~~

Replace with your values, then shift-F5 to force your browser to grab the new file.

EricClaeys, 2023-04-19:
The WebUI only allows editing `config.js` and `virtualsky.json` for an Allsky Website running on the Pi.  If you have a remote Website you'll need to edit the files on the server.
The next Allsky release makes this a gazillion times easier.

WheresWally, 2023-04-19:
it is only on the Pi.
I also dont have Timelapses woking.
could that be because I only have a 1GB RPi 4?

WheresWally, 2023-04-19:
ok that worked. Hacked the defaults to my location.
thanks Eric

WheresWally, 2023-04-19:
I went looking for the sun to see if I can align the overlay a bit now that it is day.
Of the Moon is right there too.
Oh yeah we have a solar eclipse today, shame it is on the other side of Australia.
[image]

EricClaeys, 2023-04-19:
@WheresWally    If you have any hints on how to align the constellations, let me know.  I've spent HOURS trying and writing the Wiki page, and I STILL have problems.  Any help would be appreciated.

The "Troubleshooting -> Timelapse" Wiki page will get you working.  I think it's fixed everyone who's done what it says.

WheresWally, 2023-04-19:
Thanks again. It must work well enough as most seem to have it well enough aligned.
I guess I wait now till dark.  I shall wait for the new version and then do a complete reinstall to fix the Time Lapse issue.
Is there an entry on fixing the Timelapse?

EricClaeys, 2023-04-19:
The "Troubleshooting -> Timelapse" Wiki page will tell you how to fix the timelapse issue.

WheresWally, 2023-04-19:
ah thanks.
Troubleshooting  you say.... who would have thought of that... (no me #sad)

WheresWally, 2023-04-21:
Overlay  NICE and Aligned now.
What I did was a screen cap, found two bright stars and drew lines between them using the overlay and the real ones.
That gave me two distances and dividing one by the other I had a scale, in my case 1.3 x so i multiplied 875 by 1.3 to get 1100.
And the angle difference was 10 deg so I had to add 10 to 180 to get 190.
Then I just had to fiddle topoffset to line them up vertically.
finally adjust LeftOffset to get them over the top of each other.
[image]
[image]

EricClaeys, 2023-04-21:
@WheresWally   Thanks for doing this.  I want to make sure I understand so I can add this to the documentation and give you credit, if that's ok.

1. You took a screen shot of the Allsky Website with the constellation overlay showing.
2. You put that screenshot into Photoshop or some other image editing program.
3. You picked two bright OVERLAY stars (Alpha Centauri A and Sirius in your case) and drew a red line between them and measured the length of the line in pixels and it's degree rotation (484 px and 26 degrees).  In theory you could have done this during the day.
4. You found those same ACTUAL stars in the image and drew a line between them and measured the length and rotation (649 px and 16 degrees).
5. You did some math:
   * line_2_px / line_1_px  = 1.3, which means the ACTUAL stars were 30% farther apart than the OVERLAY stars so you multiplied the default overlay size of 875 pixels by 1.3 to get 1100 pixels and used that for the overlay
   * line_1_deg - line_2_deg = 10 deg, which is added to "az" to rotate the overlay clockwise (to 190 degrees in your case).

Is the second picture just showing that the ACTUAL stars align with the OVERLAY stars?

Eric

WheresWally, 2023-04-21:
Yes, I think that is all correct.
I used Paint.net as it is a free bit of software and a good beginner tool, powerful enough with out the Mount Everest Learning curve.
I did fall back  to trail and error at the end.
I should have made the offset adjustments the same way. Take a screen gran (with GreenShot) and then make a change, Take a second screen grab, and measure the amount of movement. Then you can calculate the amount needed to get it spot one.

Yes the second picture is the stars and overlay together, but you cant see the stars as they are all hidden by the overlay, and that is what you will see when it is spot on.