AllskyTeam/allsky #2535: [BUG]Allsky Camera Installer ended with error message 'Failed to fetch...

AdKooiman, 2023-04-08:
### Environment

* Camera:  ZWO (including model)  or  RPi
* OS: Bullseye
* Allsky version:
     * The newest software downloaded on April 8, 2023
* Pi model and amount of memory (4B with 4GB)

### Bug Description

Allsky Camera Installer ended up with an error message saying:
E: Failed to fetch http://mirror.nl.leaseweb.net/raspbian/raspbian/pool/main/libc/libcdio-paranoia/libcdio-cdda2_10.2+2.0.0-1+b1_armhf.deb  Undetermined Error [IP: 5.79.108.33 80]

The installation terminated.
Your help will be appreciated!

Thanks in advance,
Ad Kooiman

### Log / configuration files

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.
[InstallationLog.txt](https://github.com/thomasjacquin/allsky/files/11184252/InstallationLog.txt)

EricClaeys, 2023-04-09:
@adKooiman, Try running the installation again but don't do another "git clone" first.  The network may have had an issue, or the mirror.nl.leasweb.net server may have had a problem.

If this resolves the problem, please close this issue.

AdKooiman, 2023-04-10:
Thanks for your help Eric, installing again fixes it.

closed 2023-04-10 by AdKooiman

snowformatics, 2023-06-16:
Hi,

I also have problem to install allsky, I got 3 times a  "Failed to fetch" error message but from different sources:

`E: Failed to fetch http://debian.bio.lmu.de/raspbian/raspbian/pool/main/d/djvulibre/libdjvulibre21_3.5.28-2_armhf.deb  Connection timed out [IP: 2001:4ca0:4300::1:22 80]`

`Failed to fetch http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian/pool/main/p/pocketsphinx/pocketsphinx-en-us_0.8+5prealpha+1-13_all.deb  Unable to connect to ftp.gwdg.de:http: [IP: 134.76.12.6 80]`

`E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libv/libvdpau/vdpau-driver-all_1.4-3_armhf.deb  Unable to connect to raspbian.raspberrypi.org:http: [IP: 93.93.128.193 80]`

Any ideas what could I try?
Thanks
Stefanie

EricClaeys, 2023-06-17:
This is either a network problem or a problem with the server that contains the packaged is downloading.  It is not an Allsky problem. 

The best thing is to wait a bit and try again.