AllskyTeam/allsky #2571: Welcome to AllSky Discussions!

Michel85300, 2023-04-27:
Bonjour
J'ai un souci avec le time lapse, apparemment il est bien créer mais il ne s'enregistre pas sue le site alors que les Kéogrammes et les Startrails eux sont bien enregistrés et accessible.

EricClaeys, 2023-04-27:
@Michel85300   Are you trying to view the timelapse video via the WebUI's "Images" link, or via the Allsky Website?

Please try the following:
~~~
systemctl stop allsky       # stop Allsky
sudo truncate -s 0 /var/log/allsky.log      # clear the Allsky log
cd ~/allsky
mkdir images/test
# copy about a dozen images from images/20230426  to  images/test
scripts/generateForDay.sh -t test     # this will create a timelapse from the dozen images you copied
# If that worked, then execute:
scripts/uploadForDay.sh -t test        # this will upload the timelapse to your Allsky Website
# ATTACH (don't copy and paste) /var/log/allsky.log
systemctl start allsky
~~~

If you still don't see the timelapse on the Allsky Website, what DO you see when you click on the timelapse icon?

EricClaeys, 2023-05-15:
@Michel85300   Any update?

EricClaeys, 2023-06-13:
Closing Issue - no response from user in 6 weeks.

closed 2023-06-13 by EricClaeys