AllskyTeam/allsky #2579: [BUG] 'LAN Dashboard' (/index.php?page=LAN_info) only shows IPv4 addresses

Elwell, 2023-05-01:
### Environment

* Camera:  ZWO ASI224MC
* OS: Bullseye / 6.1.21-v7+
* Allsky version: v2023.05.01
* Pi model: Raspberry Pi 2 Model B Rev 1.1, 1G RAM

### Bug Description

Fresh install of the 'dev' branch on a pi2 - once installed navigate to LAM Dashboard page, it's only showing ipv4 not ipv6 info
It's not being parsed by functions.php
~~~
function parse_ifconfig($input, &$strHWAddress, &$strIPAddress, &$strNetMask, &$strRxPackets, &$strTxPackets, &$strRxBytes, &$strTxBytes) {
        preg_match( '/ether ([0-9a-f:]+)/i', $input, $result );
        $strHWAddress = $result[1];
        preg_match( '/inet ([0-9.]+)/i', $input, $result );
        $strIPAddress = getVariableOrDefault($result, 1, "[not set]");
~~~

I also haven't checked how >1 ipv4 address behaves, or vlan interfaces

### Log / configuration files
example output from ifconfig eth0
~~~
pi@picam:~ $ ifconfig eth0
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.81  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ed81:11c5:e93d:ddee  prefixlen 64  scopeid 0x20<link>
        inet6 2403:5801:8055:0:e067:e45:6bf7:c04e  prefixlen 64  scopeid 0x0<global>
        ether b8:27:eb:63:2f:83  txqueuelen 1000  (Ethernet)
        RX packets 14656  bytes 3543826 (3.3 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4457  bytes 4042581 (3.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
~~~
notice none of the inet6 addresses are used

EricClaeys, 2023-06-20:
Which of the inet6 addresses should be used?
What is the difference between them?

EricClaeys, 2023-07-04:
@elwell, Which of the inet6 addresses should be used?  What is the difference between them?

@Alex-developer, Alex, something to implement when updating the WebUI.

Alex-developer, 2023-07-04:
> @elwell, Which of the inet6 addresses should be used?  What is the difference between them?
> 
> 
> 
> @Alex-developer, Alex, something to implement when updating the WebUI.

I'll add ipv6 to the new webui

Elwell, 2023-07-04:
both of them. fe80 is local lan (scopeid link) and the other is public (scopeid global). It's just like ipv4 - you may have many assigned to an interface, for example

[root@alfred ~]# ip --brief a show dev eth2
eth2             UP             192.168.1.2/24 192.168.1.250/24 2403:5801:8055::215:250/64 fe80::34e3:4cff:fec4:be1e/64
[root@alfred ~]#

see https://en.wikipedia.org/wiki/Link-local_address for an overview of link local address usage

EricClaeys, 2024-11-28:
Closing.  Moved this request to [Issue #2835](https://github.com/AllskyTeam/allsky/issues/2835) for the "new WebUI".

closed 2024-11-28 by EricClaeys