AllskyTeam/allsky #2603: Overlay editor, display sample data inop[BUG]

BobW55, 2023-05-03:
In the overlay editor, the display sample data no longer displays data.

[image]

Alex-developer, 2023-05-03:
This could be a couple of things

- There is no 'overlaydebug.txt' file in ~/allsky/tmp - If its not there check that the overlay module is enabled in the module manager. Also check the allsky settings to make sure the overlay type is set to 'module'
- a dodgy 'format' in one of the fields. Check each fields format to make sure they are valid

Cheers

Alex

BobW55, 2023-05-03:
overlaydebug.txt' file in ~/allsky/tmp
File exist.
This was working perfectly until the update tonight.
All overlays are still working as before.
Yes ovelay method is set to module.   and the daytime and night time overlays are enabled
[image]

Alex-developer, 2023-05-04:
I'll take a look when back from vacation

BobW55, 2023-05-04:
No worries.    everything else is working very nice
Enjoy your holiday

Lorenzi70, 2023-05-19:
Hi Bob, as a side note, how do you add th enice weather forecast short message in the bottom left corner? A custom made py code shared in this repositori?
Thanks
Marco

BobW55, 2023-05-19:
@Lorenzi70 
My weather station software (Cumulus) uploads a text file to my web server every minute.
I wrote a script to manipulate it into the format allsky can read.   Because the FORECAST can be a very long text string, I worte another script that breakes it up into 4 parts based on the . (period) in the text string.
I can then place it all on the overlay of my live images.
Here is a sample of it.
~~~
"rfall":"0.00",
"rrate":"0.00",
"rrateTM":"0.00",
"hum":"52",
"humTL":"50",
"humTH":"91",
"inhum":"51",
"SensorContactLost":"0",
"forecast":" Increasing clouds with little temperature change. ",
"tempunit":"F",
"windunit":"mph",
"pressunit":"in",
"rainunit":"in",
"temptrend":"+3.7",
"TtempTL":"03:16",
"TtempTH":"13:31",
"TdewpointTL":"00:00",
~~~

BobW55, 2023-05-19:
@Alex-developer 
@EricClaeys 
**Dont know why, but it is now working.**  
I know I had to do a complete power cycle the other day, I dont know if that had any beearing on it or not.
I had rebooted my PI several times after the update, and it did not work then.

Alex-developer, 2023-05-23:
Closing as cannot reproduce and user no longer has the issue

closed 2023-05-23 by Alex-developer

lumdiniz, 2023-06-26:
olá,,,, como podem ver no vídeo timelapse, https://projetoicaro.qsl.br/allsky/videos/allsky-20230624.mp4 a sobreposição sumiu, depois quase no final do vídeo voltou novamente sozinha, agora não ativa mais a sobreposição, não foi alterado nada no editor de sobreposição, alguém pode me ajudar?

EricClaeys, 2023-06-26:
@Alex-developer, can you help?
Translation:
hello,,,, as you can see in the timelapse video, https://projetoicaro.qsl.br/allsky/videos/allsky-20230624.mp4 the overlap disappeared, then almost at the end of the video it came back again alone, now it no longer activates the overlay, nothing has changed in the overlay editor, can anyone help me?

BobW55, 2023-06-26:
@lumdiniz 
What camera are you using?
What are your settings?  
Your video is beautiful.

Alex-developer, 2023-06-26:
> @Alex-developer, can you help? Translation: hello,,,, as you can see in the timelapse video, https://projetoicaro.qsl.br/allsky/videos/allsky-20230624.mp4 the overlap disappeared, then almost at the [...]

Funnily enough possible as I have just had a similar problem on mine. It was due to an error with the planets calculations. Disabling the planets fixed it for me.

Ideally we will need to the logs to see what errors are being thrown.

lumdiniz, 2023-06-27:
> @lumdiniz Que câmera você está usando? Quais são suas configurações? Seu vídeo é lindo.

obrigado... é uma camera asi224mc

lumdiniz, 2023-06-27:
> > @Alex-developer, você pode ajudar? Tradução: alô,,,, como podem ver no vídeo timelapse, https://projetoicaro.qsl.br/allsky/videos/allsky-20230624.mp4 a sobreposição sumiu, aí quase no final do vídeo voltou novamente sozinho, agora não ativa mais o overlay, nada mudou no editor de overlay, alguém pode me ajudar?
> 
> Curiosamente possível, pois acabei de ter um problema semelhante no meu. Foi devido a um erro com os cálculos dos planetas. Desativar os planetas corrigiu isso para mim.
> 
> Idealmente, precisaremos dos logs para ver quais erros estão sendo lançados.

voltou a funcionar novamente do nada... nao da para entender o que aconteceu... onde esta essa opção de cálculos dos planetas?

lumdiniz, 2023-06-27:
> > @Alex-developer, você pode ajudar? Tradução: alô,,,, como pode ver no vídeo timelapse, https://projetoicaro.qsl.br/allsky/videos/allsky-20230624.mp4 a sobreposição sumiu, aí quase no final do vídeo voltou novamente sozinho, agora não ativa mais o overlay, nada mudou no editor de overlay, alguém pode me ajudar?
> 
> Curiosamente possível, pois queria ter um problema semelhante no meu. Foi devido a um erro com os cálculos dos planetas. Desativar os planetas corrigiu isso para mim.
> 
> Idealmente, precisaremos dos logs para ver quais erros estão sendo lançados.

voce esta correto, ativei as 3 opções Include Planets, Include Sun, Include Moon, a sobreposição nao aparece na allsky. Desativar os planetas corrigiu para mim.

EricClaeys, 2023-07-03:
@Alex-developer,
See @lumdiniz's last post:
> You are correct, I activated the 3 options Include Planets, Include Sun, Include Moon, the overlay does not appear in allsky. Disabling the planets fixed it for me.